DeskPi-Team/deskpi #1: USB Hub

AllonisDave, 2020-11-25:
Do you have any plans to switch to the ASMedia USB hub instead of the JMicron one? The JMicron has serious UASP issues on the Raspberry PI requiring changes to the cmdline.txt file to make it work. The ASMedia chicpset supports UASP.

Marach79, 2020-12-05:
What kind of changes?

64Spaces, 2020-12-05:
I found this solution: setting quirk in cmdline.txt

[Fix for slow USB SSD](https://www.raspberrypi.org/forums/viewtopic.php?t=245931)

AllonisDave, 2020-12-05:
Right you have to disable UASP support if you want to use a m.2 sata stick that enables UAS. Makes the Pi work but you can double the i/o throughput by using a USB hub that is compatible with the Raspberry PIs UAS boot mode. The ASMedia chicpset is one that is Pi boot compatible.

turacos, 2020-12-07:
Same issue, with my SSD drive not an m2 stick by the way. I need to disable the uas support to be able to boot correctly with the quirks config.
Is there any fix planned for that ?

42648594, 2020-12-12:
I’m glad I checked these issues before I decided to buy one. This issue has cropped up in other RPi4 cases, such as the Nespi one, which I returned because of this problem. Massively throttling IO performance is a dealbreaker for me, especially when one of the main features of this case is a slot for a 2.5” SSD. It’s surprising to me that multiple companies could select USB controllers without checking basic protocol specs, especially for such a well-known issue for the RPi4. There’s even a stickied thread on the official Raspberry Pi troubleshooting forum for this exact issue:

https://www.raspberrypi.org/forums/viewtopic.php?t=245931

Unfortunately, there is no software fix for this. Like AllonisDave mentioned, they’ll need to choose a less outdated controller, that supports UAS. Once that happens, I’ll be a customer.

fmorency, 2020-12-13:
Same issue here with a Crucial MX500 500Gb SSD plugged in the DeskPi.

The usb-storage quirks seems to have stabilized things as I am now able to boot from USB.

However, for some reason, the system just stops working if I attempt to use the WiFi. I read that 2.4Ghz and USB3 interfere together.

WZ9V, 2020-12-14:
It is deeply frustrating to have to use USB Quirks to get this to boot from a SSD since that and cooling are the major features of this case.  At least update the instructions for SSD boot that do not mention the need for USB Quirks and cause you to waste a ton of time troubleshooting what should work out of the box.  Please consider moving to the ASMedia chipset in the adapter board in the future and making things right for those of us who put our faith in you delivering a working solution that did not require disabling needed features in order to get to to work.  Very unhappy.

theo539, 2020-12-15:
Same problem here. Slpw boot. Unstable. Chrashing. Very unhappy about that.

tapd, 2020-12-17:
I waited so long as was so excited about this case but so far I've wasted so much time and cannot get my SSD to boot consistently. Once I have it working it is not recognised after a reboot, even with the USB Quirks applied. The fan seems to also turn itself off even though the CPU temp is within range when it should be running. So disappointed with this, I hope there are plans to rectify these issues especially for us all that jumped on board hoping this was the best solution available in that price bracket.

tabish121, 2020-12-19:
I'm also experiencing extreme instability and locks when using an m.2 Kingston drive and have Wifi enabled.  I waited a long time to get this case and was excited to have a nice case for my desktop development pi.  Now I'm left with a crappy paperweight and no response from the company to my emails

vaanja, 2020-12-20:
This product is useless until JMicron chip is replaced with something working with RPi4. People are buying this product because the want to have rock solid a fast SSD drives in nice enclosure. If the only way is to use quirks which degrades overall performance it's significant change and I hope DeskPi/52pi.com will replace JMicron board for something with full UASP support for free ASAP.

WZ9V, 2020-12-20:
@vaanja - I agree. The JMicron is pretty well known not to work.  The youtube video showed it working, I wonder if it is a firmware issue with the JMicron because the Odroid forums imply there is a firmware that works with their HC1/HC2 boards and the JMicron chipset.  I know my two cases do not.

jeongsshin, 2020-12-21:
You guys should provide us with a UASP supported SSD bridge board. It's really frustrating not working with a maximum bandwidth SSD access. I didn't know about this issue before I bought this. I tried a number of firmwares for the JMICRON bridge chipset and nothing worked. The UASP doesn't seem to be supported.

yoyojacky, 2020-12-22:
> Do you have any plans to switch to the ASMedia USB hub instead of the JMicron one? The JMicron has serious UASP issues on the Raspberry PI requiring changes to the cmdline.txt file to make it work. The ASMedia chicpset supports UASP.

we are working on it. thanks for your tips.

yoyojacky, 2020-12-22:
> What kind of changes?
> I'm not very experienced in working with Linux. 😊

he means to change a chip on board.

yoyojacky, 2020-12-22:
> planned for that ?

we are working on it, but you can check out my tutorial video : https://youtu.be/wUHZb9E_WDQ

yoyojacky, 2020-12-22:
> You guys should provide us with a UASP supported SSD bridge board. It's really frustrating not working with a maximum bandwidth SSD access. I didn't know about this issue before I bought this. I [...]

Please try to refer to my tutorial video and try again? I have used a kington 120GB M.2 SATA SSD booting up my deskpi.
https://youtu.be/wUHZb9E_WDQ

closed 2020-12-22 by yoyojacky

Mister-42, 2020-12-22:
Did I miss it, or did you _not_ show uasp is active after booting from USB?

42648594, 2020-12-22:
I do not see UASP being shown as active in that tutorial video

WZ9V, 2020-12-22:
I did everything in that video and I still had to use USB Quirks to get the SSD to boot.

64Spaces, 2020-12-22:
The issue for me is that although it will boot from USB, it takes a full two minutes, instead of 20 seconds when using the micro SD card. Also, loading Chromium from the SSD takes sometimes 60 seconds or more the first time instead of 2 seconds.  After installing the quirks, it boots in 20 seconds or less and Chromium opens in 2 seconds.