DeskPi-Team/deskpi #2: /usr/bin/pwmFanControl fails to run on 64bit systems

donpezet, 2020-11-27:
I'm not sure if it is like this for the other scripts, but the script "install-ubuntu-mate.sh" installs a 32bit version of pwmFanControl which fails to run on 64bit Ubuntu.

donpezet, 2020-11-27:
I compiled pwmFanControl and fanStop from source to generate 64bit binaries. They worked perfectly without modification. You might want to consider checking for a 64bit OS to the script and throwing an error if it is a 64bit system, or add 64bit binaries.

agarthetiger, 2020-12-03:
Ditto with installing on the 64bit RPiOS. Please consider adding the built binaries for 64bit OS or run `make` on the system to build them. The RPiOS 64bit lite didn't need any additional dependencies to be installed for make to build the new binaries successfully.

yoyojacky, 2020-12-22:
Supported OS 
Raspbian OS - 32bit version: 2020-08-12 Raspberry Pi OS.
Ubuntu-mate OS - 32bit version:20.04.1-desktop--armhf+raspi Ubuntu mate 20.04 LTS 32Bit.
Manjaro OS - 32bit verison: 20.2 Manjaro XFCE 20.2
Kali-linux-arm OS - 32bit version: 2020.4 Kali Linux RaspberryPi 2, 3, 4 and 400.

Unsupport OS
For Windows IoT OS
Unsupported due to lacking of driver.
Testing version: Midnight falcon
Other 64bit is not supported at this time.

closed 2020-12-22 by yoyojacky

agarthetiger, 2020-12-22:
@yoyojacky Please consider putting this information on the product description, so that potential purchasers know there are unsupported OS versions for this hardware. 

Do you have a timeframe for supporting any 64bit OS?