DeskPi-Team/deskpi #4: Add warning to install.sh if git is not installed

agarthetiger, 2020-12-03:
On RPiOS 64bit lite, git is not installed by default. The installer fails but reboots so the output indicating why the install fails may be lost. Please add a warning and abort the install if git is not found.

yoyojacky, 2020-12-22:
Supported OS
Raspbian OS - 32bit version: 2020-08-12 Raspberry Pi OS.
Ubuntu-mate OS - 32bit version:20.04.1-desktop--armhf+raspi Ubuntu mate 20.04 LTS 32Bit.
Manjaro OS - 32bit verison: 20.2 Manjaro XFCE 20.2
Kali-linux-arm OS - 32bit version: 2020.4 Kali Linux RaspberryPi 2, 3, 4 and 400.
Unsupport OS
For Windows IoT OS
Unsupported due to lacking of driver.
Testing version: Midnight falcon

closed 2020-12-22 by yoyojacky

agarthetiger, 2020-12-22:
@yoyojacky I don't see how your response answers or addresses my question? The same thing applies to RPiOS/Raspbian 32bit lite