DeskPi-Team/deskpi #5: Ubuntu 20.10 64Bit

Marach79, 2020-12-05:
Does the installation work for Ubuntu 20.10 64Bit and not mate?
I'm not experienced in working with Linux. Only Deskpi-config seems not to work with 64Bit, right?

Marach79, 2020-12-06:
In summary, I'm a beta tester and can't work with the case without much Linux knowledge. I thought I could build a small pc with an internal 2.5 Ssd but nothing works. RaspiOS is not an option. Too bad!

mbernardi1961, 2020-12-08:
I would also like to be able to to use Ubuntu 20.10 64bit, with the front USB ports and control of the fan.

yoyojacky, 2020-12-22:
Supported OS
Raspbian OS - 32bit version: 2020-08-12 Raspberry Pi OS.
Ubuntu-mate OS - 32bit version:20.04.1-desktop--armhf+raspi Ubuntu mate 20.04 LTS 32Bit.
Manjaro OS - 32bit verison: 20.2 Manjaro XFCE 20.2
Kali-linux-arm OS - 32bit version: 2020.4 Kali Linux RaspberryPi 2, 3, 4 and 400.
Unsupport OS
For Windows IoT OS
Unsupported due to lacking of driver.
Testing version: Midnight falcon

closed 2020-12-22 by yoyojacky