DeskPi-Team/deskpi #6: Wifi And Wireless Keyboard/Mouse Problems

RasPiMon, 2020-12-05:
I have No Problems with Wifi or Wireless Keyboard/Mouse but when I Plugin USB extension for the Front USB and  SSD that's when I Loose Wifi and K/M Signals.   Any Ideas

fmorency, 2020-12-13:
I encountered the same issue. The front USB ports were working perfectly with the included SD Card, but now I can't seem to make them work using Manjaro ARM.

RasPiMon, 2020-12-13:
Have to install DeskPi-Pro script in Terminal...

cd ~/deskpi/
chmod +x install.sh
sudo ./install.sh

yoyojacky, 2020-12-22:
> I have No Problems with Built in Wifi or Wireless Keyboard/Mouse Plugged into Pi4 but when I Plugin USB extension for the Front USB and SSD that's when I Loose Wifi and K/M Signals. All this Happens when I use Twister OS. With 32GB card that was included with my DeskPi Pro it works Perfectly.

Did you use the wall charger in package ?

RasPiMon, 2020-12-22:
I got it Working Perfectly Now... Just made sure all screws were tight at Bottom of Case.  Also Need to Load DeskPi-Pro Script.

closed 2020-12-23 by yoyojacky