DeskPi-Team/deskpi #7: Workaround for slow SSD boot

bsv116, 2020-12-07:
There is a problem with SSD boot as the bridge chip from JMicron that is used in DeskPi is not UAS compatible. As a workaround, the section below can be added to the instruction "How to boot from USB SSD/HDD?". Further details and explanations can be found at [https://www.raspberrypi.org/forums/viewtopic.php?t=245931](https://www.raspberrypi.org/forums/viewtopic.php?t=245931)

**Disable UAS protocol for the JMicron chip**
`sudo nano /boot/cmdline.txt`
Insert `usb-storage.quirks=152d:0562:u ` first on the first line. Please note the trailing space.
Save and exit with ctrl+s ctrl+x

Mister-42, 2020-12-08:
Duplicate of #1

yoyojacky, 2020-12-23:
Please refer to :https://youtu.be/wUHZb9E_WDQ

closed 2020-12-23 by yoyojacky