DeskPi-Team/deskpi #8: Loss of Wifi Network with SSD expansion board plugged in

ewcook, 2020-12-09:
This is reported in the /var/log/syslog when the USB SSD board is plugged in

Dec  8 15:49:59 raspberrypi dhcpcd[441]: wlan0: carrier lost
Dec  8 15:49:59 raspberrypi dhcpcd[441]: wlan0: deleting address fe80::f92e:f60f:536:85e0
Dec  8 15:49:59 raspberrypi avahi-daemon[410]: Withdrawing address record for fe80::f92e:f60f:536:85e0 on wlan0.
Dec  8 15:49:59 raspberrypi avahi-daemon[410]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::f92e:f60f:536:85e0.
Dec  8 15:49:59 raspberrypi avahi-daemon[410]: Interface wlan0.IPv6 no longer relevant for mDNS.
Dec  8 15:50:00 raspberrypi kernel: [  235.016190] v3d fec00000.v3d: MMU error from client L2T (0) at 0x2401000, pte invalid
Dec  8 15:50:00 raspberrypi avahi-daemon[410]: Withdrawing address record for 192.168.1.17 on wlan0.
Dec  8 15:50:00 raspberrypi avahi-daemon[410]: Leaving mDNS multicast group on interface wlan0.IPv4 with address 192.168.1.17.
Dec  8 15:50:00 raspberrypi avahi-daemon[410]: Interface wlan0.IPv4 no longer relevant for mDNS.
Dec  8 15:50:00 raspberrypi dhcpcd[441]: wlan0: deleting route to 192.168.1.0/24
Dec  8 15:50:00 raspberrypi dhcpcd[441]: wlan0: deleting default route via 192.168.1.1
Dec  8 15:50:21 raspberrypi kernel: [  256.256862] usb 2-1: USB disconnect, device number 5
Dec  8 15:50:21 raspberrypi kernel: [  256.257715] xhci_hcd 0000:01:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.

Then when it is removed

Dec  8 15:50:27 raspberrypi dhcpcd[441]: wlan0: carrier acquired
Dec  8 15:50:27 raspberrypi dhcpcd[441]: wlan0: IAID 32:25:08:91
Dec  8 15:50:27 raspberrypi dhcpcd[441]: wlan0: adding address fe80::f92e:f60f:536:85e0
Dec  8 15:50:27 raspberrypi avahi-daemon[410]: Joining mDNS multicast group on interface wlan0.IPv6 with address fe80::f92e:f60f:536:85e0.
Dec  8 15:50:27 raspberrypi avahi-daemon[410]: New relevant interface wlan0.IPv6 for mDNS.
Dec  8 15:50:27 raspberrypi avahi-daemon[410]: Registering new address record for fe80::f92e:f60f:536:85e0 on wlan0.*.
Dec  8 15:50:27 raspberrypi dhcpcd[441]: wlan0: soliciting an IPv6 router
Dec  8 15:50:27 raspberrypi dhcpcd[441]: wlan0: rebinding lease of 192.168.1.17
Dec  8 15:50:27 raspberrypi dhcpcd[441]: wlan0: probing address 192.168.1.17/24
Dec  8 15:50:33 raspberrypi dhcpcd[441]: wlan0: leased 192.168.1.17 for 86400 seconds
Dec  8 15:50:33 raspberrypi avahi-daemon[410]: Joining mDNS multicast group on interface wlan0.IPv4 with address 192.168.1.17.
Dec  8 15:50:33 raspberrypi avahi-daemon[410]: New relevant interface wlan0.IPv4 for mDNS.
Dec  8 15:50:33 raspberrypi avahi-daemon[410]: Registering new address record for 192.168.1.17 on wlan0.IPv4.
Dec  8 15:50:33 raspberrypi dhcpcd[441]: wlan0: adding route to 192.168.1.0/24
Dec  8 15:50:33 raspberrypi dhcpcd[441]: wlan0: adding default route via 192.168.1.1
Dec  8 15:50:36 raspberrypi systemd[1]: Started Session 4 of user pi.
Dec  8 15:50:40 raspberrypi dhcpcd[441]: wlan0: no IPv6 Routers available

It also has trouble booting from SSD. I will try the quirk fix mentioned in another post.

fmorency, 2020-12-13:
As I mentioned in #1, I also have this issue. I just can't use the WiFi at all or else the system stops working. I read that 2.4Ghz WiFi can interact with USB3. Not sure what the source of this issue is.

yoyojacky, 2020-12-22:
> As I mentioned in #1, I also have this issue. I just can't use the WiFi at all or else the system stops working. I read that 2.4Ghz WiFi can interact with USB3. Not sure what the source of this issue is.
 
 The 2.4G USB dongle sometimes interferes with the wifi of the device. If the wifi cannot be configured at the first startup, it is because the wlan country is not set on your Raspbian system.
you can try to use `raspi-config` tool set it up.
```
sudo raspi-config
```

closed 2020-12-22 by yoyojacky