DeskPi-Team/deskpi #10: Berry boot does not install any working OS

beyondlsd, 2020-12-14:
I have tried to use berry boot to install an OS to a 2.5" SSD and nothing ever boots up properly.  

I modified the config and the cmdline files with the quirk modification and the front usb lines.  

Tried to install rasbian os 64 full, manjaro, and ubuntu.   Rasbian doesnt load up at all.  Manjaro and Ubuntu load up but network and other functions do not work.  I can start a browser but nothing connects to the internet otherwise.  Applications crash constantly

developer-haku, 2020-12-14:
> I have tried to use berry boot to install an OS to a 2.5" SSD and nothing ever boots up properly.
>
> I modified the config and the cmdline files with the quirk modification and the front usb lines. [...]

@beyondlsd BerryBoot won't work nicely with this case at this point. I tried BerryBoot with the included M.2 SATA SSD slot, it took me over 4 minutes to boot the system, and system freeze every now and then. The power also won't disconnect after system shutdown, the light of the power button won't turn red.

I emailed them, they told me the current chip is not suitable for USB booting, which I think it is a big NO-NO excuse. I think everyone who bought this case was planned to use SSD boot. 

In my case, for testing I took my M.2 SSD off the board and put it in a M.2 SSD enclosure with USB Type-C to USB-A cable connection, and connect it to the Pi through the USB 3.0 port on the Pi. The result is, BerryBoot works perfectly. Before it took me over 4 minutes to boot up the system, this time it took less than 10 seconds, just unbelievable to me.

I thought that I should just give up with the included SSD connection and using my own SSD enclosure to install the system, then another issue came up. Which is the issues you are having right now. I couldn't control the fan with BerryBoot operating system, even I installed the script that they provided from this repo. 

I took my time to do another test to see if the board have some sort connection issue with the fan. I unplug the SSD and reinstall the MicroSD card with Raspberry PI OS 32 bit, the official RaspPi OS. I reinstall the script in this new system. The fan works perfectly this time. There may be a problem with 64 bit OS, or just won't work with BerryBoot. #2 this issues mentioned about the script won't run in 64 bit OS. Maybe I will do another test with BerryBoot and install a 32 bit OS later tonight or in the next few days if I can't do it tonight.

I will update in this issue if I get something to work and have an email reply from the DeskPi team.

***

**[Additional reading if this helps someone or you want to know more about the situation]**
[RaspBerryPi 4B Cases that support HDD/SSD (NEW DeskPi-Pro), [! BAD DEAL !] ](https://diysolarforum.com/threads/raspberrypi-4b-cases-that-support-hdd-ssd-new-deskpi-pro-bad-deal.13351/)

This is a post I found a week ago. Someone named Steve post this in a forum called "DIY Solar Power Forum". He got his DeskPi Pro case early, did some testings with his units and ended up with disappointments, and some other people posted some email replies from the DeskPi team in the forum post.

I read those email replies and look like they are trying to make another updated model of fix the issues. I am not sure this is true or not. I am still waiting for their another email reply. Hoping they will make things right.

developer-haku, 2020-12-15:
UPDATE:
Couldn't find an 32 bits BerryBoot OS for the Pi4, but I tested 64 bits OS running on the MicroSD card. I ran an Ubuntu Desktop 64 bit and installed the script. Fan works fine. So the fan is just not working with BerryBoot.

@yoyojacky @deskpi They should add a default config to the fan at the firmware level. Let it just run at a normal quite speed without the script, that's better than don't spin at all.  The heat sink is hot when the fan is not spinning. Although the heat sink is a fat one, passive cooling is not good to be honest.

yoyojacky, 2020-12-22:
> I have tried to use berry boot to install an OS to a 2.5" SSD and nothing ever boots up properly.
>
> I modified the config and the cmdline files with the quirk modification and the front usb lines. [...]

This is because your system is not properly configured. What does it have to do with the DeskPi shell? This is just the fan driver.

yoyojacky, 2020-12-22:
> BerryBoot won't work nicely with this case at this point. I tried BerryBoot with the included M.2 SATA SSD slot, it took me over 4 minutes to boot the system, and system freeze every now and then. [...]

Thanks for your help, and recently we are figuring out the problem is the chip on SSD adapter board is not work very well on USB booting, our Teamates are working on it.  and we've tried following OS:
Supported OS
Raspbian OS - 32bit version: 2020-08-12 Raspberry Pi OS.
Ubuntu-mate OS - 32bit version:20.04.1-desktop--armhf+raspi Ubuntu mate 20.04 LTS 32Bit.
Manjaro OS - 32bit verison: 20.2 Manjaro XFCE 20.2
Kali-linux-arm OS - 32bit version: 2020.4 Kali Linux RaspberryPi 2, 3, 4 and 400.
Unsupport OS
For Windows IoT OS: Unsupported due to lacking of driver.
Testing version: Midnight falcon

yoyojacky, 2020-12-22:
> UPDATE:
> Couldn't find an 32 bits BerryBoot OS for the Pi4, but I tested 64 bits OS running on the MicroSD card. I ran an Ubuntu Desktop 64 bit and installed the script. Fan works fine. So the fan is [...]

The reason why we let the fan not spin when it is turned on is because some users feel that the fan is noisy when it has a sound, and hope that most of the time the fan will not be used to dissipate heat. But it should be relatively simple to debug at present, because you only need to send control commands to /dev/ttyUSB0 recognized by your Raspberry Pi to control the fan. For example, sending: pwm_50 means that the fan is adjusted at 50% PWM speed.

closed 2020-12-22 by yoyojacky