DeskPi-Team/deskpi #11: No video via full size HDMI ports

akesde, 2020-12-18:
I am unable to get any video via the full size HDMI ports.  I have verified that the onboard Micro HDMI port is working and I see activity via the green LED so the unit is reading the provided MicroSD card.  I have booted and updated the unit via the provided MicroSD card so it is not a card issue.  I am using the same monitor that I used when I connected to the onboard Micro HDMI port so there is no change in monitors.

ajscilingo, 2020-12-18:
Hi, I'm having the same issue!  I just put mine together and I am not getting video.  I tried with three different monitors and two different HDMI cables. Strangely on one of my monitors if I don't put in an SD Card and I try booting on, I do get video with a message that says "insert SD card,"  However as soon as I reboot with the SD Card the video goes away again.  The other two monitors I've tried don't show video with or without the SD Card.

ajscilingo, 2020-12-18:
I've recorded a video to further display what I'm seeing on the monitor in which I can get video, albeit briefly. 
https://www.youtube.com/watch?v=Oc9W6VTl8Vs

ajscilingo, 2020-12-18:
Got video working on one monitor for now going to test the others.  in /boot/config.txt I had to add the following 
hdmi_ignore_edid=0xa5000080
start_x=0

in the [pi4] section I had to add 
dtoverlay=vc4-fkms-v3d just above the max_framebuffers=2

ajscilingo, 2020-12-18:
Ok did further testing, so far this work around works with only one of my monitors, the other two this doesn't solve the issue.  Also It seems to only work so far with Raspbian OS, I tried this same setting in Ubuntu and it didn't work.  Can someone from DeskPi help with this.

ajscilingo, 2020-12-19:
another setting that works for me on my one monitor but not others is 
hdmi_force_hotplug=1, again seems to only be working with Raspbian OS can't get video with Ubuntu :/

ajscilingo, 2020-12-19:
Did some more testing, my other monitor that is 4K has a DVI port in addition to Display Port and HDMI.  I happen to have a DVI to HDMI Cable.  Sure enough when using this cable I get video with the DeskPi Daughter Card.  So I'm starting to think there is something faulty with the HDMI Handshake when using the DeskPi Ports?  Ubuntu still fails to boot when connected to the DeskPi Daughter Card However.  When I remove my RapberryPi from the DeskPi Daughter Card Video, Power and Audio Ports (SD Card still being read from the daughter card with cable) video works fine using the mini HDMI to HDMI whether I'm booting Raspbian OS or Ubuntu.  So my two questions now are: 1) what is allowing my 4k Monitor to work with Raspbian OS using a DVI to HDMI cable (Something faulty with HDMI Handshake??) and 2) What makes Raspbian OS different from Ubuntu when it comes to talking to the DeskPi Daughter Card's Video, Audio and Power ports?  It would seem that something about the daughter card is causing Ubuntu to kernel panic / not boot as I can't even get an IP - removing the daughter card's Video, Audio and Power ports (Still using the SD Card Connection though from the Daughter Card) and booting using the onboard Mini-HDMI port on my Raspberry pi works fine and I get an IP address right away.  DeskPi please let us know if there is a fix for this or whether we can get a replacement.

akesde, 2020-12-20:
I've had mixed success.  There is no signal EVER on the HDMI port next to the power supply.  I will call it HDMI zero since that is what the Pi4 onboard HDMI port is called.  I am getting decreasingly inconsistent video on the other HDMI  port (HDMI1).  Although I have not exhausted this pattern for the sake of not defining insanity for Einstein, I am finding that I get more favorable results when I not only turn the device off from the power button but also to interrupt power.  I have a power supply with an inline switch and I am so glad I bought this for this troubleshooting.  I estimate I am getting 90%+ success with getting video on the HDMI1 port that is working.

yoyojacky, 2020-12-22:
> I am unable to get any video via the full size HDMI ports. I have verified that the onboard Micro HDMI port is working and I see activity via the green LED so the unit is reading the provided [...]

You can solve this problem by according steps:
Modify /boot/config.txt file and adding this:
```
hdmi_force_hotplug=1
hdmi_group=2
config_hdmi_boost=4
```
Save it and reboot Raspberry Pi.
```
sudo sync
sudo reboot
```

yoyojacky, 2020-12-22:
> Ok did further testing, so far this workaround works with only one of my monitors, an Acer S231HL 1080p Display, the other two monitors are 4K and this doesn't solve the issue. Also It seems to only work so far with Raspbian OS, I tried this same setting in Ubuntu and it didn't work, it looks like this is related to HDMI EDID. Can someone from DeskPi help with this.

You can solve this problem by according steps:
Modify /boot/config.txt file and adding this:
```
hdmi_force_hotplug=1
hdmi_group=2
config_hdmi_boost=4
```
Save it and reboot Raspberry Pi.
```
sudo sync
sudo reboot 
```

yoyojacky, 2020-12-22:
Solution 2
Unplug HDMI cable and shift it to HDMI1 which is near by 3.5mm audio jack.

ajscilingo, 2020-12-22:
> You can solve this problem by according steps:
> Modify /boot/config.txt file and adding this:
>
> ```
> hdmi_force_hotplug=1
> hdmi_group=2
> config_hdmi_boost=4
> ```
>
> Save it and reboot Raspberry Pi.
>
> ``` [...]

Hi I've tried both these settings as well as switching to the second HDMI port and neither works with my 4K monitor or 4K TV. I also tried extracting the EDID information and passing it via the config.txt, again no signal over HDMI.  For your reference the 4K monitor I'm using is an Acer B286HK Display and the 4K TV I'm using is a Samsung UN65NU8000. I cannot get video out of either using HDMI, however I CAN get video out of the Acer B286HK if I switch to the DVI port and use an HDMI to DVI cable, my TV unfortunately doesn't have a DVI port so I cannot test that.  It would seem to me that something is defective with the HDMI handshake in the case of newer displays and TVs.  The Monitor that works with HDMI using the hdmi_force_hotplug=1 setting is an Acer S231HL display, it's about 8 years old, whereas my other monitor and TV are newer.

sjb5, 2020-12-30:
I am having the same problem.  Neither HDMI ports on the Deskpi Pro work.  I have tried them on both a 1080p monitor and a 1080p TV without success. I have modified the /boot/config.txt file as suggested by yoyojacky.  Still no success.  My RPi 4 miniports all work as expected when not connected to the Deskpi Pro.  I have tried using the full sized ports using both Rasbian and Ubuntu.  I get the same results, a blank screen.  Very frustrating!

ajscilingo, 2020-12-30:
> I am having the same problem. Neither HDMI ports on the Deskpi Pro work. I have tried them on both a 1080p monitor and a 1080p TV without success. I have modified the /boot/config.txt file as [...]

I experienced the same and had to do a lot of tinkering just to get Raspbian to work with one of my monitors using HDMI and the other via DVI, these full-sized HDMI ports are faulty.

akesde, 2020-12-30:
An update to my original issue of not being able to get the full size HDMI ports to work.  After I don't know how many iterations and changes I was able to finally get the full size HDMI port next to the audio jack (NOT next to the power input) able to work but only after I would completely turn off power to the unit when rebooting.  Realizing this was not going to be a working solutions for me, I went ahead and purchased a new Pi4 8Gb to try.  It works on the full size HDMI port next to the power input but not without a lot of trial and error.  I have made several observations along the way.

1.  I had been using a HDMI KVM switch which seems to be very reliable if I do not mount the Pi4 in the Deskpi case and use the micro HDMI ports booting Raspi OS with the modifications to the config.txt file noted in this thread. https://github.com/DeskPi-Team/deskpi/issues/11#issuecomment-749481040. 
2.  If I mount the Pi4 in the Deskpi case, I cannot use the full size HDMI ports unless I plug directly into the monitor's secondary HDMI port.  Nothing wrong with HDMI1 as it is what is attached to my KVM.  This tells me the Pi does not see the HDMI signal although I am using the "hdmi_force_hotplug=1" statement which tells the Pi to always think that there is an HDMI device.  Why it will work without being mounted to the Deskpi board makes me suspect something in the Deskpi board but I do not have conclusive evidence of that.  Fortunately I have two HDMI input ports on my monitor but I realize this is not a common feature.
3.  Raspi OS works better at providing an HDMI signal that Ubuntu 20.10 64bit which is what I am currently running from an SSD drive in the Deskpi case.  My goal was always to run Ubuntu and not Raspi OS since I have over 2Gb of memory.
4.  I had to test the HDMI output to my monitor after installing every case screw.  Put in a screw and reboot, put in another screw and reboot.  The tolerances on this case, board and Pi are such that if I tightened the screws in the wrong order, I was able to disturb the HDMI connection and it would not work.  I regret to say that I did it so many times that I forgot which order ended up working.  I suggest this approach as I had this working many times without all the case screws and end plates installed resulting only to break it with one more screw resulting in several tear downs to start over.  Just take the time and do baby steps.  Sounds like a lot of time but the tear downs took longer.

M-H-010, 2020-12-31:
I have done VERY LITTLE video testing from my Deskpi, as I use it as a headless device .
However it did present the Ubuntu 20.04.1 LTS cli for a clean shutdown from the console without troubles, I can not remember the used port.
Could it be that not every video mode is allowed to pass over the daughterboard ?
I also got an email that they will  replace the board in about 10 weeks.

projektdotnet, 2020-12-31:
I was also having troubles with this and think that some of the issue may be a physical PCB issue. I found that I had to get mine oriented JUST right in order for HDMI to work, if the board was pushed too far in, or not far enough, I'd lose video. If assembled outside the case it was really easy and would work each boot, but once it was assembled and secured inside the case I'd lose HDMI again. After about 5 tries I got the positioning while mounted just right and it continued to function on next boot. Sadly I'm still waiting for a solution to the USB boot needing a dongle removal and reinstall, hoping the March replacement boards fix that.

akesde, 2021-01-01:
I finally gave up and I am using my Pi4 with an external USB 3.0 SSD drive and the microSD HDMI ports are working with my KVM.  The case is a nice paper weight albeit too big for the purpose.  It boots Ubuntu 20.10 and I have attached a new cooling system.  Lesson learned about the Deskpi...

dev-0x7C6, 2021-01-07:
Just for record, I'm also getting no video via hdmi port 0/1. I tried on different monitors (1x dell, 2x samsung tv). One of my older samsung's tv showed output but with additional dialog on top of the screen that cable is damaged. I'm suspecting that connectors or board circuit isn't well designed.

Removing case and using mini-hdmi directly will result with proper output on all my monitors.

wardnuke, 2021-01-14:
So I'm having similiar??? issues. I set up my Deskpi with Kodi (running under the normal Raspberry Pi OS) since LibreElec won't work on these boxes at this time. I plugged it into a previously working HDMI plug on my LG TV (LCD, but NOT 4k) and it recognizes the Pi (it named the HDMI port it was plugged into "HDMI3-Raspberry), but I have no picture. I've tried both rear HDMI ports and did a couple of power cycles too. Any suggestions??

wardnuke, 2021-01-14:
> You can solve this problem by according steps:
> Modify /boot/config.txt file and adding this:
>
> ```
> hdmi_force_hotplug=1
> hdmi_group=2
> config_hdmi_boost=4
> ```
>
> Save it and reboot Raspberry Pi.
>
> ``` [...]

@yoyojacky 

I looked at my config.txt file and I found the three settings listed above, but they were in different places.

I already had hdmi_force_hotplug=1 and it was already uncommented.

Instead of the "hdmi_group=2" value you suggested, I have two lines: "hdmi_group=1" and "hdmi_mode=16"
Should I change this to be ONLY the "hdmi_group=2"

I already had the "config_hdmi_boost=4" command but it was commented out. I uncommented it.

Beduir, 2021-01-14:
You can also check the line "hdmi_enable_4kp60=1" in config.txt
When I add this line to config.txt, my picture disappears too. 

p.s. LibreELEC nightly builds already work with DeskPi.

yoyojacky, 2021-02-20:
> I have done VERY LITTLE video testing from my Deskpi, as I use it as a headless device .
> However it did present the Ubuntu 20.04.1 LTS cli for a clean shutdown from the console without troubles, I [...]

Regarding the video mode, it should be supported, because only the MicroHDMI interface is converted to a standard HDMI interface. No other settings

yoyojacky, 2021-02-20:
> UST right in order for HDMI to work, if the board was pushed too far in, or not far enough, I'd lose video. If assembled outside the case it was really easy and would work each boot, but once it was assembled and secured inside

In fact, because of signal interference, we have implemented shielding measures on the board that we repaired later, and there will be no abnormal HDMI display.

yoyojacky, 2021-02-20:
> I finally gave up and I am using my Pi4 with an external USB 3.0 SSD drive and the microSD HDMI ports are working with my KVM. The case is a nice paper weight albeit too big for the purpose. It boots Ubuntu 20.10 and I have attached a new cooling system. Lesson learned about the Deskpi...

We apologize for everything you have encountered, we will work hard to provide better service, and hope that the new products in March will bring you a different experience.

yoyojacky, 2021-02-20:
> Just for record, I'm also getting no video via hdmi port 0/1. I tried on different monitors (1x dell, 2x samsung tv). One of my older samsung's tv showed output but with additional dialog on top of [...]

 We apologize for everything you have encountered, we will work hard to provide better service, and hope that the new products in March will bring you a different experience. please contact sales@deskpi.com to change the daughterboard.

yoyojacky, 2021-02-20:
> So I'm having similiar??? issues. I set up my Deskpi with Kodi (running under the normal Raspberry Pi OS) since LibreElec won't work on these boxes at this time. I plugged it into a previously [...]

 please contact sales@deskpi.com to change the daughterboard.

yoyojacky, 2021-02-20:
> I already had the "config_hdmi_boost=4" command but it was commented out. I uncommented it.

yes, please try it and do remember reboot your raspberry pi after modifying this file.

yoyojacky, 2021-02-20:
> You can also check the line "hdmi_enable_4kp60=1" in config.txt
> When I add this line to config.txt, my picture disappears too.
> 
> p.s. LibreELEC nightly builds already work with DeskPi.

LibreELEC nightly builds works fine! thanks for your tips, and wish you fabulous day~

closed 2021-02-20 by yoyojacky

opjose, 2021-11-12:
Same issue with Deskpi Pro here with a Samsung TV.

The same Raspberry PI 4 with the same SDCARD installed and the configuration untouched works just fine when not using the DeskPi Pro. Once installed into the DeskPi Pro case & electronics my Samsung TV displays no picture, yet a computer monitor works fine.

The Deskpi may be suppressing the video signal levels or not carrying all of the signals to and from the TV.