DeskPi-Team/deskpi #12: cannot SSD boot until dongle is unplugged abd plugged again

j0nnnnn0, 2020-12-18:
Hi all, love the deskpi but have an issue when trying to boot from SSD 
On boot, SSD is not recognised until I remove the UBS dongle and plug it in again (then boots)
Unsure if this is related to https://github.com/DeskPi-Team/deskpi/issues/1 
Any idea?

[image]

akesde, 2020-12-19:
I am having the same issue and error.  My M.2 SSD is not recognized after a reboot.  I have to unplug the USB bridging connector that goes from the upper USB port on the assembly and the top USB 3.0 port.  If I unplug it and plug it back in, it sees the drive.  My M.2 SSD does have a JMICRON controller and I have applied the quirks command in the cmdline.txt file.

tapd, 2020-12-19:
I'm having the same issue. They need to fix this as I'm sure we all bought this case for the SSD booting & enclosure!

akesde, 2020-12-20:
So I decided to get a different SSD to try for this rig and Amazon Prime delivered (hope you appreciate the pun).  I ordered a $20 Kingston A400 120Gb SSD drive and installed it in the case.  Every time it boots now the drive is recognized and I have not seen the issue again.  For reference I was previously trying an old Mushkin Chronos 240Gb SSD and also tried a KingSpec NT-128 M.2 2242 Drive.  Not the most cost effective solution but it is cheaper than what I paid for the case so that my only consolation.

j0nnnnn0, 2020-12-20:
@akesde that's cool -  did you do anything different or did you just installed a new OS (which one?) on your kingston SSD and it just booted? I'm currently having the issue with a Crucial BX500 SSD.

yoyojacky, 2020-12-22:
hey guys,  I've made a tutorial video here: https://youtu.be/wUHZb9E_WDQ
How to USB boot your Raspberry Pi via DeskPi Pro with M 2 SATA SSD 1080

yoyojacky, 2020-12-22:
> I am having the same issue and error. My M.2 SSD is not recognized after a reboot. I have to unplug the USB bridging connector that goes from the upper USB port on the assembly and the top USB 3.0 [...]

you don't need to add quirks command in the cmdline.txt file.

64Spaces, 2020-12-22:
The issue for me is that although it will boot from USB, it takes a full two minutes, instead of 20 seconds when using the micro SD card. I can tell it’s struggling. 

Also, loading Chromium from the SSD takes sometimes 60 seconds or more the first time instead of 2 seconds.

After installing the quirks, it boots in 20 seconds or less and Chromium opens in 2 seconds.

bpicon, 2020-12-22:
I am in the same case. Booting from usb SSD takes in my case quite a few minutes (I have counted almost 10 minutes in the first boot of twister OS) Once the system is started, everything is very slow, as if we were using a faulty class 4 micro SD.
I also have the problem of loss of wifi when connecting the usb expansion board (I had to move to the side of the router to use ethernet cable and be able to do the configurations)

yoyojacky, 2020-12-23:
> The issue for me is that although it will boot from USB, it takes a full two minutes, instead of 20 seconds when using the micro SD card. I can tell it’s struggling.
>
> Also, loading Chromium from the [...]

just adding quirks will be ok?

yoyojacky, 2020-12-23:
> I am in the same case, in each reboot I have to unplug the usb dongle and plug it back in so that it detects the disk. Booting from usb SSD takes in my case quite a few minutes (I have counted almost 10 minutes in the first boot of twister OS) Once the system is started, everything is very slow, as if we were using a faulty class 4 micro SD.
> I also have the problem of loss of wifi when connecting the usb expansion board (I had to move to the side of the router to use ethernet cable and be able to do the configurations)

Did you try to boot from MicroSD card ?

j0nnnnn0, 2020-12-23:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md

Think I figured it out, looks like a bootloader issue... I followed the steps from URL above URL  in 'Enable USB mass storage boot on a Pi 4B by updating the bootloader EEPROM'... Basically, flashed the bootloader via EEPROM' recovery card... Upon reboot my SSD got recognised on boot...  Yeah 😊

j0nnnnn0, 2020-12-23:
Resolved my issue by following @akesde. Bought a Kingston A400 Data SSD seems to work fine now... 

Though I did also flash the bootloader using the instructions here https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md  in 'Enable USB mass storage boot on a Pi 4B by updating the bootloader EEPROM'... 

Working fine now... Boots in seconds... 

@yoyojacky May I suggest you put a sticky up on your site that instructs to buy SSDs that have been proved to work...

wardnuke, 2020-12-23:
> Resolved my issue by following @akesde. Bought a Kingston A400 SATA SSD seems to work fine now...
> 
> Though I did also flash the bootloader using the instructions here https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md in 'Enable USB mass storage boot on a Pi 4B by updating the bootloader EEPROM'...
> 
> Working fine now... Boots in seconds... No quirks needed
> 
> @yoyojacky May I suggest you put a sticky up on your site that instructs to buy SSDs that have been proven to work...?

My EEPROM is the latest version (Dec 11, 2020) and I'm still getting getting super long boot-up times and Chromium didn't even load after 5 minutes. I don't know what you did @j0nnnnn0 , but it's not repeatable on my system.

j0nnnnn0, 2020-12-23:
Hey @wardnuke - sorry to hear that, I understand it's fustrating...

My issues seemed to be the following:
- incompatible Sata SSD : my Crucial BX500 would start on boot (led flashed once and nothing... had to unplug/plugthe USB dongle -> replaced with Kingston A400 (maybe some kind of difference in the chipset )
- Once I switched the SSD, still didn't work (SSD not loaded on boot), so I flashed my bootloader using the Boot Recovery  and it worked...

- I didn't have the same issue as you with super long boot-up when trying to boot RaspOS directly from SSD... I did have super long boot time initially when trying to boot from BerryBoot (on SD) to RaspOS 32 (on my Crucial SSD)...

spiff72, 2020-12-24:
Same issue here - I have tried with a Samsung 860 EVO 500GB drive and a PNY 120GB SSD - both won't boot.  I haven't tried the unplugging of the dongle when it is stuck on that boot screen, but I have tried unplugging the dongle after it was booted from the SD card (while the dongle was connected - and the SSD is not mounted), and then plugged the dongle back in, and this would allow the SSD to show up and mount.

wardnuke, 2020-12-24:
I had different flashing lights when trying to boot with the SSD at first.
That coupled with VERY LONG boot times and don't even ask about Chromium.
After using the USB quirks command, the boot times now are the same as for
my MicroSD card boot time, which is about 30 seconds from power on to a
fully loaded desktop. As you can see from my post the SSD throughput speeds
are only about 30% of what the speed before USB quirks command applied, but
the boot time is outstanding. I'm working on making one of mine, I bought
2, into a Kodi Media Center. Right now I'm running a Raspberry Pi 4B 8GB
with the normal Raspbian Desktop and have Kodi 18.7 installed. I've
streamed some movies with no issues. My unit is overclocked to 2Ghz and the
GPU is set to 750Mhz. So far it's a pretty stable system, but I want to run
the latest LibreElec. I'm not sure if I can get that to work with the
DeskPi utilities, but I'm going to see. I've got one on a MicroSD card
already set up so I'll see if it works.

Good news is that DeskPi Sales support contacted me and said they were
actively working on new PCBs that have a different controller chip. They
were testing, would send new PCB's to anyone with the defective controller,
and the expected ship date is March 2021.

wardnuke, 2020-12-24:
Make sure the SSD isn't a NVMe drive. They aren't supported by this current
chipset. DeskPi Sales Support email me and said they were testing a new PCB
with a different controller chip and if it works would ship new PCBs to
anyone who has one with the old controller. Expected shipping date of March
2021. Hopefully the new chipset will support NVMe SSD's as they have wicked
speeds. My main drives on both my gaming desktop and gaming laptop are 2TB
Samsung EVO Pro SSDs. They rock.

Luck.

spiff72, 2020-12-24:
> @spiff72 Might be an issue with the Samsung controller chip on the SSD. I'm using a WD Blue M.2 2280. The description on Newegg where I got it doesn't mention a controller chip, but the specs for the Samsung 860 EVO mentions a Samsung MJX Controller. Perhaps this is causing the conflict?? On a lighter note, DeskPi Sales Support email me and said they were testing a new PCB with a different controller chip and if it works would ship new PCBs to anyone who has one with the old controller. Expected shipping date of March 2021. Hopefully the new chipset will support NVMe SSD's as they have wicked speeds. My main drives on both my gaming desktop and gaming laptop are 2TB Samsung EVO Pro SSDs. They rock. Luck.

I am currently using the Samsung EVO 860 on my Pi as a boot drive - I put the pi back into my previous case (an Argon Neo).  No issues at all with my current USB3-SATA adapter and that Samsung drive (or the PNY drive for that matter).  The USB3-SATA drive adapter I am using uses the ASMedia chipset, which I bought specifically for its good compatibility with the RasPi4.  The original USB3-SATA adapter was one that didn't work well (I bought it prior to knowing that some had these compatibility issues).  Pretty sure it also uses the JMicron chipset.

wardnuke, 2020-12-24:
@spiff72 

Yeah, it'd have been nice if they'd done a little research and got a controller chip that for sure compatible with the RasPi4s. As it is, I can't wait for this to spur on some other companies to come up with cases of their own. I've been fiddling with the Pi's for a couple of years now and I've just gotten back into it again after a lull. I have one running my 3D Printer running OctoPi, one set up for my ham radio, one set up with LibreElec, and a few I have to tinker with. I hope they get a LibreElec script out soon cause I bought these to run as Media Centers.

bobaxxx, 2020-12-26:
Hi guys,

The september eeprom is much more effective instead of the last of december

I cant even boot with one from december from my ssd (i tried with a jmicron ssd 250g, and then with a m2 wd blue 250g) i put back the eeprom from september, and ssd boot up, even its slow...and 50% time, it got some loading stuck (wpa supplicant, create volatile file..., im using libreelec

bpicon, 2020-12-28:
I provide the solution that has worked for me. Once again, the Jmicron chip has some compatibility problem, I can't wait for the new board to be shipped to us in March, as promised:

.- Modify the bootloader configuration. From a terminal run "sudo -E rpi-eeprom-config --edit"
.- Add the line: USB_MSD_PWR_OFF_TIME = 0
.- Save.
.- Reboot.

This is my bootloader configuration (installed Dec 11 2020 version): 

pi @ raspberrypi: ~ $ vcgencmd bootloader_config
[all]
BOOT_UART = 0
WAKE_ON_GPIO = 1
POWER_OFF_ON_HALT = 0
DHCP_TIMEOUT = 45000
DHCP_REQ_TIMEOUT = 4000
TFTP_FILE_TIMEOUT = 30000
ENABLE_SELF_UPDATE = 1
DISABLE_HDMI = 0
USB_MSD_PWR_OFF_TIME = 0
BOOT_ORDER = 0xf41

I still have a serious problem with the 2.4 Ghz wifi (I don't have 5 Ghz) when the usb 3.0 board is connected. I am afraid that poor isolation causes interference on the 2.4Ghz band. I have already tried on all channels and there is no way. If I unplug the dongle, it connects smoothly.

yoyojacky, 2020-12-29:
> 通过关注@akesde解决了我的问题。买了金士顿A400 SATA SSD似乎现在可以正常工作了...
> 
> 尽管我也确实按照此处的说明刷新了引导加载程序，[网址为https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md) ，“通过更新引导加载程序EEPROM在Pi 4B上启用USB大容量存储引导”。 ..
> 
> 现在可以正常工作...只需几秒钟即可启动...无需任何怪癖
> 
> @yoyojacky我是否可以建议您在您的网站上贴上一个标签，指示购买已被证明可以工作的SSD ...？

thanks for your help, At present, our team is already making the second version of deskpi pro, and we will replace the chip more suitable for USB booting. We will also provide the service of replacing the motherboard for the first batch of users.
If you have this need, you can also contact our sales. It will be distributed around the end of March.

yoyojacky, 2020-12-29:
> > 通过关注@akesde解决了我的问题。买了金士顿A400 SATA SSD似乎现在可以正常工作了...
> > 尽管我也确实按照此处的说明刷新了引导加载程序，[网址为https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md)，“通过更新引导加载程序EEPROM在Pi 4B上启用USB大容量存储引导”。 ..
> > 现在可以正常工作...只需几秒钟即可启动...无需任何怪癖
> > @yoyojacky我是否可以建议您在您的网站上贴上一个标签，指示购买已被证明可以工作的SSD ...？
> 
> 我的EEPROM是最新版本（2020年12月11日），我仍然获得超长的启动时间，并且5分钟后甚至没有加载Chromium。我不知道你在做什么@ j0nnnnn0，但是在我的系统上它是不可重复的。

you can contact our sales,At present, our team is already making the second version of deskpi pro, and we will replace the chip more suitable for USB booting. We will also provide the service of replacing the motherboard for the first batch of users.
If you have this need, you can also contact our sales. It will be distributed around the end of March.

yoyojacky, 2020-12-29:
> 嘿@wardnuke-很抱歉听到这个消息，我知道这很麻烦...
> 
> 我的问题似乎如下：
> 
> * 不兼容的Sata SSD：我的Crucial BX500无法在启动时启动（指示灯闪烁一次，什么也没...）必须拔下/插入USB软件狗->进行补救，我跟随@akesde并用金士顿A400替换了我的SSD（也许是某种芯片组的差异）
> * 切换到金士顿固态硬盘后，它仍然不起作用（引导时未加载SSD），因此我使用Boot Recovery刷新了引导加载程序，并且它起作用了……（我也尝试过回到Crucial BX500，但是无法启动）
> * 尝试直接从SSD（在两个SSD上）直接引导RaspOS时，我没有与您一样长的启动问题。但是，当我最初尝试从BerryBoot引导（在SD）到RaspOS 32（在我的Crucial SSD上）...
> 
> 如果有帮助，这是我的RPI的一些输出，我的eeprom是9月份的（我想它是Recovery中的版本）……所以也许也值得您尝试一下）
> 
> ```
> 
> pi@raspberrypi:~ $ lsusb -t
> /:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=dwc2/1p, 480M
>     |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
>         |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=ch341, 12M
> /:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
>     |__ Port 1: Dev 2, If 0, Class=Mass Storage, Driver=uas, 5000M
> /:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
>     |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
> 
> 
> pi@raspberrypi:~ $ lsblk
> NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
> sda      8:0    0 223.6G  0 disk
> ├─sda1   8:1    0   256M  0 part /boot
> └─sda2   8:2    0 223.3G  0 part /
> 
> pi@raspberrypi:~ $ vcgencmd bootloader_config
> [all]
> BOOT_UART=0
> WAKE_ON_GPIO=1
> POWER_OFF_ON_HALT=0
> DHCP_TIMEOUT=45000
> DHCP_REQ_TIMEOUT=4000
> TFTP_FILE_TIMEOUT=30000
> ENABLE_SELF_UPDATE=1
> DISABLE_HDMI=0
> BOOT_ORDER=0xf41
> 
> pi@raspberrypi:~ $ less /boot/cmdline.txt
> console=serial0,115200 console=tty1 root=PARTUUID=41d00afb-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
> /boot/cmdline.txt (END)
> 
> pi@raspberrypi:~ $ rpi-eeprom-update
> BCM2711 detected
> Dedicated VL805 EEPROM detected
> BOOTLOADER: up-to-date
> CURRENT: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)
>  LATEST: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)
>  FW DIR: /lib/firmware/raspberrypi/bootloader/critical
> VL805: version unknown. Try sudo rpi-eeprom-update
> CURRENT:
>  LATEST: 000138a1
> 
> ---
> Raspberry Pi Diagnostics - version 0.6
> Thu Dec 24 12:40:00 2020
> 
> Test : SD Card Speed Test
> Run 1
> prepare-file;0;0;274209;535
> seq-write;0;0;275361;537
> rand-4k-write;0;0;53498;13374
> rand-4k-read;51120;12780;0;0
> Sequential write speed 275361 KB/sec (target 10000) - PASS
> Random write speed 13374 IOPS (target 500) - PASS
> Random read speed 12780 IOPS (target 1500) - PASS
> Test PASS
> ```

Thank you so much, your news is very useful to us, thank you very much for your support! !

yoyojacky, 2020-12-29:
> USB_MSD_PWR_OFF_TIME = 0

Yes, we also found that the frequency band interference is caused by poor isolation. The Raspberry Pi itself uses the same GND for USB2.0 and USB3.0. We have improved this problem in the new solution to avoid interference 2.4GHz The condition of the equipment, thank you very much for the useful information, this is very important to us! Thank you for your trust and support!

bobaxxx, 2020-12-29:
> thanks for your help, At present, our team is already making the second version of deskpi pro, and we will replace the chip more suitable for USB booting. We will also provide the service of [...]

I already have contacted by mail, but no response from the store

Fortress-Labs, 2020-12-29:
I suppose it is good that they will be replacing them, but it's quite unfortunate how long it is going to take to get it all sorted out. Looks like I will need to purchase another case and sit this one on a shelf for a while.  I've been having the USB boot issue, wifi issue, and front panel USB ports not working correctly.

Pi-Mania, 2020-12-30:
M.2 Micron fails to Boot, (nothing ever displayed on screen) Samsung EVO 750 SSD Fails, (Tries but multiple errors seen) Old 500gb Hitachi 5K750-500 SATA,  Boots Just Fine.

bobaxxx, 2020-12-31:
I received an email just straight after i sent my 3r mail to the store (finally) :

Hello
Deeply sorry for the inconvenience caused to you，
We will re-send you a new version PCB board，（except the smaller Adapter Board）
At present，we are testing it，
When we finish it，we will send it to you
（Considering R&D and mass production, the holiday on the Chinese Spring Festival,it estimated to ship it for you at the middle of march 2021）
Best Regards！

This is still NOT normal to get so much problem for a case that cost so much :

- Front usb not working (or not properly)
- USB dongle fail to use correctly the internal SSD/HDD
- WIFI disturbance
- Fan control for some OS is not operationnal (Libreelec in that case)

we dont have to do so much tinker with the raspberry pi OS...this is not the expected behaviour

vinodsmenon, 2021-01-01:
Hi There, I'm one of the owners of the DeskPi Pro with a M.2 Sata adapter. What I have found is if I used the Raspberry Pi with USB boot it seems to work. The product is incomplete with bugs and issues. 

I can confirm the issues that are prevalent are as follows:

1. Front USB not working (or not properly)
2. USB dongle fail to use correctly the internal SSD/HDD
3. WIFI disturbance
4. Loading time takes too long.
5. Fan speed control should be better implemented for easy modification on OS, example on Ubuntu 20.10

Who can we contact with regards to our orders and ensuring that a functional PCB board is sent to the owners of the DeskPi Pro?
I'd appreciate your honest answer and look forward to a working unit minus the bugs and hassles.

futurejones, 2021-01-02:
+100 We need clear and precise details on who to contact to ensure we get the replacement PCB board.

Edo78, 2021-01-05:
I wrote an email and got no answer so I open an issue on paypal about a faulty product so I can block my payment if they don't solve my issue. I suggest you to do the same because end of march is about the end of the paypal protection (at least for who place the preorder like me).

john-clark, 2021-01-06:
I also have this problem and received this email.

andrewjameshough, 2021-01-27:
@yoyojacky have mailed through the support link on site about having the same issues in regards to m.2 SSD (slow and erratic booting) but have yet to receive a reply. As mentioned above it would be good if we had a process to follow for all who have this manufacturing issue that will reportedly be fixed in March. Can you help with this? 

Many thanks.

yoyojacky, 2021-02-02:
> I received an email just straight after i sent my 3r mail to the store (finally) :
>
> Hello
> Deeply sorry for the inconvenience caused to you，
> We will re-send you a new version PCB board，（except the [...]

Could you please post the /boot/config.txt file and what kind of OS did you use?

yoyojacky, 2021-02-02:
> Hi There, I'm one of the owners of the DeskPi Pro with a M.2 Sata adapter. What I have found is if I used the Raspberry Pi with USB boot it seems to work. The product is incomplete with bugs and [...]

> > > 通过关注@akesde解决了我的问题。买了金士顿A400 SATA SSD似乎现在可以正常工作了...
> > > 尽管我也确实按照此处的说明刷新了引导加载程序，[网址为https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md) ，“通过更新引导加载程序EEPROM在Pi 4B上启用USB大容量存储引导”。 ..
> > > 现在可以正常工作...只需几秒钟即可启动...无需任何怪癖
> > > @yoyojacky我是否可以建议您在您的网站上贴上一个标签，指示购买已被证明可以工作的SSD ...？
> > 
> > 
> > thanks for your help, At present, our team is already making the second version of deskpi pro, and we will replace the chip more suitable for USB booting. We will also provide the service of replacing the motherboard for the first batch of users.
> > If you have this need, you can also contact our sales. It will be distributed around the end of March.
> 
> I already have contacted by mail, but no response from the store. This is not clear, i bought deskpi pro from deskpi.com store.
> I cant contact them, chat doesnt works, they doesnt answer when we use the mail contact page.
> 
> In my paypal account, the contact mail is from the 52pi.com, and they doesnt sell deskpi from their webpage.
> 
> So what now ?

Could you please send mail to:  sales@deskpi.com our salemen will contact you immediately.

yoyojacky, 2021-02-02:
> Can you help with this?
> 
> Many thanks.

This problematic SSD conversion board will be replaced in an orderly manner in March. Please be patient. We have recorded your purchase order information here and will be replaced in order according to the order.

yoyojacky, 2021-02-02:
> I suggest you to do the same because end of march is about the end of the paypal prot

Please rest assured that we are not so shameless. Although our products have defects, we will replace them. Why is it that we will not go to work for a long time during the Chinese New Year, so the supplier will be during this time. All are on vacation. We can’t produce and test. This is hardware production. You know it will consume some time. From PCB revision to component procurement, soldering, testing, assembly, packaging and delivery, it will take time, please believe Our team will not damage our own business integrity because of this little money.

yoyojacky, 2021-02-02:
> +100 We need clear and precise details on who to contact to ensure we get the replacement PCB board.

 please contact to sales@deskpi.com, and leave your prochase number. we are going to replace the board step by step.

yoyojacky, 2021-02-02:
> Who can we contact with regards to our orders and ensuring that a functional PCB board is sent to the owners of the DeskPi Pro?
> I'd appreciate your honest answer and look forward to a working unit minus the bugs and hassl

Please rest assured that we are not so shameless. Although our products have defects, we will replace them. Why is it that we will not go to work for a long time during the Chinese New Year, so the supplier will be during this time. All are on vacation. We can’t produce and test. This is hardware production. You know it will consume some time. From PCB revision to component procurement, soldering, testing, assembly, packaging and delivery, it will take time, please believe Our team will not damage our own business integrity because of this little money.  BTW，　
If the front USB does not work, please check /boot/config.txt
In the dtoverlay part of the configuration, whether the configuration of dwc2 is added. For specific content, please refer to the readme section of github.

Edo78, 2021-02-02:
> Please rest assured that we are not so shameless. Although our products have defects, we will replace them. Why is it that we will not go to work for a long time during the Chinese New Year, so the [...]

I perfectly understand your problem and I hope you understand mine. I have already payed for a defective product. What if the replace is also defective? What if you go bankrupt while testing? I want to be protected from those issues so I contacted paypal.
I really hope we can find a solution but I'm not willing to bet my money. If you wish you can refound me and I promise to rebuy the next version at the same price. Please rest assured that I'm not so shameless

yoyojacky, 2021-02-20:
> Resolved my issue by following @akesde. Bought a Kingston A400 SATA SSD seems to work fine now...
>
> Though I did also flash the bootloader using the instructions here [...]

That would be great!

john-clark, 2021-02-20:
Still waiting for replacement board to fix the no usb on boot problem.

yoyojacky, 2021-02-20:
> Still waiting for replacement board to fix the no usb on boot problem.

It will be comming soon... you have my word.

projektdotnet, 2021-02-20:
Just wanted to add on that I am also having this issue with a Kingston SA400S37/120G

schusz, 2021-03-08:
I also have this issue with a Western Digital Blue 3D M.2 250 GB SSD.

john-clark, 2021-03-28:
Would like an update. The new version PCB board was supposed to be shipped the middle of March 2021.  I have not received an email stating it has shipped.

bobaxxx, 2021-03-28:
> > Still waiting for replacement board to fix the no usb on boot problem.
> 
> It will be comming soon... you have my word.

At least, sens mail to customers that are waiting for the new board

Mister-42, 2021-03-28:
> > Still waiting for replacement board to fix the no usb on boot problem.
> 
> It will be comming soon... you have my word.

Please don't forget to send me the whole kit, not just the replacement board. I stopped shipping it out until fixed.

vinodsmenon, 2021-04-02:
Hi I still haven't heard of when the new board is shipping. It's the first
week of April now, but no updates!

Please revert as this product is unusable right now.

Regards,
Vinod Menon

john-clark, 2021-05-11:
Thank you for the update.

> Hello,
>
> The new updated PCB Board for your DeskPi order was shipped on April 30th.
> The tracking number is _removed_, you can track here later:
> https://www.trackdog.com/?lang=en
> 
> The new updated PCB Boards fixes following problems:
> 1. The problem with USB booting --- identifying SSDs
> 2. WiFi interference issues
> 3. HDMI display problem
> 4. Compatibility issues of Wireless key & mouse
> 
> Resend List:
> 1. USB adapter
> 2. Daughter Board
> 3. T-shaped isolation foam
> 4. Gift: U-shaped bracket
> 
> Have a nice day!
> DeskPi Team

yoyojacky, 2021-05-11:
if your SSD boot still have this problem, please contact me here, and please take a snapshot of the problem, if you can provide the steps that you have already do will be better..

yoyojacky, 2021-05-12:
> me and

yes, I can understand your concern and we have already reshipped all replacement to customers one by one.

yoyojacky, 2021-05-12:
> I received the replacement. Thanks very much! Have not replaced the board yet. Will let you know when I have.

thanks for your reply and hope you have a nice day~ if you have any trouble please let me know~

akesde, 2021-05-12:
Do all customers who bought the first round of devices get the replacements or only if you requested a replacement?  I never found a way to request a replacement.

j0nnnnn0, 2021-05-19:
@Akesde As I didn't buy directly from DeskPi, I had to request the replacement via my local supplier here in Switzerland - apparently I should be receiving in the next few days 
--
All - I propose to close this issue as the solution is to request the replacement board

closed 2021-05-19 by j0nnnnn0

teopost, 2023-08-29:
I have the exact same issue, and I have contacted sales@deskpi.com.
I haven't received any response!