DeskPi-Team/deskpi #13: readme.md missing instructions for required usb quirks

WZ9V, 2020-12-18:
Because you chose to use the JMicron chipset that has issues running in UAS mode (the default) users need to add usb-storage.quirks=152d:0562:u to cmdline.txt in order for it to work properly.  Without it the SATA drive will not boot. 

Please update the readme.md with this step so that others can be spared wasting a ton of time troubleshooting a known issue.  In my case it cost me an entire Sunday.

ajscilingo, 2020-12-19:
Mine works without this quirk, my video however - see the video thread.  It looks like they don't have the exact same components for everyone.

grog-polymer, 2020-12-19:
I tried the workaround with a samsung 500GB drive. did not work. It is running usingh an external USB ->SATA cable with a  ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge. Spent a day stuffing around with it.

WZ9V, 2020-12-19:
Mine is using the JMircon chipset and would not boot without it.  It would read/write but I have my doubts about it's long term stability without USB quirks.  I'm using an Inland Professional 256GB 2.5in SATA drive.

@grog-polymer  - like you an ASMedia chipset adapter plugged into the USB3 port works perfectly without any changes.  It would have been so nice if they had done their homework and used that chipset instead.  It's not like there wasn't a ton on info on the RaspberryPi forums documenting JMicron issues.

yoyojacky, 2020-12-22:
> Because you chose to use the JMicron chipset that has issues running in UAS mode (the default) users need to add usb-storage.quirks=152d:0562:u to cmdline.txt in order for it to work properly. [...]

In fact, it can run without adding

yoyojacky, 2020-12-22:
> I tried the workaround with a samsung 500GB drive. did not work. It is running usingh an external USB ->SATA cable with a ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge. Spent a day stuffing around with it.

we will follow up this case.

closed 2020-12-22 by yoyojacky

WZ9V, 2020-12-22:
@yoyojacky - I have two of your cases where it will not despite me following the instructions in the readme and the video to the letter and wasting days troubleshooting.  The 2.5in SATA drive I'm using works perfectly with an ASMedia chipset cable so it is not the SSD.  Very disappointed that you are closing out these issues without offering a solution.

Edo78, 2021-01-15:
> In fact, it can run without adding

Mine can't. I started looking for a solution just because it didn't run. Even if I boot from an sd the speed of the ssd speed result in few kb/s. Once added the quirk the speed jump to almost 200 MB/s

This is why I opened an issue on paypal.