DeskPi-Team/deskpi #14: Fan does not start at boot

grog-polymer, 2020-12-19:
Linux raspberrypi 5.10.1-v7l+ #1383 SMP Fri Dec 18 19:05:58 GMT 2020 armv7l GNU/Linux

I have to run deskpi-config to get it to start each time I boot.

tapd, 2020-12-19:
Same here, it also stops randomly when set to option 7, auto. I basically set it to option 1 at 25% and leave it, that seems to leave it running.

yoyojacky, 2020-12-22:
> Linux raspberrypi 5.10.1-v7l+ #1383 SMP Fri Dec 18 19:05:58 GMT 2020 armv7l GNU/Linux
> 
> I have to run deskpi-config to get it to start each time I boot.
Because by default, it will rotate by detecting the CPU temperature and reaching the threshold you set. If the temperature is lower than the threshold, it will not rotate.

yoyojacky, 2020-12-22:
> Same here, it also stops randomly when set to option 7, auto. I basically set it to option 1 at 25% and leave it, that seems to leave it running.

Because by default, it will rotate by detecting the CPU temperature and reaching the threshold you set. If the temperature is lower than the threshold, it will not rotate.You can set an appropriate threshold through deskpi-config to control the fan's speed. When your CPU temperature cannot reach the threshold, it won't spin crazy. This will be smarter.

closed 2020-12-22 by yoyojacky