DeskPi-Team/deskpi #16: ISSUES WITH USB BOOT TO M.2 SSD

wardnuke, 2020-12-23:
Just received my DeskPi Pro today. Put everything together along with a WD Blue M.2 SSD. Installing went ok, but there's a learning curve.

My issue was when it came to prepare the SSD for USB Boot. I followed the directions on the GetHub Readme file. My issue came when I perfomed the changes under raspi-config. Everything seemed to go as expected, but when I did the "sudo -E rpi-eeprom-config --edit command things didn't go as expected. The boot config file pulled up, but it only had a couple of lines. It only had the BOOT_UART=0, WAKE_ON-GPIO=1, and POWER_OFF_ON_HALT=0 lines. I tried several times, but nothing "pulled" into the bootloader file about BOOT_ORDER=0fx41. When I performed all the rest of the steps and rebooted without the MicroSD card, I got the Raspberry Pi splash screen and then a blank screen with the white light in the DeskPi Pro box flashing constantly. After some digging, I watched the video called "How to USB boot you Raspberry Pi via DeskPi Pro with M 2 SATA SSD 1080p" made by Kevin Deng. At around 37:20 he used vcgencmd bootloader_config to check the BOOT_ORDER and that's where I figured out the steps in the Readme didn't work. I manually  entered the information in my bootloader file and it worked ONCE, although it's VERY SLOW on boot-up.

I did a speed test of my SSD and got the following numbers:

Raspberry Pi Diagnostics - version 0.6
Wed Dec 23 14:21:05 2020

Test : SD Card Speed Test
Run 1
prepare-file;0;0;280068;547
seq-write;0;0;259035;505
rand-4k-write;0;0;74642;18660
rand-4k-read;73061;18265;0;0
Sequential write speed 259035 KB/sec (target 10000) - PASS
Random write speed 18660 IOPS (target 500) - PASS
Random read speed 18265 IOPS (target 1500) - PASS
Test PASS

It takes almost 3 minutes from first power applied to get to the home screen (no icons) and another 1-2 minutes for the menu icons to show up. Trying to launch Chromium took another 1-2 minutes.

There needs to be a fix for this.

wardnuke, 2020-12-23:
I inserted the quirks command and the boot-up is about the same for my MicroSD card. Here are the new stats for the thd M.2 SSD with quirks activated:

Raspberry Pi Diagnostics - version 0.6
Wed Dec 23 15:23:14 2020

Test : SD Card Speed Test
Run 1
prepare-file;0;0;184089;359
seq-write;0;0;179550;350
rand-4k-write;0;0;22505;5626
rand-4k-read;18729;4682;0;0
Sequential write speed 179550 KB/sec (target 10000) - PASS
Random write speed 5626 IOPS (target 500) - PASS
Random read speed 4682 IOPS (target 1500) - PASS
Test PASS
 
As you can see the Random Write Speed is down to 30% of the pre-quirks speed and the Random Read Speed is down to about 25% of the pre-quirks speed. The Sequential Write Speed didn't suffer too much, but is down to about 70%.

In comparission, here's my MicroSD card:

Raspberry Pi Diagnostics - version 0.6
Wed Dec 23 15:34:27 2020

Test : SD Card Speed Test
Run 1
prepare-file;0;0;35890;70
seq-write;0;0;36308;70
rand-4k-write;0;0;4437;1109
rand-4k-read;10768;2692;0;0
Sequential write speed 36308 KB/sec (target 10000) - PASS
Random write speed 1109 IOPS (target 500) - PASS
Random read speed 2692 IOPS (target 1500) - PASS
Test PASS

This is bearable and better than what I'm getting on my MicroSD card which is a SanDisk Extreme A2, V30 chip, but isn't what most people expect of this device. The whole purpose of getting the DeskPi Pro was to take advantage of a good cooling option along with internal SSD support, including M.2 SSD's, and put this in a snazzy case. They need to fix this and send those of us that purchased the ones early with the incompatable controller chip with a newer one that allows us to take full advantage of the SSD's speed..

WZ9V, 2020-12-24:
Good luck.  I expect they will close this one out just like the other two pointing out the same thing.  The SSD interface is defective.

wardnuke, 2020-12-24:
I got a reply from DeskPi Sales that stated that they were actively
working on the problem and were currently testing new PCB's. They said they
were going to send out updated PCB's to people that had the defective
controller chip'd one and that the estimated ship date is March 2021. Until
then, as you can see I got my units to boot at about the same speed as a
MicroSD card boot, 30 seconds, and the transfer rates to the SSD, while not
optimal, are still better than a MicroSD card. I am having issues with my
back USB 2.0 ports not working though. I'll have to open a new trouble
ticket on that one.

wardnuke, 2020-12-24:
> Good luck. I expect they will close this one out just like the other two pointing out the same thing. The SSD interface is defective.
 
@WZ9V : 

I got a reply from DeskPi Sales that stated that they were actively  working on the problem and were currently testing new PCB's. They said they were going to send out updated PCB's to people that had the defective controller chip'd one and that the estimated ship date is March 2021. Until then, as you can see I got my units to boot at about the same speed as a MicroSD card boot, 30 seconds, and the transfer rates to the SSD, while not optimal, are still better than a MicroSD card. I am having issues with my back USB 2.0 ports not working though. I'll have to open a new trouble ticket on that one.
--

WZ9V, 2020-12-24:
Good to hear, would have been nice if they had said that when they closed the other issues.

wardnuke, 2020-12-24:
@ WZ9V

I get the impression that they are dealing with a lot of "feedback" and are afraid if they don't do something they'll lose a lot of customers if they don't fix it. Probably want to get it fixed quietly. I understand a little as the ones I ordered in Nov were the Pre-release and obviously there'd be bugs. I actually think with the SSD speed issue aside, these things are pretty good. I've been tinkering with one of mine for about 24 hours and besides the SSD speed and the fact my USB doggle for my mouse / keyboard won't work on any USB port on the back, I haven't had any issues. I do hope they work on the suggestoin about getting something set up for LibreElec. I'm running mine with Raspbain / Kodi 18.7, but I have a MicroSD set up with the latest LibreElec and it is awesome. Hopefully they come up with a script to set up a Pi with LibreElec and the DeskPi config. I'm going to try my LibreElec MicroSD card in the DeskPi to see how it reacts. Wish me l luck.

wardnuke, 2020-12-24:
@WZ9V

Well tried my LibreElec MicroSD card. Boots up ok on the DeskPi, but no USB support therefore no mouse / keyboard. Hopefully the DeskPi folks come up with a script that loads LibreElec with DeskPi scripts for support cause this thing would be an awesome LibreElec Media Center, especially with a large SSD for downloads / media.

WZ9V, 2020-12-24:
So far I've only tried mine headless and with USB3 connections.  Now I'm thinking I need to test all the USB2 ports

bobaxxx, 2020-12-26:
Dont update to the last eeprom (11/12/2020), because it does not detect at all internal sata port....the one working (partially...) is the eeprom from september. But even with that, you need to disconnect the dongle / reconnect the dongle at boot, if not, nothing is detected.

And even with that, libreelec could stuck at boot when he is loading (wpa supplicant with hdmi interference, fail create volatile file, etc etc....). I reach a successfull boot 50 % of the time.

I installed libreelec with a simple usb key just to try, and i dont see those problem, so, the real problem comes from the usb dongle and the way the system loads the internal hdd/ssd. Its got real problem to power up the internal pins.

I think its the last time i buy some "chinese" case, last time, i bought one from geekpi (X735 bundle), and the motherboard simply crash out after about 1 month with no reason, crashing my HDD too.... Nobodies replies to my mail when i say that to them, no refund, no new motherboard....

WZ9V, 2020-12-26:
I'm still waiting on the M.2 SATA drive I ordered from Amazon but I can confirm the Argon One M.2 case uses the ASMedia chipset.  I do like the cooling solution on the DeskPi Pro better though so I do hope they fix it.

bobaxxx, 2020-12-26:
i bought this one too, i hope this will works (according to other threads, it will), otherwise, its a totally pointless investments if we have a dedicated unused sata port, with no possibility of managing the fan / ir, and with front usb no working

wardnuke, 2020-12-26:
@bobaxxx 

The Dec 2020 EEPROM works fine for me although the SSD performance is about 30% of what it was before I activated USB quirks. The unit boots up just as fast as my MicroSD card boot and if you look at the speed tests, the SSD performs random read/writes faster than the MicroSD card so even operating at 30% of its abilities, it's still hands down better than the MicroSD card. I think that the DeskPi folks will get their stuff together as it's in their best interests. If they don't, some other enterprising individual will swoop in and put out a better unit, taking their business.

Pi-Mania, 2020-12-31:
My M.2 Micron drive currently just fails to Boot.  My Samsung EVO 750 SSD WORKS BUT, first boot of FENIXOS took about 15-20mins. I then applied quirks as listed here: (hxxps://jamesachambers.com/raspberry-pi-4-usb-boot-config-guide-for-ssd-flash-drives/ after a restart, its now using USB-STORAGE at least and now boots in a few seconds. Tried an Old 500gb Hitachi 5K750-500 SATA, That Boots up Just Fine. (still faster than SDCARD) Hope that helps somebody get up & running at least.

nitz, 2020-12-31:
I received an email from DeskPi on December 29th. I haven't contacted them directly before, so if you're someone who's bought one and didn't get this email from them, you might want to get in touch with them to make sure you get on this list. It seems they've re-done the PCB and plan to send out a replacement in Mid March. Here's the email in it's entirity:

```
Hello：

 

Everyone, we do apologize for not bringing a good user experience for the first batch of DeskPi Pro. For the quicker shipment of the pre-sale orders, we ignored your needs on the Deskpi function.

 

To completely solve the problem, we will re-send the re-modified PCB board for all the customers who received the DekPi Pro product. Our sales staff will contact you according to the order time. 

 

For some not shipped orders, we will contact you and adopt your ideas.

 

Considering R&D and mass production, the holiday on the Chinese Spring Festival, the new version PCB board will be shipped to you at the middle of March 2021. 

 

And we will strengthen management in design and mass production to avoid such problems. Thanks again for your trust, especially the first batch of customers, you are very important to us! 

 

In the later R&D, we are looking forward to your suggestion and supervision.  

we can produce more creative and higher quality Raspberry Pi products and provide better Customer services!
```

They don't specifically state what is being fixed, but there's a good chance it not only relates to this ticket, but may apply to #12, #11, #18 (@yoyojacky says that's fixed in the upcoming revision), #8, and maybe #1.

breitpet, 2021-01-14:
hi to all, I am also a owner of the DeskPi Pro enclosure and I have the same problem to use my SSD (e.g. **Samsung 860 EVO**, 250 GB) correctly with the delivered usb dongle. By typing the command "lsblk" there is no sda device visible! And by using the command "sudo rpi-eeprom-config" it displays the following: 
[all]
BOOT_UART=0
WAKE_ON-GPIO=1
POWER_OFF_ON_HALT=0
[all]
BOOT_ORDER=0fx41

I will try next my **SanDisk Extreme PRO M.2 NVMe 3D SSD, 500 GB** - i hope that works :-)

#stayallsafe

wardnuke, 2021-01-19:
@breitpet 

I'm pretty sure there's a note somewhere in the support documentation that the DeskPi Pro does NOT support NVMe SSD boards. It might with the upgraded controllers they're releasing in March to fix the speed issue. Might want to check. Is that Samsung 860 EVO a NVMe SSD?? If it is, then that might be your problem. If you look at the beginning of this tread, I think I lised the drives I use. Western Digital M.2 2880 SSD's worked great for me. I had no problems getting mine to work, besides the speed issue where I had to change some settings.

closed 2021-02-20 by yoyojacky

inspector71, 2021-06-19:
> 
> 
> I inserted the quirks command 

???

What is "the quirks command".

I have a DeskPi Pro they said needing replacing and have shipped the replacement, albeit it arrived but I have not picked it up yet.

Did not try the SSD or SATA via USB dongle option and unit was running LibreELEC well enough except of course, occasionally spontaneously rebooting if Kodi choked on a big video file. 

Last night thought I'd try an SSD after it appears several spontaneous Kodi choke reboots have fried the DeskPi Pro-branded MicroSD card.

Thankfully I found this issue whilst just trying to get the USB (LibreELEC USB stick to install to the M.2 SSD ... does that even work?). Would not have wanted to waste my time with a dodgy SSD interface.

Could you paste a direct link to "the GitHub README" ? Have only seen two repositories without READMEs to speak of. Unless you are referring to this:

https://github.com/DeskPi-Team/deskpi/blob/master/README.md

...  but those instructions seem minimal. They were helpful to you?

Whereas this is just empty:

https://github.com/DeskPi-Team/deskpi-team.github.io/blob/main/README.md