DeskPi-Team/deskpi #17: Please add an installer for LibreELEC

Zaroz, 2020-12-24:
Hello, I would like to request that an installer script for the LibreELEC OS be written, so that people can use the deskpi as a media center case.

wardnuke, 2020-12-24:
My LibreElec MicroSD card runs fine, but since there's no DeskPi scripts for the new SD card reader, SSD, or USB port moves I can't control the desktop. Hopefully they act on this issue cause these things would make great LibreElec Media Centers which is what I bought them for in the first place. I can live with Kodi 18.7 but running the latest Kodi would be awesome.

jojobrogess, 2020-12-24:
I was thinking about this earlier and google searched to see if there was a solution because I will need to do this later myself. Idk if it would work, but you could try ssh'ing into the LibreELEC device and installing the desk pi fan script and turning the IR interface on that way since there isn't a native terminal in libre. As for the usb boot sequencing/quirks idk if that would work with LibreELEC, so you might be limited to SD card only for the time being.

bobaxxx, 2020-12-26:
A litteraly waste 100 buckets for that case.

- No script for libreelec, only few distros can make use of fans and rear usb, deskpi team can at least make an add-on for libreelec, because there is no possibility to install .sh script on libreelec.

wardnuke, 2020-12-26:
@bobaxxx 

I'd give it some time. I'm not a big fan of "Made in China" stuff, but the unit I have put together seems to work pretty well despite the issue with the SSD speed. As I said in some other posts, they're testing another controller chip and hope to ship replacement PCB's out to past purchasers starting March 2021. They say the new chip they're testing will fix the SSD bottleneck issue. I'm sure some smart cookie will figure out and distribute a script for LibreElec to run along with the DeskPi units. It's in their best interests as that's pretty much what these boxes are designed for and like you, those of us purchased it for. In the meantime, I've run some stress tests on my overclocked Pi4 8GB and I have to say the results are VERY promising. The Ice Tower cooling is awesome. My CPU didn't make it past 57.9 deg C after 10 minutes and the fan never broke 50%. I'm very impressed and the prospects are nice. I'll stick with them for a bit and see how they do. I only paid about $110 for two boxes (I have my own Pi4's) and they only released this "model" with the updated GPIO ports and SSD M.2 support in November. I think if we, the comsumer, bears with them for a little, these boxes will turn out nice.

yoyojacky, 2020-12-28:
> @bobaxxx
>
> I'd give it some time. I'm not a big fan of "Made in China" stuff, but the unit I have put together seems to work pretty well despite the issue with the SSD speed. As I said in some other [...]

Thank you very much for your affirmation of our products, we will continue to work hard.

yoyojacky, 2020-12-28:
> A litteraly waste 100 buckets for that case.
> 
> * No script for libreelec, only few distros can make use of fans and rear usb, deskpi team can at least make an add-on for libreelec, because there is no possibility to install .sh script on libreelec.

We are preparing to contact libreELEC official, hope they can provide dtoverlay like Raspberry Pi official to support front USB interface, fan driver is provided through C or Python, we are still working hard to test, please be patient, we are Try to support more distributions.

yoyojacky, 2020-12-28:
> My LibreElec MicroSD card runs fine, but since there's no DeskPi scripts for the new SD card reader, SSD, or USB port moves I can't control the desktop. Hopefully they act on this issue cause these [...]

Yes, it works on kodi.

yoyojacky, 2020-12-28:
> Hello, I would like to request that an installer script for the LibreELEC OS be written, so that people can use the deskpi as a media center case.

https://forum.libreelec.tv/thread/23217-i-want-to-try-to-make-libreelec-support-deskpi-what-should-i-do/  
I was posted a post here, and try to figure out how to add the funciton to enable `dwc2` on libreELEC.

yoyojacky, 2020-12-28:
> Hello, I would like to request that an installer script for the LibreELEC OS be written, so that people can use the deskpi as a media center case.

> My LibreElec MicroSD card runs fine, but since there's no DeskPi scripts for the new SD card reader, SSD, or USB port moves I can't control the desktop. Hopefully they act on this issue cause these [...]

The dwc2 driver isn't included in LibreELEC, but with LibreELEC 10 you'll be able to use otg_mode=1 - this is already working in development builds, see here [BUG] Missing configuration parameter for DWC2 at RPI4 · Issue #4722 · LibreELEC/LibreELEC.tv · GitHub

Zaroz, 2020-12-31:
> I was thinking about this earlier and google searched to see if there was a solution because I will need to do this later myself. Idk if it would work, but you could try ssh'ing into the LibreELEC [...]

I tried ssh'ing into it and installing there, but it said that the installer script required apt, which libreELEC does not have.

> A literaly waste 100 buckets for that case.
> 
> * No script for libreelec, only few distros can make use of fans and rear usb, deskpi team can at least make an add-on for libreelec, because there is no possibility to install .sh script on libreelec.
That's not entirely true, the Argon One uses .sh script and, while not exactly the same as the raspbian version, it **does** work on LibreELEC.

bobaxxx, 2020-12-31:
Somebody is comparing the argon 1, i got 2 argon1 cases, 20 bucks each :

- Works flawlessly without some "quirks" command, or anything else
- argon1 script fan control is intended at start to works on Libreelec (launch as a service at start), and it works (install, working and uninstall script)

How do you explain argon1 script is functioning and then deskpi-team script doesn't function ?

wardnuke, 2021-01-02:
@bobaxxx 

I think the "script" they're referring to "not work" for the DeskPi is the one that loads and activates the various drivers and such that make the extension board work. I think there are controllers and such that must be activated and set up so that the board they use to move the HDMI / USB and such along with the IR controller and power button have to be loaded and this is the issue. Like I've said in previous posts, this is the initial launch and I'm sure they'll get things right. I think the biggest issue they have working against them is they're based overseas and getting stuff to the US market takes a bit. My two DeskPi Pro boxes took almost a month to get to me. Hopefully they'll get the bugs worked out and the supply chain flowing so that they can deliver a better product in a more timely manner. Otherwise some enterprising person is going to set up something here in the US and dominate the business. If there is any sort of trade war between the US and China it'll impact their supply flow. Someone here in the US that makes a similar product, using the lessons learned from this box, would make a killing as the UK would probalby follow with the US in any dispute with China. Since Raspberry Pi's are very popular in the UK, that would be real bad for the DeskPi folks if they were cut out of the US market along with the UK market because of flaws and supply issues. They'd better get this fixed or someone will swoop in and take their business.

yoyojacky, 2021-02-02:
> 我想请求编写LibreELEC OS的安装程序脚本，以便人们可以将deskpi用作媒体中心盒

Thanks very much.... and what can i do for you?

yoyojacky, 2021-02-02:
> * 在

The working methods of the two products are different, and the same scripts cannot be used to achieve functions.

Zaroz, 2021-02-06:
> > 我想请求编写LibreELEC OS的安装程序脚本，以便人们可以将deskpi用作媒体中心盒
> 
> Thanks very much.... and what can i do for you?

We are asking for you to write an install script for LibreELEC OS, so the device is able to **_work_**.

yoyojacky, 2021-03-12:
> > > 我想请求编写LibreELEC OS的安装程序脚本，以便人们可以将deskpi用作媒体中心盒
> > 
> > 
> > Thanks very much.... and what can i do for you?
> 
> > > * 在
> > 
> > 
> > The working methods of the two products are different, and the same scripts cannot be used to achieve functions.
> 
> We are asking for you to write an install script that will work on LibreELEC OS, so the device is able to **_work_**.

oh, sorry for my misunderstanding, we are working on it, it is a little bit hard for me due to the LibreELEC OS is not a normal linux  distribution...

yoyojacky, 2021-05-11:
> I tried ssh'ing into it and installing there, but it said that the installer script required apt, which libreELEC does not have.

I have tried several times but no luck too. it requires some packages as well.

jojobrogess, 2021-06-25:
@yoyojacky what drivers are you referring to? 

I'm going to attempt at making an installer. Idk how to code, and I barely understand Kodi addon/plugins. Or how they operate but it shouldn't be too difficult considering the Argon One m.2 does the same thing and there is an installer script for it. 

So I'm basically going to attempt at pirating that script and then hacking it for the deskpi pro case. 

I already found the argon one libreelec installer script and ported over a few terms to deskpi already. 

https://github.com/jojobrogess/deskpi/blob/Libreelecinstaller/libreelec.sh
 
Later on tonight I will follow this guide: 

https://www.argon40.com/learn/index.php/2020/03/10/argon-one-installation-guide-for-libreelec/

Hopefully I won't have to customize too much. 

It honestly doesn't look too complicated. 

The Argon one m.2 adapter board seems to work from what people are posting online.

And libreelec 10 beta has already been released so the script can be built using a dev build. 

I'm also in the process of trying to understand how Kodi addons work. If I can get this installed and working correctly. I will try to create an addon(and repo), so you can easily change the temp values within Kodi. 

And hopefully, just design it as an addon rather than an installer script that has to be SSH'd or CLI'd in. Because that would be lame to have to SSH into your device just to change the fan speed or temp ranges. 

Wish me luck!

jojobrogess, 2021-06-25:
yeah so i really do not know how to code. lol
but from the research I keep getting, I honestly think it's possible. just a little more complicated than the other installer scripts since there it isn't a normal OS at all. It's literally a ported over xbmc(kodi), with just enough OS to run everything on hardware.

but i honestly think it's possible, it just has to be done by someone smart than I

To get the Python GPIO library, all you have to do is to install the "Raspberry Pi Tools" add-on. Start your Python script with this lines to import that add-on lib:

>#!/usr/bin/python
>
>import sys
>sys.path.append('/storage/.kodi/addons/virtual.rpi-tools/lib')
>import RPi.GPIO as GPIO

`serial_port='/dev/ttyUSB0'`
 This is the serial port that connects to the deskPi mainboard and it will
 communicate with Raspberry Pi and get the signal for fan speed adjusting.

yoyojacky, 2021-06-30:
> yeah so i really do not know how to code. lol
> but from the research I keep getting, I honestly think it's possible. just a little more complicated than the other installer scripts since there it [...]
In fact, I tried many methods, but none of them worked, because the user needs to configure two parts of deskpi: on the one hand, the dwc2 function needs to be enabled, so that the /dev/ttyUSB0 device can be created in the Linux system, and This device is the key to the communication between the Raspberry Pi and the MCU of the daughter board. It needs to send the fan speed information and the information of cutting off the power after shutdown to the MCU. I like libelect very much, but there are not enough interfaces for me to perform these operations, maybe I don't understand how it works. There is no detailed introduction in the official documents, so it is more difficult.

jojobrogess, 2021-06-30:
> In fact, I tried many methods, but none of them worked, because the user needs to configure two parts of deskpi: on the one hand, the dwc2 function needs to be enabled, so that the /dev/ttyUSB0 [...]

_Originally posted by @HiassofT in https://github.com/LibreELEC/LibreELEC.tv/issues/4722#issuecomment-739526268_
> Just add a line with `otg_mode=1` to the end of /flash/config.txt, just like you'd add the dtoverlay line.
> 
> This option will enable the builtin XHCI controller on the USB-C port instead of the DWC2 controller.

Yeah this will have to wait until Libreelec v10 is fully released. But as of right now, I am on `LibreELEC (official): 9.95.4 (RPi4.arm)` and after enabling otg_mode=1 this is what what is returned when I use `lsusb -t -v`

[image]

and it looks like this located at:

[image]

Might need to `mount -o remount,rw /dev` before you can use it. 

This might help:
[image]

[image]

BTW usb boot on `LibreELEC (official): 9.95.4 (RPi4.arm)` works fine, I'm just not running it that way atm. 

I wish I knew more to help, I'll keep searching and trying. hopefully eventually I get it.

jojobrogess, 2021-06-30:
After a bunch of tinkering I was able to get the install script to install the files correctly in the correct locations, but I'm having problems with installing the drivers. 

I'm not sure if I should use c or python, I would assume python but what do I know lol. 

With the C driver: idk if the make command works within libreelec. So I'll either have to build one outside of libreelec and include it in the installer or figure out how everything works so I can break it and do it by hand. 

With the python driver: it might have to be changed around slightly or include it within an addon for pyserial to work correctly 
Like it's suggested here: 

https://forum.kodi.tv/showthread.php?tid=203883

When that doesn't work, I'll have to install the drivers by hand and then modify the install code to match that "working" process. 

I would also care to assume that both pwmFanControl and the shutdown script need to be heavily modified to work within the libreelec file system. 

I've already had to remove all "sudo" calls, and find workarounds for the "read-only" file system. 

Hopefully later on tonight I'll get a little closer to getting the fan to turn on 

If anyone has any type of information they can share to help me out, or point me in the right direction I would greatly appreciate it! I honestly don't know what I'm doing lol

jojobrogess, 2021-07-01:
@yoyojacky do you have any information on the GPIO configuation? 

because I found this:

https://forum.libreelec.tv/thread/9472-cooling-fan-control-raspberry-pi-libreelec/

and it seems promising..

jojobrogess, 2021-07-01:
I was able to find a pyserial that was made for libreelec matrix 

https://mirrors.kodi.tv/addons/matrix/script.module.pyserial/

jojobrogess, 2021-07-04:
Yeah I think I got it working. let me figure out how to clean everything up and I'll post a follow up instruction manual with files.

jojobrogess, 2021-07-04:
************************************************************************************************************************************

## https://github.com/jojobrogess/deskpi/tree/Libreelecinstaller 
### If you want to help out, or take for yourself

************************************************************************************************************************************

### !!!PREFACE!!!:
I have absolutely **NO** idea what I'm doing. 
Use all of this at **your own risk**!
I also for some reason can't seem to be able to uninstall it fully.
I do NOT know if the power button works, my device has the pins set to `always on`. The blue light _is_ on though.

I added in a lsb library from my own personal RPI 4 (found in: /lib/lsb/).
The Pyserial module and url was found by GOOGLE SEARCHING so idk if it's safe, but its hosted on a kodi mirror site soooo...

I do NOT (currently) use the Deskpi Pro case for Libreelec. 
I just got bored and was curious as to why this would be complicated in the first place.
(Simple answer there just isn't that much info on the OS) 
BUT if I get bored again, I might update, clean, and fix the installer. 
As well as create an addon so you don't have to ssh into your device everytime if you want to change the temp ranges. 
I already have some code sketched out, I just need to learn a little more about how to create an addon and then how to get that addon to operate less like an addon and more like simple bash/python. Oh yeah plus everything else in-between.
But I digress...

************************************************************************************************************************************

### How to install DeskPi Pro script for Power Button and Fan Control:

************************************************************************************************************************************
REQUIREMENTS:
You MUST be on at least `LibreELEC-RPi4.arm-9.95.4`
extra computer or at least ssh terminal
extra usb

You will need to load a matrix ready version of pyserial on a USB and install it as an addon. 
`https://mirrors.kodi.tv/addons/matrix/script.module.pyserial/` 
MAKE SURE you choose the latest(as of right now it's):
Script.module.pyserial-3.4.0+matrix.2.zip | 100.0 KiB | 2020-May-01 14:30

**IF you're `ADVANCED` GO TO BOTTOM**

************************************************************************************************************************************

### After you do that, then follow these instructions:

************************************************************************************************************************************

Connect LibreElec to the internet.
        This can be done with either ethernet or WiFi
        To check if the Ethernet or WiFi adapters are enabled, go to Settings>LibreElec>Network 

Enable SSH. You need another computer to access the terminal of LibreElec
        This can be done upon installation during wiki. Default Username= `root` Password=`libreelec`
        To check if SSH is enabled, go to
        Settings>LibreElec> Services 

You might also want to make sure to allow addons/updates from any source. 
        Settings> System> Addons

Install Raspberry Pi Tools and System Tools
        These can be installed by going to Addons>Install from Repository>LibreElec Addons>Program Addons 

Plug the USB device you loaded with pyserial into your device.
Install Pyserial from usb
         Go to Addons>Install from a zip file>Navigate to usb drive>pyserial.zip (should install like an addon) but it's just a library.(AFAIK)

Connect via SSH. The default username for LibreElec is `root` while the default password is `libreelec`
        For Windows users, the best way to use SSH is through Putty using the IP address of the Pi and port 22.
        Linux users `ssh root@[ip address of device]`.
            If connecting to the device for the first time, you will be asked if you're sure you want to connect to the device. 
[... 21 lines cut ...]
Reconnect through SSH and run this:
`./Libreelecinstaller/Deskpi-config`

Use the config menu normally. 
At this moment, idk what happens if you try to set custom values or if the options would even be persistent after reboot.

************************************************************************************************************************************

### FOR ADVANCED

************************************************************************************************************************************

1. Download `https://mirrors.kodi.tv/addons/matrix/script.module.pyserial/` put on USB.
2. Install .zip through addons page.
3. Install `Raspberry Pi Tools` and `System Tools` from Kodi repo.
4. ssh into device
5. `wget https://github.com/jojobrogess/deskpi/archive/refs/heads/Libreelecinstaller.zip`
6. `unzip Libreelecinstaller.zip`
7. `chmod +x deskpi-Libreelecinstaller/lib/lsb/init-functions && chmod +x deskpi-Libreelecinstaller/install-libreelec.sh`
8. `./deskpi-Libreelecinstaller/install-libreelec.sh`
9. Wait for reboot. Reconnect ssh.
10. `./Libreelecinstaller/Deskpi-config`
11. Adjust speed and temps.

Note (as of right now): I have not been able to make an addon to be able to adjust the deskpi-config files so you can change the temperature ranges without having to SSH into your libreelec every time. If I can build one, I will either post in this thread or make a new one.

jojobrogess, 2021-07-05:
IF THIS WORKS FOR YOU, pleeeaaase let me know.
this is my first ever published script.
@bobaxxx @wardnuke @bobaxxx and anyone else

haan2787, 2021-07-10:
i have used pinn with libreelec but usb works only deskpi-config not no temp control
but thanks for the great work

jojobrogess, 2021-07-10:
> i have used pinn with libreelec but usb works only deskpi-config not no temp control
> 
> but thanks for the great work

I really wish I could understand what you wrote, maybe I could help out.
I've never heard of Pinn before, but with a simple google search it looks to be like NOOBS, in that it's a boot loader for rpi's to multi boot. 

As long as you are running at least libreelec 9.95.4, everything should work. 

Usb boot options(/update eeprom) are in the librelec icon in settings tab of Kodi.

haan2787, 2021-07-12:
thanks for youre help pinn is like noobs 
front usb is working 
only deskpi-config in putty is not working
so i can not control the temp
i will try again maybe i do something wrong
gr  haan2787

jojobrogess, 2021-07-12:
@haan2787 yeah you're correct. 
I connected up my deskpi last night and tried to enter custom values, it didn't work. 
I'll take a look at it today. My guess would be it's because of how the driver scripts work. 

TBH I didn't really test that part of the script, I was a little bit more concerned with @yoyojacky statements that it was not possible to even control the fan. Which I had to prove wrong 

It should work if you set it to the default values:
`sh -c "echo pwm_025 > $serial_port"`
`sh -c "echo pwm_050 > $serial_port"`
`sh -c "echo pwm_075 > $serial_port"`
`sh -c "echo pwm_100 > $serial_port"`
`sh -c "echo pwm_000 > $serial_port"` (this turns the fan off, or at least sets the fans value to 0)

Regardless, I'll take a look at this. But it might take me a little bit. 

I am currently trying to figure out If I can get this installed with an ADDON instead of the process I tutorial'ed above, and then control the fan and it's speeds and ranges within that addon. (if I'm able too, I will make it compatible with the Argon One fan control script as well).

It IS possible btw, I just don't know how to code so it will take me a little bit.

haan2787, 2021-07-14:
thanks for youre hard work i wil follow you when you are ready take youre time

inspector71, 2021-07-23:
Like to add my thanks. First I've seen of this Issue, thread just now. Congrats on giving it a go!

FWIW, AFAIK LibreELEC runs on BusyBox which is a Linux distribution used often by embedded system OEMs.

inspector71, 2021-08-07:
> Idk if it would work, but you could try ssh'ing into the LibreELEC and turning the IR interface on 

has anyone experienced the IR function working well because I am astounded at how poorly the sensor appears to be placed and thankful for CEC or DeskPi would be useless as a Kodi box for me.

inspector71, 2021-08-07:
> I tried ssh'ing into it and installing there, but it said that the installer script required apt, which libreELEC does not have.

yep, LibreELEC is based on BusyBox. Same distro a lot of routers and the like use.

@yoyojacky thanks for continuing to update us on this. I can understand people are frustrated but it's also good to have some feedback on the efforts you are making.

inspector71, 2021-08-12:
> If anyone has any type of information they can share to help me out, or point me in the right direction I would greatly appreciate it! I honestly don't know what I'm doing lol

If you've considered turning this into a Kodi Add-on, this post / tool might help:

https://forum.kodi.tv/showthread.php?tid=339544

yoyojacky, 2021-10-13:
> > In fact, I tried many methods, but none of them worked, because the user needs to configure two parts of deskpi: on the one hand, the dwc2 function needs to be enabled, so that the /dev/ttyUSB0 device can be created in the Linux system, and This device is the key to the communication between the Raspberry Pi and the MCU of the daughter board. It needs to send the fan speed information and the information of cutting off the power after shutdown to the MCU. I like libelect very much, but there are not enough interfaces for me to perform these operations, maybe I don't understand how it works. There is no detailed introduction in the official documents, so it is more difficult.
> 
> _Originally posted by @HiassofT in [LibreELEC/LibreELEC.tv#4722 (comment)](https://github.com/LibreELEC/LibreELEC.tv/issues/4722#issuecomment-739526268)_
> 
> > Just add a line with `otg_mode=1` to the end of /flash/config.txt, just like you'd add the dtoverlay line.
> > This option will enable the builtin XHCI controller on the USB-C port instead of the DWC2 controller.
> 
> Yeah this will have to wait until Libreelec v10 is fully released. But as of right now, I am on `LibreELEC (official): 9.95.4 (RPi4.arm)` and after enabling otg_mode=1 this is what what is returned when I use `lsusb -t -v`
> 
> [image]
> 
> and it looks like this located at:
> 
> [image]
> 
> Might need to `mount -o remount,rw /dev` before you can use it.
> 
> This might help: [image]
> 
> [image]
> 
> BTW usb boot on `LibreELEC (official): 9.95.4 (RPi4.arm)` works fine, I'm just not running it that way atm.
> 
> I wish I knew more to help, I'll keep searching and trying. hopefully eventually I get it.

wow, that's cool, and we need to enable dwc2 overlay to generate /dev/ttyUSB0 as raspberry Pi usb-c port to MCU on daughterboard, your device is using usb port to generate serial port ....
Did you generate serial port `/dev/ttyUSB0` mapped to USB-C port on Raspberry Pi ?

yoyojacky, 2021-11-04:
> Somebody is comparing the argon 1, i got 2 argon1 cases, 20 bucks each :
>
> * Works flawlessly without some "quirks" command, or anything else
> * argon1 script fan control is intended at start to works [...]

Because the working mechanism is different

inspector71, 2021-11-04:
On Thu., 4 Nov. 2021, 2:21 pm yoyojacky, ***@***.***> wrote:

> Somebody is comparing the argon 1, i got 2 argon1 cases, 20 bucks each :
>
>    - Works flawlessly without some "quirks" command, or anything else
>    - argon1 script fan control is intended at start to works on Libreelec
>    (launch as a service at start), and it works (install, working and
>    uninstall script)
>
> Thanks!

Which version of LibeELEC though?

The difference between 9.2 and 10 seems to be a show stopper in three
regards: yoyojacki cannot find a way to support 9.2; massive Kodi graphics
drivers overhaul for 10; add-ons require  at least recompilation in Python
3 for 10.

If argon works, I can consign my DeskPi to a limited failure not fit for
the purpose I purchased it for (ease of use as a boot to Kodi solution with
... you know, a fecking revolutionary device called a ... built in, fully
functioning power button! Yep, in Raspberry Pi context ... what an
expectation, can you believe it 🤫🤔🤐😯☹️😏 ?).

I do not even really care much about ...

- IR receiver (using HDMI-CEC)
- front USB (though it's damn shitty to have ports viewable that don't work)
- SATA3 (though downgrading after purchase is bullshit)
- GPIO
- Dual HDMI (not necessary for a Kodi box but handy if I was ever desperate
enough to use it for half-arsed desktop)
- 4K (no screens to use it yet)

1. I just want any storage option actually suitable for running an OS,
unlike an SD card or USB stick.

2. a power button with user friendly simple features like killing a frozen
OS with a 'long press' or whatnot, turning the device on!

3. a decent, functional cooling system ... not because I want / need to
overclock it or anything but just because, you know, it's surely sensible
to moderate the heat of any silicon to hopefully improve lifespan!

jojobrogess, 2021-12-17:
I am just so upset at this whole endeavor..

This **issue** was opened ALMOST a YEAR AGO.

6 MONTHS AFTER I took 2 WEEKS to create a PROOF OF CONCEPT, to prove it COULD be done.

6 MONTHS HAVE PASSED SINCE, and still nothing. 

@inspector71 your best bet at this point is either to use my code or give this idea up. 
I HIGHLY doubt @yoyojacky is going to do anything with this.

inspector71, 2022-01-23:
> I am just so upset at this whole endeavor..
>
> This **issue** was opened ALMOST a YEAR AGO.
>
> 6 MONTHS AFTER I took 2 WEEKS to create a PROOF OF CONCEPT, to prove it COULD be done.
>
> 6 MONTHS HAVE [...]

@jojobrogess I *feel* you, mate, as they say these days.

I was a bit intimidated by your code - well, not the code itself so much but the task overall as I needed / wanted this hardware to be a solution to the decade-long piecemeal, half-arsed RPi ballache of promise but failing to live up to it - but the rebel in me almost wants to try it just to confirm the poor service was not technically based as is the excuse that seems to be put forward. That and patience. Sorry, patience has run out. 

Two weeks on your behalf is a major, impressive investment and I personally appreciate it that much that I would really like to have the capability to use it. Can you confirm it works with LibreELEC 9.2.* ? I cannot see any benefit from going to LibreELEC 10 as the architectural changes are not helpful in the immediate term. Rather, a hinderence. Though I understand why they have been made.

Zaroz, 2022-01-23:
> I am just so upset at this whole endeavor..
>
> This **issue** was opened ALMOST a YEAR AGO.
>
> 6 MONTHS AFTER I took 2 WEEKS to create a PROOF OF CONCEPT, to prove it COULD be done.
>
> 6 MONTHS HAVE [...]

If you think you're upset, try to imagine how I'm feeling: I created this issue over a _year_ ago, thinking it wouldn't be a long wait until it was resolved. After all, the Argon ONE, which they were always comparing their product to, had quickly released a installer script for LibreELEC OS. Little did I expect that not only would it not get fixed, they even upgraded the hardware, but didn't fix this. At this point I've long given up and bought an Argon ONE instead. I also left a scathing review for this product. That will be the last time that I pre-order this type of product.

inspector71, 2022-01-24:
@Zaroz This thread has become a support group for dissatisfied buyers :)

Are you using the Argon ONE with LibreELEC 9.2.* as your 'daily driver' ?

Until I'm absolutely certain, I'm not making any moves. Wasted too many resources (mental, money) on RPi based media products.

Zaroz, 2022-01-24:
@inspector71, if by 'daily driver' you mean my primary media player, then yes. I donwnload/store the videos from sonarr/radarr on my nas then use the pi to play them

jojobrogess, 2022-01-27:
They just released the Deskpi Lite. "Driver Compatible with DeskPi Pro". 

Hopefully they'll look into this, but again I doubt it.

inspector71, 2022-01-27:
Yeah, they've been better at regurgitating new variations than supporting
the first one properly.

yoyojacky, 2022-03-02:
> 

Sorry for the slow response due to i am working on testing those deskpi Pro individually ....
I have try to install deskpi driver in  libreELEC, but i am not an expert on this OS. so it tooks me a lot of time to learn about it. and i am still have no idea how to compile the properly binary file and running on it....
I may need some of your help...

inspector71, 2022-03-03:
> > 
> 
> sorry for let you down...

@yoyojacky sounds like a difficult time for you. Thank you for your honesty and updating us. Whilst my disappointment remains, it's good to know you are trying your best. I'm still wondering if you could look at outsourcing the problem but perhaps that is not an option.

At present my unit is working within the limits already discussed. Probably helps that it is underneath a split system air conditioning unit on cool / dehumidify mode, out of the case, in lieu of a fan. The older Pi my Mum relies on, which I wanted to improve with the deskpi, is for the moment behaving itself. 

I feel your genuine intent to do the ethically legitimate best by your customers and appreciate this. 

Stress is not healthy for the body so if you are stressed due to this issue, feel free to take a few steps back if pressure from me is adding to the stress.

I am willing to help with testing anything you may come up with, if I can.

yoyojacky, 2022-03-10:
@inspector71 thanks bro, I am trying to learn about libreELEC and try to make it works, hope i can fix this issue..
TODO:
1. add dtoverlay=dwc2,dr_mode=host to libreELEC in a easy way.
2. Try to make serial module works on libreELEC.
3. Add python3 code to control the fan and sending power off signal to MCU and getting poweroff signal from MCU when double clicking the power button.

jojobrogess, 2022-03-11:
Just more levels upset, especially when it feel like manipulation is afoot. 

> 
>     1. add dtoverlay=dwc2,dr_mode=host to libreELEC in a easy way.
> 

@yoyojacky posted this comment ON [Dec 27, 2020](https://github.com/DeskPi-Team/deskpi/issues/17#issuecomment-751616281)
 alex1702 closed this [on Dec 8, 2020](https://github.com/LibreELEC/LibreELEC.tv/issues/4722#event-4084614184)
> https://forum.libreelec.tv/thread/23217-i-want-to-try-to-make-libreelec-support-deskpi-what-should-i-do/
> I was posted a post here, and try to figure out how to add the funciton to enable `dwc2` on libreELEC.

Corresponding Libreelec github issue [4722](https://github.com/LibreELEC/LibreELEC.tv/issues/4722)

HE ALREADY KNOWS DWC WON'T BE SUPPORTED AND TO USE `otgmode=1`
THIS WAS POSTED 2 YEARS AGO.

**************************************************

> 2\. Try to make serial module works on libreELEC.

I posted this June 30th 2020

> I was able to find a pyserial that was made for libreelec matrix
> 
> https://mirrors.kodi.tv/addons/matrix/script.module.pyserial/

**************************************************
> 3\. Add python3 code to control the fan and sending power off signal to MCU and getting poweroff signal from MCU when double clicking the power button.

https://github.com/DeskPi-Team/deskpi/blob/eb761f269d2db607968463c45d908d30e3627d01/drivers/python/pwmControlFan.py
Yeah idk about this though, but shouldn't take 2 years to switch from python 2 to python 3.

jojobrogess, 2022-03-11:
I'm pretty sure I figured out how to do this last night without ssh and through an addon.

jojobrogess, 2022-03-12:
oh yeah, totally got it. 
halfway there...

the addon has been made.

Now I just have to re-rewrite my modified deskpi service to actually work this time.

inspector71, 2022-03-12:
> oh yeah, totally got it. halfway there...
> 
> the addon has been made.
> 
> Now I just have to re-rewrite my modified deskpi service to actually work this time.

Sounds exciting! For LE10 users at least :)

jojobrogess, 2022-03-12:
> Sounds exciting! For LE10 users at least :)
> 
> Thanks for your ongoing efforts!

`LibreELEC 9.2.0 (Leia) the final version has arrived based upon Kodi v18.5`

Any particular reason why you are staying on Leia?

There have been an intense amount of updates from 9.2-10.0.2 in libreelec and even more from kodi 18.5-19.4. 

Plus any addons that you're staying for, that haven't updated to work with python 3 will be dead in nexus. Since legacy python 2 support ends there.

inspector71, 2022-03-13:
> @inspector71
> 
> > Sounds exciting! For LE10 users at least :)
> > Thanks for your ongoing efforts!
> 
> `LibreELEC 9.2.0 (Leia) the final version has arrived based upon Kodi v18.5`
> 
> Any particular reason why you are staying on Leia?

Any particular reason to install Matrix? Nope :)
 
> There have been an intense amount of updates from 9.2-10.0.2 in libreelec and even more from kodi 18.5-19.4.
> 
> Plus any addons that you're staying for, that haven't updated to work with python 3 will be dead in nexus. Since legacy python 2 support ends there.

Yeah, yeah, I know the IT industry make it almost impossible to keep a perfectly stable, solid system. Instead, forcing people to risk regressions / 'features' they don't want or hate, by 'up' grading. 

I'll review my analysis in time. 

Willing to feck around with a perfectly working system to test DeskPi integrations. That's the reason I'm staying subscribed to this issue :)

inspector71, 2022-03-13:
Update:

Reasons not to 'up' grade? 

`Rpi4 crashes at H264 and H265 playback`

Errrm, she dun play videos mate! Kiiiiinda a key feature of a media player? :)

That's a release note for version 10.0.1 ... can you imagine how much stress I would have experienced trying 10 and experiencing that alpha level bug ... in a release version!

I'm just super lucky, because it's not always been the case, that I somehow fluked reading the changes behind Matrix were fundamentally huge and thus LE10 would be a ballache, in particular for RPI users.

jojobrogess, 2022-03-13:
Yeah can confirm, I'm having the hardest time at the moment. 
4 hours of dropped ssh connections/wifi network, trying to install system-tools. 
Just finally got it. 
And now every window load freezes kodi...

inspector71, 2022-03-14:
> Yeah can confirm, I'm having the hardest time at the moment. 4 hours of dropped ssh connections/wifi network, trying to install system-tools. Just finally got it. And now every window load freezes kodi...

Sounds painful. I hate it when Kodi freezes.

Best to take plenty of breaks if you can?

jojobrogess, 2022-03-15:
@yoyojacky you never fully looked at/ ran my code did you?

jojobrogess, 2022-03-15:
I guess it doesn't matter, it's been a year since I released it. 
You had ample time to. 
But now I 100% know you didn't..

[53 more replies]