DeskPi-Team/deskpi #18: REAR USB 2.0 PORTS DO NOT WORK

wardnuke, 2020-12-24:
Neither of the USB 2.0 ports on the back of the DeskPi Pro work with my mouse / keyboard doggle. I had to move the doggle to the front of the panel. The USB 3.0 and 2.0 ports on the back work just fine for USB sticks and a USB SSD drive I had, but if I put the doggle for my mouse / keyboard in it doesn't work.

Exscotticus, 2020-12-26:
I noticed that the FRONT USB ports didn't work when I first booted up on an SSD. Updates and reboot and all was well again.

yoyojacky, 2020-12-28:
> I noticed that the FRONT USB ports didn't work when I first booted up on an SSD. Updates and reboot and all was well again.

This is because the 2.4GHz of the USB dongle interferes with the communication of the USB3.0 interface. We have made improvements in the next generation of products. Please contact our sales staff and provide your order number when you buy. They will be in March next year. The circuit board will be reissued for you around mid-term, replacing the current version.

yoyojacky, 2020-12-28:
> Neither of the USB 2.0 ports on the back of the DeskPi Pro work with my mouse / keyboard doggle. I had to move the doggle to the front of the panel. The USB 3.0 and 2.0 ports on the back work just [...]

This is because the 2.4GHz of the USB dongle interferes with the communication of the USB3.0 interface. We have made improvements in the next generation of products. Please contact our sales staff and provide your order number when you buy. They will be in March next year. The circuit board will be reissued for you around mid-term, replacing the current version.

closed 2020-12-28 by yoyojacky