DeskPi-Team/deskpi #19: Fan buzzes and rattles

Exscotticus, 2020-12-26:
I managed to get everything working including booting from a 2.5" SSD (which required the USB "quirks command" mentioned in other threads).

My only issue now is that the fan makes a buzzing rattling noise on any speed other than 100%. Fortunately, the software provides significant control, so I just don't run the fan until things get hot. Then I run it at 100%.

wardnuke, 2020-12-26:
@Exscotticus 

I had issues putting the assembled board / Pi4 in the case. It was pretty tight and the heatsink looks like it's right up against the case. Is it possible you bent the copper heatpipes when assemblying and the fan is rubbing against something?? I noticed when I was getting ready to put mine in the box that the heatsink seemed to be "cocked" and discovered that while I was handling the board to tighten the various screws and such that the copper heatpipes deformed a little and the  heatsink was at an angle. You might want to disassemble the board and see if something is rubbing.

On another note, I ran Stress and a CPUBurn-A53 stess test on my board that I got from a Pi website: 

https://core-electronics.com.au/tutorials/stress-testing-your-raspberry-pi.html

I'm running a Pi4b 8GB with CPU overclocked to 2Ghz and GPU overclocked to 750Mhz. After 10 minutes of the highest stress testing the CPU only ever got to 57.9 deg C and the fan never went past the 50% setting. I have the 75% fan setting set to come on at 60 deg C. Hope this helps.

Exscotticus, 2020-12-26:
Thank you.

I must admit that I found the assembly difficult—particularly the part where the "heat exchanger" brackets must flex quite a bit to actually screw down, putting a disconcerting amount of force on the CPU chip. At first I thought maybe they had sent me the wrong lengths of standoffs.

I'll remove the chassis and power up to see if I can locate the source of the noise. I just hope it's not a fan with faulty bearings. I've encountered so many noisy muffin fans in recent years...

yoyojacky, 2020-12-28:
> My only issue now is that the fan makes a buzzing rattling noise on any speed other than 100%. Fortunately, the software provides significant control, so I just don't run the fan until things get hot. Then I run it at 100%.

That's because when the PWM is working, the voltage loaded on the fan changes in frequency, and when the fan rotates, it is controlled according to the current CPU temperature. By default, it is based on the configuration in /etc/deskpi.conf Perform temperature-speed corresponding speed adjustment. If you want to change the working state of the fan, you can adjust the state of the fan by typing in the terminal: deskpi-config.

yoyojacky, 2020-12-28:
> @Exscotticus
>
> I had issues putting the assembled board / Pi4 in the case. It was pretty tight and the heatsink looks like it's right up against the case. Is it possible you bent the copper heatpipes [...]

Well, you can set the speed through the deskpi-config command.

yoyojacky, 2020-12-28:
> I must admit that I found the assembly difficult—particularly the part where the "heat exchanger" brackets must flex quite a bit to actually screw down, putting a disconcerting amount of force on [...]

For the heat sink bracket, it is 0.8mm. We have designed it as a slightly stressful way to ensure better heat dissipation and will not damage your Raspberry Pi CPU. Please rest assured to use it. 
In addition, please pay attention to the position of the copper pillar during installation. M2.5x5+3mm is placed in the two positioning holes near the network card and CSI interface, while M2.5x4+4 is the two positioning near the TF card slot. hole.
If the fan is noisy, please execute: deskpi-config
Then choose 4 and press Enter.
Run for 1-3 minutes to fully dissolve the lubricating oil. By default, the oil in the bearing may solidify at low temperatures. When the temperature rises and dissolves, the sound will be much lower. This fan is specially customized and silent The effect will be very good. Especially after the speed is up.

closed 2020-12-28 by yoyojacky