DeskPi-Team/deskpi #20: Fan control install issues

arsirantala, 2020-12-29:
Hi, in ubuntu server I am having problems getting the install.sh to work without problems.

This is the output of the install.sh:

` ubuntu@ubuntu    ~/deskpi     master  sudo ./install.sh                                                                                                                                                                                                                                          1 ↵   16:23:42 
fatal: destination path 'deskpi' already exists and is not an empty directory.
 * DeskPi Fan control script installation Start.
 * Enable dwc2 on Host Mode
sed: can't read /boot/config.txt: No such file or directory
sed: can't read /boot/config.txt: No such file or directory
 * DeskPi main control service loaded.
 * DeskPi Service configuration finished.
 * DeskPi Service Load module.
 * DeskPi PWM Fan Control and Safeshut Service installed successfully.
 * System will reboot in 5 seconds to take effect.
Job for deskpi.service failed because the control process exited with error code.
See "systemctl status deskpi.service" and "journalctl -xe" for details.`

It would seem to me that the install.sh was meant to work with the Raspbian, and not with say ubuntu server. This is a problem, as I cannot use raspbian with my raspberry pi - so I wonder how am I going to control the cpu fan now? I can see that it without load is running already at 49 degrees celcius.

akesde, 2020-12-29:
I am having the same issue with the fan control on Ubuntu Desktop 20.10 64bit.  I do not think that it was meant to be run on a 64bit OS but I have an 8Gb Pi4 and I want to take advantage of the 64bit memory addressing.  Mine is hovering around 52 degrees centigrade without the fan so it is tolerable but would like it to be in the 40s.

arsirantala, 2020-12-30:
I'd prefer that Deskpi team would offer a solution how to be able to configure the fans in ubuntu, but if thats now somehow problematic, I did some googling and found out this: 
https://ograblog.wordpress.com/2019/10/09/attaching-a-cpu-fan-to-a-rpi-running-ubuntu-core/
Problem is that I have no idea which GPIO pins are used in deskpi pro to control the fan. Any ideas?

M-H-010, 2020-12-31:
I noticed the fan is not connected to the GPIO extender board, but to the daughter board that is connected to the pi its power usb-c port ( and hdmi & 3.5 mm jack ) In the scripts they send commands to a serial port ( ser = serial.Serial("/dev/ttyUSB0", 9600, timeout=30) ) .
So I wonder if they are programming a controller though that USB-C connection that will set the fan speed.
You can test this by omitting the GPIO riser board installation and setting fan speeds with the supplied OS image and tools.

arsirantala, 2020-12-31:
I highly recommend (for time being) to wire the fan directly to the gpio 5v and ground pins, otherwise you end up risking frying your raspberry pi (mine was with load close to 76 degrees celcius).
As the fan is not spinning at all otherwise!

M-H-010, 2020-12-31:
Arsirantala, a lot of pi users and engineers do not share you point of view. There is  no risk in frying you pi, as it is designed to run without a cooler. ( f.i. see https://www.raspberrypi.org/forums/viewtopic.php?t=262023)  
Also using the icetower without a spinning fan, does add a lot of extra cooling capacity. 
My set does not thermal throttle at all , even during a 12 minute 4 core stress test it did not pass 65 degrees Celsius.

arsirantala, 2020-12-31:
> Arsirantala, a lot of pi users and engineers do not share you point of view. There is no risk in frying you pi, as it is designed to run without a cooler. ( f.i. see [...]

Maybe so, in non-overclocked use case. But I'm using some overclocking so in my case I need the cpu fan to protect my raspberry pi :)

mbernardi1961, 2020-12-31:
Given there is now an official fan control built in to Raspberry Pi OS can this be used by DeskPi?

slightlyloos3, 2020-12-31:
im having the same issue but im using the Fenix OS. im completely new to using linux based systems

M-H-010, 2021-01-02:
@ mbernardi1961 you mention there is "official fan control built in to Raspberry Pi OS" can you link me to any documentaion of that ?
I have enabled the serial access to the power and fan controller of the deskpi, and can manually set the 5 fan levels. Most other controlled raspi fans I have seen are controlled from a  PWM controlled pin on the GPIO, and need a transistor to amplify the current for the fan motor. But these have many more fan level steps.
Still we can see if any existing software can be altered to control the deskpi fan.

mbernardi1961, 2021-01-02:
https://www.raspberrypi.org/blog/new-raspberry-pi-4-case-fan/

M-H-010, 2021-01-05:
@ mbernardi1961
Ok I have done a readup on the pi official fan, It is a 3 wire small fan. It has the feature to be switches on and off from a gpio pin. In raspbian there is a new software option to switch it on above a certain treshold, and it wil turn off at about 10 degrees celcius below that high temperature. Allthough not intended to be powered through PWM it should work that way.
The deskpi icetower fan is not controlled nor powered from the pi directly, but from the dauhterboard. And it is 2 wire.
Controlling the deskpi icetower from the raspbian software means you need to rewrite that program. I do not think it is likely that te raspberry foundation will put time and effort in this, but surely there will be other enthusiasts that can help you if you want to rewrite that. Please report any progress you gain, so others can use your knowledge.
M-H

yoyojacky, 2021-02-02:
> 嗨，在ubuntu服务器中，我无法顺利安装install.sh。
> 
> 这是install.sh的输出：
> 
> ```
> ubuntu@ubuntu    ~/deskpi     master  sudo ./install.sh                                                                                                                                                                                                                                          1 ↵   16:23:42 
> fatal: destination path 'deskpi' already exists and is not an empty directory.
>  * DeskPi Fan control script installation Start.
>  * Enable dwc2 on Host Mode
> sed: can't read /boot/config.txt: No such file or directory
> sed: can't read /boot/config.txt: No such file or directory
>  * DeskPi main control service loaded.
>  * DeskPi Service configuration finished.
>  * DeskPi Service Load module.
>  * DeskPi PWM Fan Control and Safeshut Service installed successfully.
>  * System will reboot in 5 seconds to take effect.
> Job for deskpi.service failed because the control process exited with error code.
> See "systemctl status deskpi.service" and "journalctl -xe" for details.
> ```
> 
> 在我看来，install.sh是用于Raspbian的，而不是用于ubuntu服务器的。这是一个问题，因为我无法在树莓派上使用raspbian-所以我想知道现在如何控制cpu风扇？我可以看到它在没有负载的情况下已经在49摄氏度下运行了。

The whole fan works in this way. The system sends "pwm_xxx" through the serial port, where xxx is the number 000 ~ 100 to the onboard MCU. The MCU parses the PWM signal and then drives the fan to rotate. You can refer to the content provided in the python script to write the control script yourself. Under normal circumstances, I recommend setting the temperature at 55 degrees to set 100% speed, so that the temperature can be controlled while also reducing noise. Low-frequency pwm does There is noise, but the whole problem can be solved by changing the pwm information. When the fan rotates at high speed, there is basically no noise.

yoyojacky, 2021-02-02:
> 我在Ubuntu Desktop 20.10 64bit上的风扇控件遇到相同的问题。我不认为它应该在64位操作系统上运行，但是我有一个8Gb Pi4，我想利用64位内存寻址的优势。没有风扇，矿井在52摄氏度左右徘徊，因此可以接受，但希望在40年代左右。

sorry, if you want to use it under 64bit OS, you may need to re-compile the C code so that it can works on your Platform.

yoyojacky, 2021-02-02:
> 我希望Deskpi团队会提供一个解决方案，以便能够在ubuntu中配置风扇，但是如果这现在仍然有问题，我会进行一些谷歌搜索并发现以下问题：
> [https](https://ograblog.wordpress.com/2019/10/09/attaching-a-cpu-fan-to-a-rpi-running-ubuntu-core/) : [//ograblog.wordpress.com/2019/ 10/09/10将一个cpu风扇连接到一个rpi运行ubuntu核心/](https://ograblog.wordpress.com/2019/10/09/attaching-a-cpu-fan-to-a-rpi-running-ubuntu-core/)
> 问题是我不知道在deskpi pro中使用了哪个GPIO引脚来控制风扇。有任何想法吗？

The whole fan works in this way. The system sends "pwm_xxx" through the serial port, where xxx is the number 000 ~ 100 to the onboard MCU. The MCU parses the PWM signal and then drives the fan to rotate. You can refer to the content provided in the python script to write the control script yourself. Under normal circumstances, I recommend setting the temperature at 55 degrees to set 100% speed, so that the temperature can be controlled while also reducing noise. Low-frequency pwm does There is noise, but the whole problem can be solved by changing the pwm information. When the fan rotates at high speed, there is basically no noise.

yoyojacky, 2021-02-02:
> 鉴于现在Raspberry Pi OS内置了一个官方的风扇控件，DeskPi可以使用它吗？
NOP, DeskPi 's fan is controlled by a MCU on board , so you just control it via Serial port like : /dev/ttyUSB0  and just send to the port with "pwm_xxx" xxx is digital number, for example : send "pwm_058" means tell the MCU generate 58% PWM signal to drive the fan ,so we called that: fan speed level will be 58%. you can refer to the python code, it will be clearly.

yoyojacky, 2021-02-02:
> @ mbernardi1961您提到“在Raspberry Pi OS中内置了官方风扇控制”，您可以将我链接到该文件的任何文档吗？
> 我已启用对Deskpi的电源和风扇控制器的串行访问，并且可以手动设置5个风扇级别。我见过的大多数其他受控raspi风扇是由GPIO上的PWM控制引脚控制的，并且需要一个晶体管来放大风扇电机的电流。但是这些还有更多的风扇级别步骤。
> 仍然可以看到是否可以更改任何现有软件来控制Deskpi风扇。

Hi Dear MH-010, please refer to my python code, the fan of DeskPi is controlled by MCU onboard ,so just send "pwm_xxx" signal to the serial port will change the behavior of the fan. you can try to use pyserial liberay  to control the fan in python script.

yoyojacky, 2021-02-20:
> Hi, in ubuntu server I am having problems getting the install.sh to work without problems.
>
> This is the output of the install.sh:
>
> ```
> ubuntu@ubuntu    ~/deskpi     master  sudo [...]

please try to use install-ubuntu-mate.sh instead of install.sh script and try again.

closed 2021-02-20 by yoyojacky

camerling, 2021-03-08:
PI 4 / 8go / ubuntu 20.10 /USB Boot

I tried install with username and root: 

linux:~$ systemctl status deskpi.service
● deskpi.service - DeskPi PWM Control Fan Service
     Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Mon 2021-03-08 16:17:05 CET; 34min ago
    Process: 1466 ExecStart=/usr/bin/sudo /usr/bin/pwmFanControl & (code=exited, status=1/FAILURE)
   Main PID: 1466 (code=exited, status=1/FAILURE)

Mar 08 16:17:05 camerlinux systemd[1]: Starting DeskPi PWM Control Fan Service...
Mar 08 16:17:05 camerlinux sudo[1466]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/pwmFanControl &
Mar 08 16:17:05 camerlinux sudo[1466]: pam_unix(sudo:session): session opened for user root by (uid=0)
Mar 08 16:17:05 camerlinux sudo[1466]: pam_unix(sudo:session): session closed for user root
Mar 08 16:17:05 camerlinux sudo[1466]: sudo: unable to execute /usr/bin/pwmFanControl: No such file or directory
Mar 08 16:17:05 camerlinux systemd[1]: deskpi.service: Main process exited, code=exited, status=1/FAILURE
Mar 08 16:17:05 camerlinux systemd[1]: deskpi.service: Failed with result 'exit-code'.
Mar 08 16:17:05 camerlinux systemd[1]: Failed to start DeskPi PWM Control Fan Service.

Any idea ?

yoyojacky, 2021-03-12:
> I noticed the fan is not connected to the GPIO extender board, but to the daughter board that is connected to the pi its power usb-c port ( and hdmi & 3.5 mm jack ) In the scripts they send commands to a serial port ( ser = serial.Serial("/dev/ttyUSB0", 9600, timeout=30) ) .
> So I wonder if they are programming a controller through that USB-C connection, that will set the fan speed by PWM value.
> You can test this by omitting the GPIO riser board installation and setting fan speeds with the supplied OS image and tools.

yes, you can just program it via USB-C　connection and you can find out that is a /dev/ttyUSB0 device file in your system once you put the dtoverlay=dwc2, dr_mode=host to /boot/config.txt file, and you can use PＷＭ to control your fan.

yoyojacky, 2021-03-12:
> PI 4 / 8go / ubuntu 20.10 /USB Boot
>
> I tried install with username and root:
>
> linux:~$ systemctl status deskpi.service
> ● deskpi.service - DeskPi PWM Control Fan Service
> Loaded: loaded [...]

please uninstall it and  refresh the code by : git pull and then install the driver  again.