DeskPi-Team/deskpi #21: List of hardware issues - copy of my mail to Deskpi

DerSpatz, 2020-12-30:
First, let's start with the things that are good:

I like the overall design and concept of the case. It has some neat extra functions like the additional USB ports and the IR receiver. Also, the illuminated power button (two colors!) was a great surprise, as it wasn't visible on the render images on your website. I also like the interior design, how everything comes together in the small case.

But there are some serious problems with the case, as well:

Obviously, USB booting is bugged, and you will probably need a new controller. Jeff Geerling has tested a lot of controllers and USB storage solutions in his blog, so maybe you can use his knowledge to your advantage when selecting the next controller IC.
When googling for a solution for the problem, I also found that sometimes the Raspberry Pi won't boot from USB when there is a powered USB hub connected to it, as cheap USB hubs will feed back 5V to the USB output of the Raspberry Pi, and this stops the Raspi from booting. Maybe the problem with the USB boot is actually related to the USB hub IC you are using, and not to the USB-to-SATA bridge. If this is the case, you could maybe find a solution with the parts you already have in stock, by adding a single MOSFET and a resistor to keep the hub from backfeeding.

If you could post the schematics for the Deskpi Pro on Github, it would be far easier for people to find bugs like these, and to find solutions.

Furthermore, the HDMI output is problematic, sometimes I do not get any video output at all, and when I have video and lift the case, the video disappears.
Also, 4K video shows a faulty image. Both problems could also be caused by the HDMI cable that was delivered with the Deskpi Pro, as I did not test any other cable, but I assume it is more of a problem of the board routing or the microHDMI connectors.

I could not test the IR receiver, yet, but it is pretty far in the back of the case, so I assume reception will not be that good. Maybe you can also find a different position for this part when you redesign the PCBs.

There are also some mechanical problems I'd like to address:

The GPIO board that gives external access to the pins has a slight bend that will probably not affect the function of the case, but is still looking not that good.
Placing the mounting holes just a tiny bit closer to each other will probably remove that bend. Also, it seems like the 5V on the pins is only supplied by the thin FFC cable, which might cause problems with devices that draw (or deliver, like an UPS) a lot more power. Maybe the power pins of the GPIO board could also be powered from the main power source, and you can select with two jumpers where the 5V and 3V3 for the GPIO pins are coming from.

Also, the cutouts in the front and back panel are very tight. Adding 0,25mm on each side would still look good, but it would make assembly a bit easier.

Finally, the fan that was delivered with the cooler is crap. Below 100% it makes some high pitched scratching noise. I already tried to mount the fan with only one bracket in the hopes that it would stop scratching then, but even when the fan and the cooler have very little contact, the noise remains. I will replace the fan with one from Noctua.

Some people call for support for NVMe for the next board version, but I think this is counterproductive, as it would mean to drop support for SATA and 2.5 inch drives. So please stick to SATA, as this is the special thing about your case.

wardnuke, 2020-12-31:
@DerSpatz 

Good points about NOT adding support for NVMe. The gains from just plain M.2 / 2.5 SSD's is more than enough. There aren't huge gaines from supporting NVMe and as you said, we'd lose the ability to do standard SATA / 2.5" drives.

I have had no sound issues from my fan. I have 2 boxes of which I've only assembled the first one. I'm still waiting on my Vilros Pi4 8GB kit to come in, but it seems to be held up by the holiday shipping backlog. As it is, I've been runing the one I built first for almost a week now and I've had no issues. I have the settings for the fan autospeed set at 40/25% 45/50% 60/75% 70/100%. While streaming video with load (the CPU is overclocked to 2Ghz / GPU is at 750Mhz) my unit runs about 55 deg C. I've done some stress testing with Stress and CPU-Burn and the highest the unit got was 57.9 deg C and that was with the 50% fan setting at 60 deg C ( I changed it just to give more margin). I have had the fan at all levels of speeds while playing with the settings to figure out what I'd like (tolerate) for fan noise and have never heard this noise people talk of. Maybe I just got lucky and got a good one. I did notice when assembling my unit that the heatpipes were causing the cooler to contact the case and "bent" them back down, but I don't know if that is what's making mine not make bad noise. I guess I'll find out when I assemble my second box.

I really hope they listen to our suggestions. From other posts and from the communications I've had with them back and forth, I'd say they're pretty engaged and seem to want our business. I'll be happy when the other things like the controller chip / USB issues are fixed with the new PCB they're talking about sending out in March. Hopefully by then they'll have the LibreElec issue figured out so I can use the boxes I bought for what I entended which is as Media Centers.

DerSpatz, 2020-12-31:
I think I will remove the fan from the cooler today to find out what is causing the noise. I will also try to make a recording of the sounds the fan makes.
I'm glad that you got a case that does not have the fan problems, but maybe there's a problem with QC on the manufacturers side, or you need to mount it in a certain way to reduce noise. I will look into it.
I assume the fan needs 5V, or is it a 12V fan?

Regarding the fan control and IR receiver: I've seen a CH340G USB-to-UART IC on the board, and I assume the TX side is used to control the fan controller. Maybe you could use the RX side and connect it to the IR receiver. Of course this would make the software a bit more complicated, as the IR receiver is not connected directly to the GPIO anymore, but it would free up the UART pins on the GPIO, and you could place the IR receiver more freely.

actionschnitzel, 2021-01-03:
I am glad that not only I have these problems. I turned the fan to blow the air away from the ARM (the noises are really bad). That led to better cooling. Even with SSD, I felt like the Pi booted very slowly.
The IR installation was always faulty. I develop software for the Pi and immediately add the fan speed option to my software  because I believe in the guys and think that many will use the Deskpi.

Occasionally the undervoltage symbol appears.
My GeeekPI case with X825 board already needs a 12V connection to function properly. How does the Deskpi do that with a 5V plug?

It seems we will get a replacement in March.(E-mail from their sales department)

rdiezv, 2021-01-03:
I got this email too and I'm very mad. After waiting all these months all I have is a brick. HDMI output and the M2 SSD don't work, so this case is useless. And now I have to wait until ¿March? to get a new version.

pd: and we have to write our issues in Github because they don't answer emails. shitty experience.

ralieghnet, 2021-01-05:
Here is what I have found. 

The USB controller for the daughter board is not usable for USB boot. 

The mounting post next to the power button needs to be unscrewed in order for the unit to have video out. With the post securely mounted it tweaks the daughter board enough to loose contact with the HDMI ports. 

The acrylic panel on the rear (eth side) is not cut correctly. The hole for the usb 2.0 is being pressed down witch also causes the HDMI to loose contact.

Last, the clearance between the fan and the case is zero. This places unnecessary force on the chip set. 

I did manage to get Ubuntu 20.10 to boot from a SSD, but the SSD is located outside the case.

DerSpatz, 2021-01-05:
I will try the fix for HDMI in the next days, thank you!

When you say "USB controller", do you mean the USB hub IC or the USB-to-SATA bridge? I ask because my suspicion is that the USB hub is the main problem, not the USB-to-SATA bridge...

ralieghnet, 2021-01-05:
I would have to say it is the USB hub IC. I tried a simple mouse/keyboard on one of the front port to check functionality. It to was next to non responsive. When I plugged the same mouse/keyboard into a PI port it function correctly.

ppareit, 2021-01-09:
@DerSpatz I confirm your positive points. I also like the case and how everything is structured.

I would like to add that the soldering on the board is not good. The headset plug on the back came lose in the box. While installing everything, the sd card mount on the front came lose. I hope that the next batch also fixes the quality.

jasonthomashorn, 2021-01-23:
@ralieghnet how did you get 20.10 working?
I concure with the hdmi clearance issues.
I am running with the back and front face plate off.
And using a cheap iotech usb3 to sata to boot from m.2. like others ssd on board would lock up.
But fan runs great no noise will try flipping it for better flow

ralieghnet, 2021-01-23:
@jasonthomashorn When I first setup the external drive using the raspberrypi imager. I could not get it working. This was mainly due to the HDMI output not passing the video. As I found out the image was working I just didn't realize until a little later. 

So what I did was use a second RPi4 to setup the external drive then I moved it back to the desired RPi4. I mainly use the RPi4 in a headless state so video is not much use other than the initial setup.

wmichel0706, 2021-02-03:
I've two cases and they behave different but they also have common - severe - hardware issues:
1. SD cards: Raspi often cannot boot from sd card. I get strange messages like unexpected interrupts, data corruption messages or the boot process just hangs. I didn't figure out so far what the issue is but it looks like a temperature problem. System cold -> problems; System warm -> better and/or no problems. When I remove the sd card extender and put the card directly into the raspberry 4 slot everything works fine in all cases!
2. USB-drive: With one case I'm using a SSD drive to boot. Everything works fine besides the fact that I can't use the USB_to_USB dongle -> data corruption (also with the dongle of the other case). Using instead a 1m USB cable solves the issue.
3. HDMI outputs work fine
4. IR receiver: not checked so far
5. GPIO-extender: disconnected -> data corruption problems

I spend many hours to find a robust and reliable configuration but was not successful. (I changed raspi's, power supplies, sd-cards, ssd's, usb drives, cables, etc.)

Current situation: The case looks great and solid but it doesn't work reliably (at least for all temperatures)

yoyojacky, 2021-02-20:
> @DerSpatz
>
> Good points about NOT adding support for NVMe. The gains from just plain M.2 / 2.5 SSD's is more than enough. There aren't huge gaines from supporting NVMe and as you said, we'd lose the [...]

Thank you very much for your trust in our products. I just got news that Liberelec nightly builds can support deskpi. It's still under test, and you will be notified as soon as there is news.

yoyojacky, 2021-02-20:
> I think I will remove the fan from the cooler today to find out what is causing the noise. I will also try to make a recording of the sounds the fan makes

It does not work as you expect. In fact, the control of the fan is executed on the daughterboard through a single-chip microcomputer, while the Raspberry Pi recognizes a serial device through the function of the dwc of the USB-C dtoverlay. Pi sends the "PWMXXX" signal to the serial device, where XXX is the PWM value from 000 to 100 to notify the MCU to control the fan speed, and the DESKPI driver is adjusted by reading the current temperature of the CPU, you can also customize The mapping relationship between speed and temperature is through the deskpi-config command.

yoyojacky, 2021-02-20:
> I think I will remove the fan from the cooler today to find out what is causing the noise. I will also try to make a recording of the sounds the fan makes.
> I'm glad that you got a case that does not [...]

It does not work as you expect. In fact, the control of the fan is executed on the daughterboard through a single-chip microcomputer, while the Raspberry Pi recognizes a serial device through the function of the dwc of the USB-C dtoverlay. Pi sends the "PWMXXX" signal to the serial device, where XXX is the PWM value from 000 to 100 to notify the MCU to control the fan speed, and the DESKPI driver is adjusted by reading the current temperature of the CPU, you can also customize The mapping relationship between speed and temperature is through the deskpi-config command.

yoyojacky, 2021-02-20:
> mean the USB hub IC or the USB-to-SATA bridge? I ask because my suspicion is that the USB hub is the main problem, not the USB-to-SATA bridge...

the main problem is the chip of SSD driver, we have already fixed up this issue and reproduced the product during those days..

yoyojacky, 2021-02-20:
> I would have to say it is the USB hub IC. I tried a simple mouse/keyboard on one of the front port to check functionality. It to was next to non responsive. When I plugged the same mouse/keyboard into a PI port it function correctly.

1. what kind of OS did you use? 
2. Could you please show me the /boot/config.txt file if you are using Raspberry Pi OS?

yoyojacky, 2021-02-20:
> I would like to add that the soldering on the board is not good. The headset plug on the back came lose in the box. While installing everything, the sd card mount on the front came lose. I hope that the next batch also fixes the quality.

You mean the 3.5mm headphone jack is off? And the SD card slot is also broken?  If so, please contact sales@deskpi.com to provide photos or videos and request hardware replacement,

yoyojacky, 2021-02-20:
> @ralieghnet how did you get 20.10 working?
> I concure with the hdmi clearance issues.
> I am running with the back and front face plate off.
> And using a cheap iotech usb3 to sata to boot from m.2. like others ssd on board would lock up.
> But fan runs great no noise will try flipping it for better flow

please contact sales@deskpi.com to provide photos or videos and request hardware replacement,

yoyojacky, 2021-02-20:
> When I first setup the external drive using the raspberrypi imager. I could not get it working. This was mainly due to the HDMI output not passing the video. As I found out the image was working I [...]

Your HDMI output is not normal? If this is the case, please contact sales@deskpi.com and provide a test video, and you will be replaced with a new board with a fixed bug.

yoyojacky, 2021-02-20:
> I've two cases and they behave different but they also have common - severe - hardware issues:
>
> 1. SD cards: Raspi often cannot boot from sd card. I get strange messages like unexpected interrupts, [...]

Raspi usually cannot be booted from the SD card? Can you try to boot directly from the Raspberry Pi without connecting the deskpi shell to test if the system is correct. If the problem still exists, please check whether the SD card adapter board is inserted properly and locked, and check whether the TF card extension cable FPC cable is installed in place.
If the GPIO expansion board is not connected, it should not cause data damage. Please provide the installation icon and test video to sales@deskpi.com. After verification, we will provide a replacement device. We are very sorry for the inconvenience caused.

closed 2021-02-20 by yoyojacky

wmichel0706, 2021-02-20:
@yoyojacky Thx for reply. I spent more or less two full days (a couple of weeks ago) to check all possible combinations, connections and to permute different sd cards, sata devices (SSD's, HDD's) and the behaviour was pretty much always the same:

1. It looks like there is a temperature dependence for correct boot from sd-card: Warm system -> boot; Cold system -> no boot: a lot of strange messages like "unexpected interrupts...; corrupted memory blocks;... etc.). In all this cases the raspberry booted without any issues when I plugged the SD card directly in the pcb slot of the raspberry.

2. If I manage to boot the system I can also install the OS on SSD and/HDD with the restriction that one of my two USB dongles generates datacorruption. If I use a 1m cable instead of the dongle everything is fine.

Unfortunately I don't have videos and/or more detailled information. It took me so many hours to bring at least one case to the point of functionality where I can use now that I don't want to touch it again and/or invest more time.