DeskPi-Team/deskpi #24: Install Script to control fan, front usb & power for Official OSes

mbernardi1961, 2020-12-31:
Given that the Raspberry Pi OS, Ubuntu 20.10 Desktop, and LibreElec
 are all Officially supported by The Raspberry Pi Foundation, it would seem that these should be supported at minimal, rather then other unsupported distros.

I want to use Ubuntu 20.10 Desktop as my default internal SSD OS but if I can't use all the features there is no point.

M-H-010, 2021-01-01:
I second your idea to expand support for the known distro's for the RPI4. However I do think the Deskpi team should be the one responsible to come up with all the solutions. We can make this possibe much faster , when we combine our knowedge on Linux distro's, with the knowledge of the design of the case and its electronics.
I myself have started to get the fan to work with Ubuntu 20.04.1 LTS, as I have used the case to get a proper mount of the pi , an external HDD and powersupply as a home server in a cabinet. Although I do not need the fan to spin now, it would be good to be able to control it. The other parts that need to have  software support are the IR reader and the USB devices connected to the regular usb3 port. 
If we understand the connection an techniques used we can find what is needed to get the parts working for Ubuntu too.
On the fan controller part, it seems connected via /dev/ttyUSB0 which is a usb to serial device connected from the USB-c 'power' connector. However I have not jet found a way to recognize it and install the serial port /dev/  from Ubuntu.
The HDD is recognized automatically, and did not need any configuration for me. Though I boot from the sd-card and still need to upgrade the HDD, so will not move to USB boot till then.
The IR receiver seems a basic vishay as used in f.i. the adafruit explinations: https://learn.adafruit.com/using-an-ir-remote-with-a-raspberry-pi-media-center

Please share your efforts and progress.
M-H

M-H-010, 2021-01-02:
for Ubuntu 20.04.1 LTS
you need to enable the dwc2 kerneldriver to access the serial port on the daughterboard.
by adding the line:
`dtoverlay=dwc2,dr_mode=host`
to the file:
`/boot/firmware/usercfg.txt`

Please note there are no spaces around the comma in the line with the dwc2 driver.
Also we do not alter the config.txt or sysconfg.txt to maintain system integrity

yoyojacky, 2021-02-02:
> 对于Ubuntu 20.04.1 LTS，
> 您需要启用dwc2内核驱动程序以访问子板上的串行端口。
> 通过
> `dtoverlay=dwc2,dr_mode=host`
> 在文件中添加以下行：
> `/boot/firmware/usercfg.txt`
> 
> 请注意，在dwc2驱动程序的行中，逗号周围没有空格。
> 另外，我们不会更改config.txt或sysconfg.txt来维护系统完整性

Thanks very much for your help!!!  It works.

yoyojacky, 2021-02-20:
Thank you very much for your suggestions and comments. We are also trying to support other systems other than the Raspberry Pi OS, but it takes time. Different systems deal with serial ports in different ways, which makes our installation scripts take more time to test and debugging.
Since we are using the function provided by a dtoverlay of USB-C in the Raspberry Pi hardware, the part of dwc, we are not sure whether the Raspberry Pi hardware can also be seamlessly supported in ubuntu20.10, and the system and raspberry Pi OS There are still some differences between them. If we want to control the fan, we can send a similar signal to the onboard MCU through the serial port identified by the system: "pwmXXX". It can debug pwm linearly with a baud rate of 9600. The XXX is a number from 000 to 100, which means pwm information. You can customize settings according to the source code provided by github.

closed 2021-02-20 by yoyojacky