DeskPi-Team/deskpi #25: Findings about fan noise

DerSpatz, 2020-12-31:
As I mentioned in the other thread, my fan is rather noisy. It makes scratching noises on low levels (unexpected) and wind noises on high levels (expected). I try to concentrate on the low level noise in this issue, as the wind noise is natural for fans of this size.

To find out what's going on, I tried different situations, and want to share my findings here:

- The fan PWM is probably controlled by U8, that controls a MOSFET (Q5). Unfortunately, U8 is not labeled, so I can't find out what it is and how it works
- My fan makes a high-pitched, scratchy noise at PWM levels 25%, 50% and 75%. I can't tell if this noise is still there at 100%, as the fan is quite loud then, but I think I can't hear the scratchy noise then anymore. A quick and dirty measurement with a spectrometer on my smartphone shows a peak at 4800 Hz, which strangely correlates with the 9600 baud rate used in the program to control the PWM.

**ASSUMPTION:** The high pitched noise is caused by PWM. By adjusting the PWM frequency, the noise could be pushed into an inaudible range. As I can't tell what the PWM controller IC is, I can't tell if the IC is able to provide sufficiently high PWM frequencies (>20kHz). Also the fan might not support such high frequencies.
(The correlation between the fan noise frequency and the UART baud rate let's me assume that the PWM signal is directly connected to the UART signal, so maybe changing the UART baud rate might improve the noise)

- The fan is less loud when the module is placed outside of the metal case; the scratchy noise is less audible, but still there
- After pushing the module back into the case, the fan was much louder, the noise was amplified by the case
- In the third try, I tried to adjust the cooler as good as possible so that it does not touch the case, which is not entirely possible, but it still helped a bit

Now I tried to attach the 40x40x**10**mm Noctua fan, only to find out that it does not fit. The original fan is even thinner than 10mm, and I am too lazy to get a ruler right now...

Next, I added a small rubber band between fan and cooler, so that the fan does not directly touch the cooler. This also helped a lot, but the fan is still audible on low levels.

**FINDINGS:** The PWM sound is amplified by the case. This might be, because the case's shape and material causes a resonant frequency that is in the audible range and is excited by the fan/cooler combo. This could be reduced by gluing damping material to the case (which I don't have) or by reducing the contact between case and cooler, for example by adding two rubber rings like in this image (when you put them in the right position, they are almost invisible from the outside):

[image]
[image]

At last, I will try to mount the Noctua fan directly onto the case with two screws, and add some small passive heatsinks directly on the ICs on the Raspberry Pi. After that, I will compare the two solutions (ICE cooler with rubber bands and Noctua fan with passive heatsinks) regarding temps and volume/noise. By holding the Noctua fan onto the case, I already found out that the slots in the case also amplify fan wind noise, but this fortunately only happens at 100% PWM.

**CONCLUSION:** Damping the contact between fan and cooler and cooler and case reduces fan noises by a lot and is a quick and easy fix for those complaining about fan noise.
Manufacturing-wise, chamfering the slots on the top of the case might help with wind noises at high PWM levels, but also adds manufacturing costs to the case. Also adding mounting holes for a 40mm fan directly to the case would be a cool thing in case the other cooling solution works well.

**TODO:** Edit the baud rate in the code and see if and how it affects the PWM rate. Unfortunately, I am not code savvy enough, and the function of U8 is still unclear. I assume the worst case with adjusted baud rates would be U8 not working properly anymore, but I can't tell...

CrunchyDoodle, 2021-01-01:
On my system, the fan is generally inaudible at any speed or system temperature. Even manually setting for full speed, it's pleasantly quiet. Swapping in another fan might be the right fix for you.

wardnuke, 2021-01-05:
@DerSpatz 

I assembled my second DeskPi Pro box yesterday and had no fan noise issues. I ran the fan control at various levels up to 100% and I did not experience any noise issues. At 100% the fan is actually not too bad. I took care when assembling the fan / heatsink and inserting the assembled boards into the case so the heatsink didn't get "bent" like the first case I assembled, but it actually went together pretty well and I have no noises. Looks like perhaps you have a bad fan.

solskogen, 2021-01-21:
I've got the same problem as @DerSpatz

bpicon, 2021-01-21:
I also have the problem of fan noise. It's like a rubbing or scraping sound at low revs (mostly 25%). It looks like a manufacturing defect.
I requested a replacement, as there are more buyers who are not affected, but the seller refused, claiming that this sound is normal.

yoyojacky, 2021-02-02:
> * The fan PWM is probably controlled by U8, that controls a MOSFET (Q5). Unfortunately, U8 is not labeled, so I can't find out what it is and how it works

The whole fan works in this way. The system sends "pwm_xxx" through the serial port, where xxx is the number 000 ~ 100 to the onboard MCU. The MCU parses the PWM signal and then drives the fan to rotate. You can refer to the content provided in the python script to write the control script yourself. Under normal circumstances, I recommend setting the temperature at 55 degrees to set 100% speed, so that the temperature can be controlled while also reducing noise. Low-frequency pwm does There is noise, but the whole problem can be solved by changing the pwm information. When the fan rotates at high speed, there is basically no noise.

yoyojacky, 2021-02-02:
> I also have the problem of fan noise. It's like a rubbing or scraping sound at low revs (mostly 25%). It looks like a manufacturing defect.
> I requested a replacement, as there are more buyers who are not affected, but the seller refused, claiming that this sound is normal.

The whole fan works in this way. The system sends "pwm_xxx" through the serial port, where xxx is the number 000 ~ 100 to the onboard MCU. The MCU parses the PWM signal and then drives the fan to rotate. You can refer to the content provided in the python script to write the control script yourself. Under normal circumstances, I recommend setting the temperature at 55 degrees to set 100% speed, so that the temperature can be controlled while also reducing noise. Low-frequency pwm does There is noise, but the whole problem can be solved by changing the pwm information. When the fan rotates at high speed, there is basically no noise.

yoyojacky, 2021-02-20:
> which I don't have) or by reducing the contact between case and cooler, for example by adding two rubber rings like in this image (when you put them in the right position, they are almost invisible from the outside):

contact sale@deskpi.com and change the fan.

yoyojacky, 2021-02-20:
> I also have the problem of fan noise. It's like a rubbing or scraping sound at low revs (mostly 25%). It looks like a manufacturing defect.
> I requested a replacement, as there are more buyers who are not affected, but the seller refused, claiming that this sound is normal.

There will be a small amount of lubricating oil inside the fan during production. Please adjust and run at 100% speed for a few minutes. If the noise persists, please slightly press the fan blade, because it may loosen or scratch the heat sink during transportation. All fans are packaged after testing. Please rest assured of quality.

closed 2021-02-20 by yoyojacky

ivanmarban, 2021-07-23:
What are the specs of the original fan? I would like to replace it because of that noise and would be nice to kown the specs to try to find a replacemente that fits.