DeskPi-Team/deskpi #26: deskpi.service cannot be installed under Volumio (Raspbian)

orsib78, 2021-01-04:
**I tried to install the DeskPI Pro driver under Volumio 2.861, which is technically a Raspbian based music player (Raspbian GNU/Linux 8 (jessie)**

**During installation I've got the following output:**

[info] DeskPi Fan control script installation Start..
[info] Enable dwc2 on Host Mode.
[info] dwc2 has been setting up successfully.
[info] DeskPi main control service loaded..
[info] DeskPi Service configuration finished..
[info] DeskPi Service Load module..
Failed to start deskpi.service: Unit deskpi.service failed to load: Invalid argument. See system logs and 'systemctl status deskpi.service' for details.
[ ok ] DeskPi PWM Fan Control and Safeshut Service installed successfully..
[info] System will reboot in 5 seconds to take effect..

**Using systemctl, the following error message displayed:**

Jan 04 12:39:40 volumio systemd[1]: [/lib/systemd/system/deskpi.service:7] Executable path is not absolute, ignoring: sudo /usr/bin/pwmFanControl &
Jan 04 12:39:40 volumio systemd[1]: deskpi.service lacks both ExecStart= and ExecStop= setting. Refusing.

**How could I circumvent this issue?**

yoyojacky, 2021-02-02:
> **I tried to install the DeskPI Pro driver under Volumio 2.861, which is technically a Raspbian based music player (Raspbian GNU/Linux 8 (jessie)**
>
> **During installation I've got the following [...]

Maybe the version of the entire system is a bit old, because the environment we tested is the version of raspberry pi OS (buster), can you provide us with the content of lsb_release -a? I find a time to test it.

orsib78, 2021-02-02:
Hello.

**This is the output of lsb_release:**
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie

**This is the output of uname -a:**
Linux volumio 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux

Thanks,

yoyojacky, 2021-02-20:
> Hello.
> 
> **This is the output of lsb_release:**
> No LSB modules are available.
> Distributor ID: Raspbian
> Description: Raspbian GNU/Linux 8.0 (jessie)
> Release: 8.0
> Codename: jessie
> 
> **This is the output of uname -a:**
> Linux volumio 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux
> 
> You can easily test it by flashing the latest version of volumio to an SD-CARD. This is exactly the same version I am currently using (2.861).

OK, we are working on it.

closed 2021-02-20 by yoyojacky