DeskPi-Team/deskpi #27: Please Support Ubuntu 20.10

VsmashJ, 2021-01-18:
64bit os for rpi4 64bit

NuLogicSystems, 2021-01-19:
I had the same issue with Manjaro, I had to delete the drivers and recompile them myself prior to installing them.
I also had to install everything manually and create the systemd services myself as the installer script fails on Manjaro. I think this is due to Manjaro not supporting /lib/lsb/init-functions in any, way, shape, or form.

yoyojacky, 2021-02-02:
> 我在Manjaro（也是64位）上遇到了同样的问题，我不得不从/ deskpi / drivers / c /中删除驱动程序二进制文件，并在安装之前自己重新编译它们。我还必须手动安装所有内容，将dtoverlay = dwc2，dr_mode = host添加
> 到config.txt，然后自己创建两个systemd服务，因为安装脚本在Manjaro上失败。我认为这是由于Manjaro不以任何方式，形状或形式支持/ lib / lsb / init-functions所致。

Oh, thanks for your feedback, i am going to modify the installation scripts.

closed 2021-02-20 by yoyojacky