DeskPi-Team/deskpi #28: Ice tower cooler mount measurements are ill-fitting; mounting pillars are too short.

Kuratius, 2021-01-28:
Installing the ice tower provided with the desk pi pro requires signficant in order to mount the two back brackets, such that they bend a lot and add stress to the PCB.
I've already seen reports from other people about damaging their raspberry pi due to sloppy manufacturing tolerances in the desk pi pro & the low profile ice tower cooler that they sell separately, which is what the desk pi ice cooler is based on, uses significantly thicker brackets and longer mounting pillars so that it doesn't have this issue. For example, it uses 6mm pillars instead of 5mm, etc.

craldaz, 2021-02-01:
This was also a problem in my device.

yoyojacky, 2021-02-02:
> Installing the ice tower provided with the desk pi pro requires signficant in order to mount the two back brackets, such that they bend a lot and add stress to the PCB.
> I've already seen reports [...]

> Installing the ice tower provided with the desk pi pro requires signficant in order to mount the two back brackets, such that they bend a lot and add stress to the PCB.
> I've already seen reports [...]

At that time, the length of the entire screw was tested and calculated, and it was just tightened, otherwise the internal height of the copper pillar on the bracket would not be enough to lock the screw.

closed 2021-02-02 by yoyojacky

Kuratius, 2021-02-02:
> At that time, the length of the entire screw was tested and calculated, and it was just tightened, otherwise the internal height of the copper pillar on the bracket would not be enough to lock the [...]

Are you saying you the copper pillars on the bracket are the wrong size? That would be easy to fix.

johntdavis84, 2021-08-23:
I ran into this, too. 

I didn't damage the Pi, as far as I can tell, but there was a quite substantial gap between the copper pieces on the cooler mount and the Pi itself.

I had to pretty much bend them into place.

I've installed a standard ice tower cooler on another Raspberry Pi 4b, and this was not an issue.

YourFriendCaspian, 2021-12-08:
This was also a problem in my device.

I didn't damage the Pi, as far as I can tell, but there was a quite substantial gap between the copper pieces on the cooler mount and the Pi itself.

yoyojacky, 2022-03-02:
> This was also a problem in my device.
> 
> I didn't damage the Pi, as far as I can tell, but there was a quite substantial gap between the copper pieces on the cooler mount and the Pi itself.

did you solve this problem?