DeskPi-Team/deskpi #29: SSD is extremely slow

mgrouch, 2021-02-08:
I’ve followed all set up steps per deskpi pro instructions. However when I try to boot from SSD it is unusably slow.  Way way slower than SD card. 
thx

wardnuke, 2021-02-09:
@mgrouch 

You need to look at some of the other Issues here. There are ways to pick up the speed for now. There's an issue with the controller chip (it's incompatable with Raspberry) so the manufacturer is making new controller boards and sending them out sometime after March. There's a setting (USB Quirks) you can put into a config file that will boost the speed a little. It'll be about 30% of the peak performance of a M.2 SSD, but it's still 2X faster than the MicroSD card boot times.

Look at ISSUES WITH USB BOOT TO M.2 SSD #16 that I put up in December. It'll show my issues with SSD and the test results I had after using the USB Quirks command.

You'll find that fix in the Closed Issues as Workaround for slow SSD boot #7.

Make sure you look at the settings for the EEPROM too. That's near the end of my Issue #16 post.

Hope this helps.

mgrouch, 2021-02-09:
I’ve run install.sh from deskpi.git 
as per instructions. Shouldn’t those hot fixes be included into it while you are working on correct one?

Thank

mgrouch, 2021-02-09:
I’ve put the quirks and they did nothing. Absolutely the same issue. 
I’m ready to try other ideas though. 
Thanks

wardnuke, 2021-02-09:
I’ll pull up my config file and send it to you. I remember there was an issue I had to figure out also.

mgrouch, 2021-02-09:
This solution actually works for me! 
I did couple of mistakes previously while testing it.

1. I've edited /boot/config.txt instead of /boot/cmdline.txt
2. I've put required change on a separate line in  /boot/cmdline.txt. It has to single line

Not full speed but at least it's useable

Thanks a lot

wardnuke, 2021-02-10:
@mgrouch 

Yeah, it's not as fast as it could be. I was able to run mine at full speed initially and got some data on speed. I then ran the USB Quirks command and got some speed data from that. It was about 30% of what it did at full speed. I can't wait for the new boards to come to see the full potential for this thing. The only thing I'll need then is for them to figure out how to install and run LibreElec so I can run this thing like a media center. As it is, I'm runing Kodi from the Raspberry OS Desktop.

mgrouch, 2021-02-10:
I’ve built a new distro for my boat. It has music player. Mopidy. And other goodies. You can try installing it. Image is here. 

https://cloudsmith.io/~bbn-projects/repos/bbn-repo/packages/?q=lysmarine

Some documentation here

https://github.com/bareboat-necessities/lysmarine_gen

yoyojacky, 2021-02-20:
> as per instructions. Shouldn’t those hot fixes be included into it while you are working on correct one?

sorry for that, we are changed the chip of the SSD driver in the latest version of DeskPi Pro.

closed 2021-02-20 by yoyojacky