DeskPi-Team/deskpi #30: Problem using pwmFanControl in Ubuntu server 20.04

diego-G, 2021-02-08:
The service doesn't get executed due to the next problem:

`sudo: unable to execute /usr/bin/pwmFanControl: No such file or directory`

I tried using the script for Ubuntu Mate but it doesn't work well. I had to execute the commands by hand. Therefore, I imagine the scripts should be prepare for Ubuntu 20.04.

yoyojacky, 2021-02-20:
sudo cp -rf $installationfolder/drivers/c/pwmFanControl /usr/bin/pwmFanControl
sudo cp -rf $installationfolder/drivers/c/fanStop  /usr/bin/fanStop 
please copy the binary file from the installation folder. and give them execute permissions.

closed 2021-02-20 by yoyojacky

diego-G, 2021-02-20:
I had to recompile the executables before compiling them. Thanks anyways!