DeskPi-Team/deskpi #32: deskpi.service as distibuted never allows startup to finish

iainaff, 2021-02-15:
deskpi.service service file never releases - this mean the multi-user.target startup is never completed.

Patch below:

--- /lib/systemd/system/deskpi.service~	2021-02-15 12:21:13.841558454 +0000
+++ /lib/systemd/system/deskpi.service	2021-02-15 12:47:53.408371617 +0000
@@ -2,8 +2,8 @@
 Description=DeskPi PWM Control Fan Service
 After=multi-user.target
 [Service]
-Type=oneshot
-RemainAfterExit=true
-ExecStart=sudo /usr/bin/pwmFanControl &
+Type=simple
+RemainAfterExit=no
+ExecStart=sudo /usr/bin/pwmFanControl 
 [Install]
 WantedBy=multi-user.target

yoyojacky, 2021-02-20:
> atch below:

But if remove :
"-Type=oneshot
-RemainAfterExit=true
-ExecStart=sudo /usr/bin/pwmFanControl &" 
The standalone program will not monitor the temperature of CPU.

closed 2021-02-20 by yoyojacky

iainaff, 2021-02-20:
You might not like the fix -- but the issue still exists.

The service script, as distributed, blocks completion of startup:

iainf@hopper:~$ uptime
 17:31:42 up 4 min,  1 user,  load average: 0.34, 0.24, 0.11
iainf@hopper:~$ systemd-analyze
Bootup is not yet finished (org.freedesktop.systemd1.Manager.FinishTimestampMonotonic=0).
Please try again later.
Hint: Use 'systemctl list-jobs' to see active jobs
iainf@hopper:~$ systemctl list-jobs
JOB UNIT           TYPE  STATE  
136 deskpi.service start running

1 jobs listed.
iainf@hopper:~$

iainaff, 2021-02-20:
And, using the patched script, after reboot:

iainf@hopper:~$ ps aux | grep pwmFanControl
root         944  0.0  0.0  14464  4256 ?        Ss   17:37   0:00 /usr/bin/sudo /usr/bin/pwmFanControl
root         947  0.0  0.0   1960   576 ?        S    17:37   0:00 /usr/bin/pwmFanControl
iainf       1141  0.0  0.0   8608   824 pts/0    S+   17:38   0:00 grep --color=auto pwmFanControl
iainf@hopper:~$ uptime
 17:39:12 up 1 min,  1 user,  load average: 0.44, 0.23, 0.09
iainf@hopper:~$ systemd-analyze
Startup finished in 4.483s (kernel) + 16.107s (userspace) = 20.591s 
graphical.target reached after 16.079s in userspace
iainf@hopper:~$ systemctl list-jobs
No jobs running.
iainf@hopper:~$ ps aux | grep pwmFanControl
root         944  0.0  0.0  14464  4256 ?        Ss   17:37   0:00 /usr/bin/sudo /usr/bin/pwmFanControl
root         947  0.0  0.0   1960   576 ?        S    17:37   0:00 /usr/bin/pwmFanControl
iainf       1148  0.0  0.0   8608   824 pts/0    S+   17:39   0:00 grep --color=auto pwmFanControl
iainf@hopper:~$ 

The standalone program is running and monitoring fan control.