DeskPi-Team/deskpi #33: DeskPi Pro - SSD is not recognized

starmagoo, 2021-02-16:
Hello, I have purchased the DeskPi Pro and assembled it without any problems. Unfortunately, the add-on board (PCB) for the SATA SSDs does not recognize the SSDs. Therefore, I can not boot from SSD or USB. Raspberry Pi OS does not display the SSD so I assume a hardware defect.

How to solve the issue?

Greetings

Mister-42, 2021-02-16:
Step 1) Read previous issues
Step 2) Wait for the middle of march 2021

More information [here](https://deskpi.com/products/deskpi-pro-for-raspberry-pi-4).

starmagoo, 2021-02-16:
Thanks for the Hint. I ordered at Seeedstudio. So Seeedstudio will contact me?

closed 2021-02-20 by yoyojacky

futurejones, 2021-02-20:
@yoyojacky You have closed this without answering the question about SeeedStudio.
What do the people who purchased from SeedStudio do??