DeskPi-Team/deskpi #35: deskpi.service as distributed never allows Bootup to finish

iainaff, 2021-02-20:
deskpi.service script blocks completion of Bootup:

iainf@hopper:~$ uptime
 17:34:06 up 6 min,  1 user,  load average: 0.21, 0.21, 0.10
iainf@hopper:~$ systemd-analyze
Bootup is not yet finished (org.freedesktop.systemd1.Manager.FinishTimestampMonotonic=0).
Please try again later.
Hint: Use 'systemctl list-jobs' to see active jobs
iainf@hopper:~$ systemctl list-jobs
JOB UNIT           TYPE  STATE  
136 deskpi.service start running

1 jobs listed.
iainf@hopper:~$

yoyojacky, 2021-02-26:
Could you please tell me what kind of OS did you use?

yoyojacky, 2021-02-26:
ok, i have already modified the code, and please refresh you browse and try again.

closed 2021-02-26 by yoyojacky