DeskPi-Team/deskpi #36: USB port are now dead !?

constantdupuis, 2021-03-01:
Hi,

I installed my Raspberry pi 4 in the DeskPi Pro enclosure, not using SSD/M.2 HDD yet.
But my USB port where not working properlly, sometime front ones where working, soetime back one where working, sometime USB2 but not USB3 or other way around.

So I remove my PI from the enclosure, and now none of the USB ports are working any more ?

Please advise,

Constant

yoyojacky, 2021-03-12:
you mean front USB works sometime and sometimes out of work? or never work properly ? if so, you'd better to contact sales@deskpi.com and we are going to reship a daughter board to you.

constantdupuis, 2021-03-12:
> properly ? if so, you'd better to contact ***@***.*** and we are

Hi,

Now, none of them are working any more. I removed the PI from the case, and
the USB ports are not working anymore, look's like the daughter killed my
USB ports :-(
I try with a fresh install of Rasp Os on another SD card, same issue ....

I'm a bit pissed-off, I will try to send mail to sales.

Thanks

yoyojacky, 2021-03-17:
> Hi, Now, none of them are working any more. I removed the PI from the case, and the USB ports are not working anymore, look's like the daughter killed my USB ports :-( I try with a fresh install of [...]

The power supply of the USB interface on the Raspberry Pi is not enough to support the simultaneous use of your HDD and SSD. If you use both at the same time, your USB control chip may be burned....

G6EJD, 2021-05-03:
I have two off the new cases, both with updated (I believe) boards, the fan works normally when no USB devices are connected, but when I plug-in a keyboard instead of using the RPi headless the new USB device gets allocated to USB0 and the fan control is disabled.
I cannot find a way off ensuring that new USB devices use USB1 upwards to prevent this, do you know how to solve it  please>

yoyojacky, 2021-05-11:
> I haven't used any HDD yet. I was just trying my Pi 4 in the DeskPi Case with only the SDCard and a wireless keyboard/mouse combo and a screen. Le mer. 17 mars 2021 à 16:15, yoyojacky ***@***.***> a écrit :
> […](#)

you mean the raspberry pi's usb port are out of working ?

yoyojacky, 2021-05-11:
> I had the issue with the first board, [sales@deskpi.com](mailto:sales@deskpi.com) is helpy my to sort it out, but so far, USB port are still dead.
> :-( I received the new board (very fast by the way), but I'm afraid to use it, and kill another Rasp 4 USB ports...

It will not kill your usb port due to we have already test it for several times. if so, please check if there is something wrong when you assemble it, i am not sure, but i have never met that before.

Pi-Mania, 2021-06-17:
> Hi,
>
> I installed my Raspberry pi 4 in the DeskPi Pro enclosure, not using SSD/M.2 HDD yet.
> But my USB port where not working properlly, sometime front ones where working, soetime back one where [...]

Hi.

If you are booting from SDcard ONLY then you will probably need to edit the **config.txt** file & add : **dtoverlay=dwc,dr_mode=host** to that file.  I had the same issue running O/S from SDcard but this resolved for me. hope it helps others. Drew / Pi-Mania

yoyojacky, 2021-06-30:
it should be :    `dtoverlay=dwc2,dr_mode=host`  but not `dtoverlay=dwc,dr_mode=host` , you've lost a `2`  ....
do remember reboot your pi after configuring.

closed 2021-06-30 by yoyojacky

njc009, 2024-02-11:
I know this is an old thread but I thought I'd post it for anyone else wondering. I'm currently using a Raspberry Pi 4 Model B with a DeskPi runnning RaspiOS version "30f0c5e4d076da3ab4f341d88e7d505760b93ad7" and had no response from the front USB ports. [I found an article from DeskPi](https://deskpi.com/blogs/learn/trouble-shooting-how-to-check-front-usb-port) pertaining to a command to update with those functions. In this case, the proper command to get these working is to use `sudo sed -i '$a\dtoverlay=dwc2,dr_mode=host' /boot/firmware/config.txt` as they've moved the boot file to `/boot/firmware/config.txt`

yoyojacky, 2024-02-26:
> d a way off e

In a Linux operating system on a Raspberry Pi, you can use udev rules to assign a fixed device name to a USB device based on its properties. Follow these steps to achieve this:

Identify the USB device attributes:
Connect the USB device and use the lsusb command to list the connected USB devices. Note down the Vendor ID and Product ID of the USB device you want to assign a fixed name.

```bash 
lsusb
```
* Create a udev rule:
Create a new udev rule by creating a file in the /etc/udev/rules.d/ directory. You can use a text editor like nano or vi. Replace VENDOR_ID and PRODUCT_ID with the actual values from step 1.

```bash 
sudo nano /etc/udev/rules.d/99-usb-serial.rules
```
Add a line similar to the following, replacing VENDOR_ID and PRODUCT_ID with the actual values:

```bash
SUBSYSTEM=="tty", ATTRS{idVendor}=="VENDOR_ID", ATTRS{idProduct}=="PRODUCT_ID", SYMLINK+="my_usb_device"
```
Save the file and exit the text editor.

Reload udev rules:
After creating the udev rule, you need to reload the udev rules to apply the changes.

```bash
sudo udevadm control --reload-rules
```
Reconnect the USB device:
Disconnect and reconnect the USB device. After reconnecting, you should be able to access the device using the specified symlink. In the example above, it would be /dev/my_usb_device.

Now, regardless of the order in which USB devices are connected, the specified USB device should have a fixed symlink. Adjust the rule as needed based on your specific device's attributes.

sampatteson, 2024-04-06:
> sudo sed -i '$a\dtoverlay=dwc2,dr_mode=host' /boot/firmware/config.txt

My man! That was the problem. Makes you wonder why deskpi hasn't updated their troubleshooting page. Does this company even QA their product?

yoyojacky, 2024-04-09:
> > sudo sed -i '$a\dtoverlay=dwc2,dr_mode=host' /boot/firmware/config.txt
> 
> My man! That was the problem. Makes you wonder why deskpi hasn't updated their troubleshooting page. Does this company even QA their product?

Yes, we are checking the repository and fix the bug which cause by new os 's change .

iamLazyCode, 2025-01-17:
> I know this is an old thread but I thought I'd post it for anyone else wondering. I'm currently using a Raspberry Pi 4 Model B with a DeskPi runnning RaspiOS version "30f0c5e4d076da3ab4f341d88e7d505760b93ad7" and had no response from the front USB ports. [I found an article from DeskPi](https://deskpi.com/blogs/learn/trouble-shooting-how-to-check-front-usb-port) pertaining to a command to update with those functions. `sudo sed -i '$a\dtoverlay=dwc2,dr_mode=host' /boot/config.txt` In this case, the proper command to get these working is to use `sudo sed -i '$a\dtoverlay=dwc2,dr_mode=host' /boot/firmware/config.txt` as they've moved the boot file to `/boot/firmware/config.txt`

even after this mine doesnt work is there any else i need to enable/disabe sir ?

yoyojacky, 2025-02-06:
> even after this mine doesnt work is there any else i need to enable/disabe sir ?
yes, if your deskpi case's front USB ports are not work properly ,please try to add dtoverlay=dwc2,dr_mode=host to /boot/firmware/config.txt file in [all] section, and reboot your raspberry pi, after that ,try to connect your usb device to front USB ports and try again.

yoyojacky, 2025-02-06:
/boot/firmware/config.txt  here, the /boot/config.txt is an symblic link to /boot/firmware/config.txt file.

iamLazyCode, 2025-02-06:
> /boot/firmware/config.txt here, the /boot/config.txt is an symblic link to /boot/firmware/config.txt file.

though i have tried earlier but it dint worked will try again