DeskPi-Team/deskpi #37: Auto-Power-On after Power Loss

adamoutler, 2021-03-23:
Is there a way to enable an automatic power on after power loss.  I realize it's not an advertised feature, but there must be some way to connect some test points, or replace a capacitor, and cause the device to power on automatically when power is applied? 

This case is a great form factor for a RPI Server because it is stackable and has cooling.  This is a necessary feature for a server.

adamoutler, 2021-03-24:
What problems would this cause?  
[image]

closed 2021-05-11 by yoyojacky

paulpeeters, 2022-02-16:
I'm also looking for a way to automatically power on the DeskPi after a power failure. Can this be made possible somehow?

adamoutler, 2022-02-16:
just a note: I've been running the mod successfully for about a year and it works.

@yoyojacky can you tell me the source of the signal I'm using to make this work from schematics?