DeskPi-Team/deskpi #38: Any news about the new pcb board ?

bobaxxx, 2021-03-24:
Hi,

We dont have any news about new pcb board. Deskpi team doesnt answer to mail too.

i need also another USB3 rear adaptater

Can you make an answer for that

thx

nitz, 2021-03-24:
I'm wondering the same. I came here to ask this yesterday, but never actually created the issue, so I'm glad to +1!

tedder, 2021-03-24:
on 3/17 they replied to an email from me and said:
> There are some delay.
>
> The new PCB Board is in mass production.
>
> We will arrange shipments from the end of March according to the order date.

bobaxxx, 2021-03-24:
You have pretty more Luck than me because i sent 3 mails Without a single answer ...

jojobrogess, 2021-03-28:
Yeah the website was silently updated from "middle of March" to "after the end of March" without any notification or courtesy email's. IMO, that's extremely upsetting business practices. And the sparse, broken, or non-existent communication for months and then the shutdown for Chinese New Years or whatever Holliday it is, is extremely pathetic. 3 months with a handful of comments and an email or two to fix a pcb board. For an issue that should have been caught PRIOR to development, because that chip had WELL-KNOWN problems for SBC's. 
I get it though, they wanted it out before the m.2 argon or just faster in general, so things can happen. I get it. 
But STILL ZERO communication on what's going on, any type of eta, or even a "hey! We are still here! Working hard." That's what gets me the most. It would take less than 5 minutes to (weekly?hell monthly) post an update for better customer relations. But no? Do we not deserve it? Or is it just too much to ask for? Are you working _that_ hard that you can't give us an update? 
So yeah, I'm like a week or two away from requesting a refund because I was promised a fixed PCB board by mid march, and if that won't work then a fraudulent charge will be enacted because I purchased a product/service and the agreed upon product was not fully provided. 
It might seem like I'm salty, but it literally takes less than 5 minutes to post an update. We all have cell phones and internet connectivity. So that shows me how little they care

nitz, 2021-03-30:
I may be wrong, but I doubt it's that they don't care. I agree an update seems like a minimal amount of effort, but having been on small-team projects where people are waiting to hear status updates, the longer it goes without one the tougher it gets to tell folx it's delayed (or worse, that you don't know how long it will take.) I imagine cultural differences might make a difference too.

I'm sure the CNY exacerbated the issue, but the silicon supply chain problems right now are ridiculously painful. At work, we have some chips on order that we have used for years that have a lead time of 42+ weeks now. They usually are an easy overnight from Digi-Key, but _everything_ is backed up.

Checking out [the website](https://deskpi.com/products/deskpi-pro-for-raspberry-pi-4) though, It looks like they posted an update on March 26. I've duplicated it below

---

**Updated News on 26th March, 2021:**
We will start to re-sent of the updated replacement PCB board according to the order date from this week.

**Resent List:**
1. USB adapter
2. Daughter Board
3. T-shaped isolation foam
4. U-shaped support frame

Please kindly keep patience, you will receive your new PCB Boards soon.

**For new orders:**

We will start shipping in mid-April.

Thanks a lot for your all support and understanding!

**We will also send all of you a DeskPi bracket for free!**

[image]
[image]

Mister-42, 2021-03-30:
I also interested in what is NOT there.

No front IR receiver???

Pi-Mania, 2021-04-03:
Received tracking and can confirm my replacements are on the way. (AUS Reseller) 👍

M-H-010, 2021-04-11:
@Pi-Mania, you are lucky, Sales at Deskpi stopped responding to me as they clearly did not met the mid or end March deadline.
And as jojobrogess mentioned, an update on their website would take so little time and make a more mature impression.
I just hope the new PCB's do really solve all the issues....
Mine just develloped a new one:, undervoltage warnings and unexpexted reboots. :-(

bpicon, 2021-04-14:
New PCB received and installed. The bridge is identified as 174c: 1153, ASMedia Technology Inc. ASM1153 SATA 3Gb / s bridge (not 1153E SATA 6Gb / s bridge)
At least in my case (SATA XRayDisk SSD 240 GB connected, twister os 2.0.1, kernel 5.10.17-v7l +) using usb-storage driver by default, not UAS driver.
I don't know what firmware the bridge uses.

pi@raspberrypi:~ $ lsusb
Bus 003 Device 003: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Bus 003 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 174c:1153 ASMedia Technology Inc. ASM1153 SATA 3Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:~ $ lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=dwc2/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=ch341, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M

Pi-Mania, 2021-04-20:
Here's what i know:

1. The old version of the DeskPi Pro disk chip DID support UASP, but it does not support USB booting well. (low/poor  compatibility)  The new version of the disk chip does NOT support UASP and TRIM. (DeskPi Pro V2) but has HIGH compatibility.
2. asm1153 does not support 6Gbps,only supports 3Gbps. Since the 6GB chip is currently unavailable it had to be replaced with a 3GB chip. 

maybe not what some of you wanted to hear but, i still like this DeskPi case a lot. If the 6gb chip becomes available, perhaps they will switch to that? With new V2 boards i can now boot without quirks and i get slight speed improvements. I can also boot & read from sata/ssd drive attached to rear external USB 3 port. I could not personally do this with old boards. 

Regards

bpicon, 2021-04-20:
I don't know if I'm wrong, but according to the manufacturer's specifications (AsMedia) there is UASP support for this chip. Perhaps a firmware update can activate this support:

https://www.asmedia.com.tw/product/995Yq54sX2YI1th2/d1Eyq85QN8GhBwRC

- Compliant with USB3.0 Specification Revision 1.0
- Compliant with USB Specification Revision 2.0
- Support USB Super-Speed, High-Speed and Full-Speed Operation
- Support Mass Storage Class, Bulk-Only Transport Specification Revision 1.3
- **Support USB Attached SCSI Protocol Specification Revision 1.0**
- Compliant with Serial ATA Specification Revision 2.6Serial ATA bus up to 3Gbps Signal bandwidth
  ...
--

Station010, 2021-04-20:
Thanks to @Pi-Mania and @bpicon for the updated information.
A good spot on the asmedia page re USAP support but I wonder why they would go to all the effort of re-creatng the board and not test the frmware is up to date and gives the support for USAP and TRIM. Cheap USB adapter cases with ASMEDIA asm1153E chips come with the support ready to use out of the box.
So unless a firmware update and instructions are suppplied with the new board in the near furture I guess its a no sale for me,
what a disappointment from a normally very reliable company.

csmale, 2021-04-20:
This is now on the News page of the website:

> The Old Version DeskPi supports UASP. 
> The New Version DeskPi Pro on sale does not support UASP or TRIM.

Looks like a backward step if they are admitting that.

Mister-42, 2021-04-20:
That's the final straw. I have requested a refund.

mjeshurun, 2021-04-20:
Hello everyone, 
Sorry if my question is Noobiesh, but I'm currently running my Raspberry Pi 4 using a [Sandisk Extreme Portable 500GB SSD](https://shop.westerndigital.com/products/portable-drives/sandisk-extreme-usb-3-1-ssd#SDSSDE60-250G-G25) 
which appers to support UAS:
```
pi@raspberrypi:~ $ lsusb
Bus 002 Device 003: ID 8564:4100 Transcend Information, Inc. 
Bus 002 Device 002: ID 0781:558c SanDisk Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 1b1c:1b6d Corsair 
Bus 001 Device 004: ID 0658:0200 Sigma Designs, Inc. Aeotec Z-Stick Gen5 (ZW090) - UZB
Bus 001 Device 003: ID 8564:4100 Transcend Information, Inc. 
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:~ $ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Mass Storage, Driver=uas, 5000M
    |__ Port 2: Dev 3, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 1: Dev 4, If 1, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 1: Dev 4, If 0, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 3: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 3: Dev 5, If 1, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 3: Dev 5, If 2, Class=Human Interface Device, Driver=usbhid, 12M
```
and here are the hdparm test results for the Sandisk Extreme SSD:
```
pi@raspberrypi:~ $ sudo hdparm -Tt /dev/sda

/dev/sda:
 Timing cached reads:   1512 MB in  2.00 seconds = 756.64 MB/sec
 Timing buffered disk reads: 922 MB in  3.00 seconds = 307.03 MB/sec
```

My initial plan was to use the DeskPi Pro with a [Transcend 830S M.2 SSD SATA III 2TB](https://www.transcend-info.com/Products/No-982).
Based on your experience, will the Transcend 830S run in the same/slower/faster speeds as the Sandisk Extreme SSD? 
What else should I consider now that we know that the new Deskpi Pro version will lack UASP/TRIM support? 
I'd like to know if lack of UASP/TRIM are really "deal breakers"?
I guess, if my system will run at the same speeds as it runs today with the Sandisk Extreme SSD, I might still consider purchasing the Deskpi, but if the Deskpi ends up running slower, then there is no point of changing my current setup.

Pi-Mania, 2021-04-20:
I am on V2 hardware on a Samsung 750EVO 128gb SSD.  Timing cached reads:   1620 MB in  2.00 seconds = 809.81 MB/sec
 Timing buffered disk reads: 676 MB in  3.00 seconds = 225.25 MB/sec. My TechData m.2 is pretty much same result.

mjeshurun, 2021-04-20:
> I am on V2 hardware on a Samsung 750EVO 128gb SSD. Timing cached reads: 1620 MB in 2.00 seconds = 809.81 MB/sec
> Timing buffered disk reads: 676 MB in 3.00 seconds = 225.25 MB/sec. My TechData m.2 is pretty much same result.

Thank you @Pi-Mania. 
From your speed test, am I correct to assume I should expect similar speeds to those I have today with the Sandisk Extreme SSD?

Aasikki, 2021-04-21:
I'm planning to use this case for my home assistant server. What's  your opinion should I cancel my order because of this or will it be a big deal for that use?

> 
> 
> This is now on the News page of the website:
> 
> > The Old Version DeskPi supports UASP.
> > The New Version DeskPi Pro on sale does not support UASP or TRIM.
> 
> Looks like a backward step if they are admitting that.

Mister-42, 2021-04-21:
The lackof TRIM support will slow down SSDs and degrade their lifespan faster.

Aasikki, 2021-04-21:
Not sure if I want to keep waiting for my deskpi or should I just keep my current mess with external ssd enclosure and 3d printed case with cables coming from two different sides.Sigh... I wish there were other cases like this with proper ssd support.

risa2000, 2021-04-21:
@Aasikki FWIW I was planning to use desk-pi case for HA too, and since it did not work, I went ahead with Argon M.2 case. The major bummer was finding out that my Zigbee, Z-Wave (USB sticks) and (integrated) BT basically stopped working, because they were totally scrambled by EMI from USB 3.0 SSD traffic. So I am still running my HA from RPi 3B (no plus) with old USB 2.0. Depending on your needs and hardware you may actually have already the best setup ;).

Aasikki, 2021-04-21:
Haha that's pretty much the exact setup I have currently. Rpi b3+ and usb 2.0 ssd. I managed to get rpi 4 for cheap though and really want to use it for home assistant and throw the old one in my 3d printer for octoprint.

mjeshurun, 2021-04-21:
@Pi-Mania can you please comment what type of M.2 socket connector is available on the Deskpi Pro?
Is it B key, M key or B&M? 
I would like to make sure the SSD I order will fit properly.

Aasikki, 2021-04-22:
@risa2000 Do you mean that home assistant didn't work at all with the deskpi or what went wrong? If that is the case I'm definitely going to have to cancel my order.

risa2000, 2021-04-22:
@Aasikki No, I did not bother with desk-pi considering it did not work with USB SATA disks. My observations are based on the similar setup run on RPi 4 in Argon M.2 case, which also sports USB 3.0 to SATA interface (only to M.2 slot though).

Aasikki, 2021-04-22:
@risa2000 ok in that case I think I'll give deskipi a go then.

Pi-Mania, 2021-04-23:
> @Pi-Mania can you please comment what type of M.2 socket connector is available on the Deskpi Pro?
> Is it B key, M key or B&M?
> I would like to make sure the SSD I order will fit properly.

The DeskPi Case can accept either M or B+M SSD's Via a supplied daughterboard that slots into SATA port. (But NOT NVMe drives)

mjeshurun, 2021-04-23:
Thanks @Pi-Mania 🙏
Could we potentially change the 2.5inch SSD expansion board and the M.2 connector that comes with the Deskpi Pro with a different M.2 connector board that supports SATA 6Gb, UASP and TRIM?

Marach79, 2021-04-24:
I requested a refund and they ignored that in February. The case was useless for me and I had thrown away. Three days ago I received the replacement board. Ironic 🤣 My request to buy the missing parts.... They want to answer ASAP.... I'm still waiting. Very bad support in my opinion.

Mister-42, 2021-04-25:
Hello @yoyojacky @deskpi 

Saying >refunded< on the order page does not make the money magically appear back on my PayPal account. Ignore emails also does not fix this. PROCEED WITH THE REFUND AT ONCE!

Marach79, 2021-04-25:
After opening an PayPal case they want from me to send back the case to them. Sure.... paying for a not functional case and the delivery twice. Definitely no! And a support that's not able to answer ASAP.... I'm very angry about the not working communication.

leicray, 2021-04-26:
I bought one of the new spec cases about ten days ago (from a UK supplier) and find the my Pi now freezes once or twice a day with the white disk-activity light flashing furiously. Previously, it was absolutely stable when installed in a simple case with a fan and with the SSD drive connected via a StarTech USB32SSAT3CB (Chipset: ASM1153E) USB3 to SATA adapter. That chipset supports UASP and TRIM.

Removing the drive (Kingston 120 GB SA400S37/120G) from the DeskPi Pro case and reconnecting it externally with the adapter cable to a USB3 port has restored the stability that I enjoyed previously. Boot time is subjectively faster too.

There is clearly still something wrong with the revised board. I will be requesting a refund.

Mister-42, 2021-04-26:
> Hello @yoyojacky @deskpi
> 
> Saying >refunded< on the order page does not make the money magically appear back on my PayPal account. Ignore emails also does not fix this. PROCEED WITH THE REFUND AT ONCE!

Reminder for @yoyojacky @deskpi. I'm gonna repeat this over and over until you have done as promised.

babyheysus, 2021-04-29:
Got my replacement board and it doesn't see my SSD at all. The pi sees the ASMedia board though. Was there anything you did differently @Pi-Mania? Outside the deskpi instructions. Everything is being pumped by a OnePlus charger so power should be good, SSD is a 120GB SanDisk SSD Plus and that was readable on my PC. OS is stock from what deskpi gave.

Any help at all would be great

Aasikki, 2021-04-29:
@babyheysus OnePlus chargers only give 2A unless you are charging a oneplus devie as they don't support Qualcomm Quickcharge. Try another charger that supports QC.

Mister-42, 2021-04-29:
> Reminder for @yoyojacky @deskpi. I'm gonna repeat this over and over until you have done as promised.

Still waiting @yoyojacky @deskpi

simonduane, 2021-05-01:
> I bought one of the new spec cases about ten days ago (from a UK supplier) and find the my Pi now freezes once or twice a day with the white disk-activity light flashing furiously. Previously, it [...]

I can confirm this behaviour. New DeskPi delivered 1 week ago, assembled only today. Works, unless SSD is installed inside case and USB plug is fitted. Then no HDMI video on either port. It boots ok, because remote SSL connection is possible. But not VNC, which only says "Cannot currently show the desktop".
Strangely, rebooting without the USB plug brings up the desktop on a monitor (and via VNC), inserting the plug blanks the monitor, but the remote desktop via VNC survives. I suspect that by the time the HDMI signal has been routed out the back of the DeskPi case, it's too marginal to drive the monitor. I've seen inconsistencies a bit like this with pi-zero.
This is too flakey to be useful for me. I bought the case to run the pi as a desktop. With a monitor. At this price, it should just work. I'm sending it back.

Aasikki, 2021-05-01:
@simonduane fuck that I'm going to ask for a refund now.

G6EJD, 2021-05-03:
> I can confirm this behaviour. New DeskPi delivered 1 week ago, assembled only today. Works, unless SSD is installed inside case and USB plug is fitted. Then no HDMI video on either port. It boots [...]

VNC not able to display is nearly always because the RPi display resolution is not set correctly to say 1920x1080i if your monitor cant' display 60Hz signals.
My two cases are working correctly, I assume you used the new board supplied (two boxes arrived) to replace the driver chips, etc. I get good SSD performance too, not excellent but adequate:
pi@raspberrypi:~ $ sudo hdparm -Tt /dev/sda

/dev/sda:
 Timing cached reads:   1306 MB in  2.00 seconds = 653.14 MB/sec
 Timing buffered disk reads: 588 MB in  2.00 seconds = 294.0 MB/sec

The only issue I've found is the fan stops working the moment a USB device is plugged in and is allocated to USB0 also used for the fan control. I tried changing the USB address in the Python code, but no difference.

simonduane, 2021-05-03:
> VNC not able to display is nearly always because the RPi display resolution is not set correctly to say 1920x1080i if your monitor cant' display 60Hz signals.

Nearly always, maybe, but not in this case. I was in an extended and very helpful exchange with RealVNC support last year over "Cannot currently show the desktop" messages - sometimes this message clears itself but only after a really long time (and sometimes never). Perhaps you didn't read my comment as far as the "... rebooting without the USB plug brings up the desktop on a monitor (and via VNC)." part? Meaning, it's not because of any settings that VNC isn't working.

All I can say is, your two cases work?
Fine. Mine doesn't.
And yes, of course I swapped out the old board for the new. I also fitted the RF interference-suppressing conductive cotton insert between the USB ports. If it looks like a bodge to get over flaky behaviour, maybe that's exactly what it is ... and maybe it doesn't always do the job.

Mister-42, 2021-05-03:
> Still waiting @yoyojacky @deskpi

Email doesn't help. Posting in this issue doesn't help. This is very unprofessional.

**@yoyojacky @deskpi process that F@%$ing refund already!!!**

leicray, 2021-05-03:
Update on my experience:

Previously, my Pi would crash once or twice a day with the white disk-activity light flashing furiously when installed in the DeskPi Pro case. I also posted my experience (also as leicray) on another forum hoping for guidance from there too: [TechForum Support](https://forum.digikey.com/t/deskpi-pro-bad-sata-controller/13490/14) It might be useful to read through that thread to understand the sequence of events and to see the experiences of other users.

The best advice from TechForum Support was to keep trying with the Pi in the DeskPi Pro case and wait for a crash and then take a look at the contents of the kernel log just before the crash: <code>/var/log/kern.log</code>

Before we get to that, it took nearly three days this time before it crashed again. Luckily, I had a a terminal window open on the desktop. However, some commands worked within the terminal, but others did not. For example I was able to <code>cat</code> the contents of kern.log but could not save it, as <code>nano</code> would not load and redirecting to output from <code>cat</code> to a file would not work either.

At this point, I could not connect to the pi using ssh or VNC. In addition, the desktop was frozen apart from the open terminal. It's probably best to say that the pi was only just alive at this point.

The content of kern.log shortly before the "crash" is:
<pre>
pi@raspberrypi:~ $ uptime
13:12:00 up 2 days, 21:16, 2 users, load average: 13.02, 10.45, 9.56
pi@raspberrypi:~ $ cat /var/log/kern.log
ay 3 11:46:09 raspberrypi kernel: [244220.207312] xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
ay 3 11:46:09 raspberrypi kernel: [244220.207325] xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
ay 3 11:46:09 raspberrypi kernel: [244220.358262] usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
pi@raspberrypi:~
</pre>

There are ten virtual bonus points on offer for anybody who can guess how I did capture the the content of kern.log and it was not by copying manually from the screen to a plain text file.

The good news is that the UK supplier (PiHut) has issued an RMA number and provided a pre-paid return label. In an ideal world, I would like to have a working case, but there is clearly something very wrong even with the revised board.

Any additional comments or encouragement would be most gratefully accepted.

simonduane, 2021-05-03:
> Update on my experience:
...
> The good news is that the UK supplier (PiHut) has issued an RMA number and provided a pre-paid return label. In an ideal world, I would like to have a working case, but there is clearly something [...]

Thanks for the update.
Well I'm encouraged anyway - I got my DeskPi from the same source. Fingers crossed they'll be similarly helpful to me. I went through enough time-consuming log-capturing games in trying to pin down the cause of my VNC problems last year. I'm really not up for more of that - I was really hoping for a nice tidy pi desktop system that just works (and am planning on trying the Argon ONE M.2 next)...

nitz, 2021-05-03:
>  I was really hoping for a nice tidy pi desktop system that just works (and am planning on trying the Argon ONE M.2 next)...

I actually ordered the Argon One M.2 at the same time I ordered the DeskPi, so I've had a pi in it basically the whole time. It's my "daily driver" pi, and is honestly fantastic. The M.2 version has the case that does the micro hdmi to hdmi conversion too, which was a big deal for me and part of what made me go for it as well. But it does *fantastic* with thermals, is incredibly clever in it's design and machining, and still has a lot of the features I had hoped for. Sure no SATA, but if you're looking to go M.2 anyways, I don't think there's a better option.

simonduane, 2021-05-03:
> I actually ordered the Argon One M.2 at the same time I ordered the DeskPi, so I've had a pi in it basically the whole time. It's my "daily driver" pi, and is honestly fantastic. The M.2 version has [...]

Yet more encouragement - cheers for that, it's just what I want to hear. (I only bought the SSD to go with the case, so SATA vs M.2 was an arbitrary choice for me ... I'm hoping that PH will look favourably on my request to swap SSDs too. It's not like it's even been formatted or anything yet ...)

leicray, 2021-05-03:
I am slightly wary of going with the Argon One M.2 case for three reasons:

1. As far as I can see, the Argon case does not provide an accessible microSD slot: please correct me if I'm wrong.
2. I would now have to buy an M.2 drive to replace the SATA drive that I currently use in the DeskPi Pro case.
3. I cannot see how to write an image of the SATA drive to the M.2 drive without buying an M.2 to USB adapter.

On the plus side, it does have the ASmedia ASM1153E chipset which does provide UASP support, and presumably TRIM support too.

Any thoughts?

nitz, 2021-05-03:
It doesn't have an easy to get ot SD slot, no. That was a non-issue for me since I intended to boot from the SSD, and attach any removable storage I need via USB. That said, It's just the base screws (4 of them) to remove the bottom which will let you get to the SD slot if you need to. The non-M.2 base does have access to it, so I imagine it's probably at least a concern for them if they ever do another revision.

Your point 2 is completely right. I was already using a M.2 SSD (and an adapter to a 2.5" SATA when I needed it,) so that's an easy move for me, but certainly more difficult for people in your case. That said, eBay is a great way to get close to if not exactly what you paid for if you flip your SSD there.

Your third point: you kinda get one automatically with the case: the base of the case/M.2 board is connected to the Pi via a USB 3 bridge the same way the DeskPi does, so you can absolutely use it to write your image to. I did it with mine directly from the Pi before I cased it up!

And for booting, I've only used two different M.2 drives in it so far, but neither of them had issues booting!

mjeshurun, 2021-05-03:
> I am slightly wary of going with the Argon One M.2 case for three reasons:
>
> 1. As far as I can see, the Argon case does not provide an accessible microSD slot: please correct me if I'm wrong.
> 2. I [...]

Based on this blog post and comments discussion, it appears the Aragon One M.2 currently doesn't supports TRIM.
[https://www.jeffgeerling.com/blog/2021/argon-one-m2-raspberry-pi-ssd-case-review](https://www.jeffgeerling.com/blog/2021/argon-one-m2-raspberry-pi-ssd-case-review)

leicray, 2021-05-03:
Hi @nitz,

Thank you for the prompt reply!

> Your third point: you kinda get one automatically with the case: the base of the case/M.2 board is connected to the Pi via a USB 3 bridge the same way the DeskPi does, so you can absolutely use it to write your image to. I did it with mine directly from the Pi before I cased it up!

The fact that I can use the base of the case as a temporary adapter makes perfect sense. Kicking myself that I did not work that out for myself.

leicray, 2021-05-03:
Hi @mjeshurun

> Based on this blog post and comments discussion, it appears the Aragon One M.2 currently doesn't supports TRIM.
https://www.jeffgeerling.com/blog/2021/argon-one-m2-raspberry-pi-ssd-case-review

I might hold off for a bit in the hope that the makers of the Argon case do update the firmware to support TRIM.

jamsoft, 2021-05-04:
Thought I'd chip in here.

I just bought a DeskPi Pro, SSD and Pi 4B 8Gb last week from PiHut I had read about the lack of TRIM etc but this is really just a 'nix scratch pad play thing so totally not mission critical at all.

Built it all and installed Ubuntu 64 bit pi build. I'm typing this on it now. I have seen the crazy flashing USB light a number of times, requiring a power cycle to fix.

WiFi is dog slow and I've turned BT off for now as that seemed to be impacting WiFi ... but then that's also true of my stupid mac ... and it's not so magic mouse but that's another story.

The fan control install under Ubuntu 64 is pretty much broken. I've managed to get manual control working but really, what use is that? Cooling has to be controlled dynamically or it's "just a fan".  Someone else suggested in another ticket the drivers need recompiling for 64bit but I don't know where to start with that at the mo. Also, anything other than 100% fan speed has an annoying vibration resonance that I haven't identified and fixed yet.

Seems off manual control works but automatic doesn't.

All in all, case looks and feels great but there are definitely issues to be ironed out and my experience overall is definitely on the slightly frustrated side.

Currently eyeing up a NUC to be honest ...

G6EJD, 2021-05-04:
Same experiences here, but I discovered I get good WiFi connections on 5 GHz and now the WiFi is fast. The metal case is not conducive to good RF links.
I have the same annoying buzz / resonance from the fan, not solved that yet.
When I plug in a USB device the fan stops, examination of the script reveals it uses usb0 which is also being assigned to the first device plugged in. I am going to allocate and name a USB port say usbfan and change the script to use that to solve the problem. The cpu runs at about 40° most of the time and the fan is off, so cooling is good. Overall a neat case with issues.

jamsoft, 2021-05-04:
Forgot to mention. The power light always stays blue now I'm running Ubuntu. Would power down and turn red under Raspian so definitely issues there in the config script.

Currently trying to get GPU accelration working and failing at that but that's nothing to do with this case.

I'm not driven enough to get a refund but the experience is a bit lackluster for what is THE most expensive Pi case on the market ...

I'm a bit picky with my hardware builds so these issues are actually quite annoying on that level.

jamsoft, 2021-05-04:
@G6EJD  Hmm ... 5Gz is pretty useless unless you're in the same room as the WiFi source.

jamsoft, 2021-05-04:
On another issue here a chap posted this list of issues:

    Front USB not working (or not properly)
    USB dongle fail to use correctly the internal SSD/HDD
    WIFI disturbance
    Loading time takes too long.
    Fan speed control should be better implemented for easy modification on OS, example on Ubuntu 20.10

That's an issue WITH EVERY SINGLE FUNCTION OF THE CASE other than it being a case ... anyway, enough moaning.

leicray, 2021-05-05:
I have just heard from The Pi Hut, here in the UK, that they have decided to suspend sale of the case until matters are resolved.

simonduane, 2021-05-05:
@leicray interesting - I just sent mine back with its RMA yesterday.

mjeshurun, 2021-05-05:
I sent Deskpi a refund request two days ago. 
Didn’t hear back from them yet, but I’ll give this another week, before contacting them again about it.

risa2000, 2021-05-05:
FWIW, I emailed sales on April 20, asking for a refund and instructions how to proceed. They never replied, but sent me an advance notice about shipping the replacement board. So I went to PayPal customer protection and raised a complaint there while requesting a full refund.

G6EJD, 2021-05-05:
I’m wondering what’s different with my two cases which apart from the fan control using USB0 which I fixed by allocating the port to the  fan and changing the Python script all is working as expected. I’ve since found if I take the RPi out of the case it won’t connect to a 2.4G access point, well unless I force it to do so via wpa_supplicant. 5G range is a good 6M but I’m using a wifi6 mesh. I note the new boards have a screening can over components directly below the Pi USB ports. In one case I’m using an M.2 drive the other a 2.5” SSD both boot in identical times of about 7.5secs according to system-analyse and I’m getting near maximum r/w speeds for the two drives. Processor temp is a steady 39°c at most 46°. There are matching serial number stickers on the top and bottom pcbs the originals were blank. I think the version number is 3.1 on the new boards. The fan buzz is of very low sound level, but doesn’t other me as at 1M it’s inaudible.

[71 more replies]

closed 2021-06-30 by yoyojacky