DeskPi-Team/deskpi #39: Ubuntu Install Script Fails mix of users.

muckypaws, 2021-04-12:
Hello,

I'm trying to install the ubuntu script on my Pi4-8GB, however the script fails if I follow the instruction

`sudo ./install-ubuntu-mate.sh`

This is because it seems to look for the drivers where they aren't, in they assume the drivers are located in user root and not the current $USER

jason@jason-desktop:~/deskpi$ **sudo ./install-ubuntu-mate.sh**
 * DeskPi Fan control script installation Start.
 * DeskPi main control service loaded.
./install-ubuntu-mate.sh: line 25: **cd: /home/root/deskpi/drivers/c/:** No such file or directory
**cp: cannot stat '/home/root/deskpi/drivers/c/pwmFanControl': No such file or directory
cp: cannot stat '/home/root/deskpi/drivers/c/fanStop': No such file or directory
cp: cannot stat '/home/root/deskpi/deskpi-config': No such file or directory
cp: cannot stat '/home/root/deskpi/Deskpi-uninstall': No such file or directory**
 * DeskPi Service configuration finished.
 * DeskPi Service Load module.
Created symlink /etc/systemd/system/multi-user.target.wants/deskpi.service → /lib/systemd/system/deskpi.service.
Created symlink /etc/systemd/system/halt.target.wants/deskpi-safeshut.service → /lib/systemd/system/deskpi-safeshut.service.
Created symlink /etc/systemd/system/shutdown.target.wants/deskpi-safeshut.service → /lib/systemd/system/deskpi-safeshut.service.
Created symlink /etc/systemd/system/poweroff.target.wants/deskpi-safeshut.service → /lib/systemd/system/deskpi-safeshut.service.
 * DeskPi PWM Fan Control and Safeshut Service installed successfully.
 * System will reboot in 5 seconds to take effect.
Job for deskpi.service failed because the control process exited with error code.
See "systemctl status deskpi.service" and "journalctl -xe" for details.
Reboot system for changes to take effect

If I try running the script without **SUDO** that's going to fail on permissions issues.

Kind Regards
Jason

donpezet, 2021-04-15:
You have to do a few workarounds to get this up and running under Ubuntu. 
1. The driver that it comes with is 32-bit so you have to recompile it to get a 64bit binary. Just delete `pwmFanControl from ~/deskpi/drivers/c` and recompile with `make`. On a default install of Ubuntu Server you will need to do a `sudo apt install gcc make` first. 
2. It is trying to write the `dtoverlay=dwc2,dr_mode=host` boot configuration settings to `/boot/config.txt` or `/boot/firmware/config.txt`. Both are wrong. You can manually add the lines yourself to `/boot/firmware/usercfg.txt`.
3. When you execute the script with `sudo` it will try to run everything from `/home/root` which doesn't exist with the default Ubuntu install. I just hard coded the `$USER` variable in the script to be my username. 

Once all that is done the script will complete properly. From there you can reboot (the automatic reboot doesn't work) and then run `deskpi-config` to setup the fan controls as desired.

muckypaws, 2021-04-15:
Hi Don, 

Thank you for your reply. 

I resolved the issue, however if I make make the following suggestions?

1) If you use $SUDO_USER instead of $USER

installationfolder=/home/$SUDO_USER/deskpi

This will resolve the first issue. 

I did recompile the binary for 64 bit, however given the diversity of Ubuntu deployments may I suggest a check in the script for 

uname -m 

If 64 is returned in the resultant architecture, then change the copy for pre-compiled 64 Bit versions of fanStop and pwmFanControl

The changes are easy to make and I’m happy to submit a fixed version as this will make the enduser experience more seamless for those that aren’t as ’nix savvy.

Kind Regards
Jason
www.muckypaws.com

jamsoft, 2021-05-04:
Gets my vote. I'm pretty tech savvy but 'nix is still a mystery to me.

How do I recompile with make?

yoyojacky, 2021-05-11:
> The changes are easy to make and I’m happy to submit a fixed version as this will make the enduser experience more seamless for those that aren’t as ’nix savvy.

that's great, I am still using 32bit raspberry pi OS, so it a little bit difficult for me to fix this issue.. what a shame..

yoyojacky, 2021-05-11:
> Gets my vote. I'm pretty tech savvy but 'nix is still a mystery to me.
> 
> How do I recompile with make?

there is a Makefile , just typing: make will ok.

yoyojacky, 2021-05-11:
> Hi Don, Thank you for your reply. I resolved the issue, however if I make make the following suggestions? 1) If you use $SUDO_USER instead of $USER installationfolder=/home/$SUDO_USER/deskpi This [...]

Hi muckypaws, thanks for your tips ..

closed 2021-06-30 by yoyojacky