DeskPi-Team/deskpi #40: Automatic fan control needs hysteresis

spiff72, 2021-05-06:
The automatic fan controller really could use some hysteresis built in so it doesn't cycle on and off unnecessarily when it is hovering around one of the temperature thresholds.  For example if the 25% fan speed threshold is 45 degrees, the fan should turn on at 45c but should stay on until the temp drops to below a value of, say 43 or 42 degrees.  Just pick a hysteresis value of 2 or 3 degrees and apply that to all the threshold values.

yoyojacky, 2021-05-11:
hi spiff732,  Because there is currently an MCU on the deskpi motherboard that linearly detects the signal sent by the Raspberry Pi, if you want to start the fan at a specific CPU temperature, you can use deskpi-config to set a higher threshold. You can make the fan rotation change less, because considering the different needs of many people, the range of customization is large, and you can also control the fan by sending pwm data to the serial port recognized by the Raspberry Pi.

closed 2021-06-30 by yoyojacky

jojobrogess, 2022-03-24:
Adding hysteresis behavior to code is not difficult. 
You just need to store the state you're in and make the thresholds for transitioning into another state dependent on that.

You could also solve this by implementing small dead bands around the change levels. When the value is in a range near the transition threshold, do nothing. Only when the value is certainly above or certainly below the transition, make the change.

As for adding hysteresis to the fan speed itself(instead of ramping from 40%to100, or whatever, it slowly gets to speed) yeah idk if that's possible.

jojobrogess, 2022-03-24:
actually on second thought, you might be able to get it to slowly ramp fan speed.
By "pipe"ing the difference in  let's say "pwm_040">"pwm_100" and inputting every xx0/xx5 up.
so using my example. instead of going from sending "pwm_040" then when temp changes you send "pwm_100" you would send:
"pwm_040"
"pwm_040"
"pwm_040"
"pwm_040"
"pwm_040"

mdickers47, 2023-01-26:
this is just a drive-by, but for the sake of anyone else that happens to drive by, the feature @yoyojacky and @jojobrogress are describing is something else, not hysteresis, and not the problem @spiff72 is pointing out.