DeskPi-Team/deskpi #41: NEW BOARD DOESN'T IMPROVE SSD SPEED

wardnuke, 2021-05-07:
I received the updated boards yesterday. I just changed out the old with the new boards and booted up. No issues there, but after performing a diagnostic test on the SSD, I found the speeds the same as when we had to run Quirks to get them to function. I removed the Quirks command from my config.txt file and rebooted. No change in speed. When I first got this DeskPi Pro I was able to boot up once with the SSD's in "normal" mode and my speeds were comprable to what you'd expect for a SSD.

Raspberry Pi Diagnostics - version 0.6
Wed Dec 23 14:21:05 2020

Test : SD Card Speed Test
Run 1
prepare-file;0;0;280068;547
seq-write;0;0;259035;505
rand-4k-write;0;0;74642;18660
rand-4k-read;73061;18265;0;0
Sequential write speed 259035 KB/sec (target 10000) - PASS
Random write speed 18660 IOPS (target 500) - PASS
Random read speed 18265 IOPS (target 1500) - PASS
Test PASS

After I ran Quirks I was getting these speeds:

Raspberry Pi Diagnostics - version 0.6
Wed Dec 23 15:23:14 2020

Test : SD Card Speed Test
Run 1
prepare-file;0;0;184089;359
seq-write;0;0;179550;350
rand-4k-write;0;0;22505;5626
rand-4k-read;18729;4682;0;0
Sequential write speed 179550 KB/sec (target 10000) - PASS
Random write speed 5626 IOPS (target 500) - PASS
Random read speed 4682 IOPS (target 1500) - PASS
Test PASS

At this time, with the new board installed and with or without Quirks set, I'm still getting speeds that are 25-30% lower than when I was able to utilize the full potential of the SSD.

So is there some other driver I need to install to utilize the controller on this new board??? I really don't want to do a full reinstall of the system on my DeskPi as I've gotten it set up just the way I want it while waiting for the fixed boards to arrive.

Help.

yoyojacky, 2021-05-11:
> I received the updated boards yesterday. I just changed out the old with the new boards and booted up. No issues there, but after performing a diagnostic test on the SSD, I found the speeds the same [...]

Hi, Could you please execute following command in terminal and sent the feedback to me?
1.  lsusb -t
2.  lsblk

wardnuke, 2021-05-12:
lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=dwc2/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=ch341, 12M
        |__ Port 3: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 3: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 3: Dev 4, If 2, Class=Human Interface Device, Driver=usbhid, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M

lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda      8:0    0 465.8G  0 disk 
├─sda1   8:1    0   256M  0 part /boot
└─sda2   8:2    0 465.5G  0 part /

yoyojacky, 2021-05-13:
> lsusb -t
> /: Bus 03.Port 1: Dev 1, Class=root_hub, Driver=dwc2/1p, 480M
> |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
> |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=ch341, 12M [...]

ok, please try : lsusb  without `-t` and send it to me again.

wardnuke, 2021-05-13:
lsusb
Bus 003 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 003 Device 003: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Bus 003 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 174c:1153 ASMedia Technology Inc. ASM1153 SATA 3Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 214b:7250  
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

pjk51, 2021-05-17:
I have the same problem. I read somewhere that the ASM1153 detected as ASM1051 and breaking UAS? 
I found the driver for it: https://linux-hardware.org/?id=usb:174c-1153&page=2#status
It is included in Majaro and ubuntu, but not in raspberry pi OS?

CrunchyDoodle, 2021-05-21:
I have the new boards installed and removed the special quirks. I have Ubuntu MATE 20.10 aarch64 (Kernel: 5.8.0-1024-raspi). When I check USB devices I get Bus 002 Device 002: ID 174c:1153 ASMedia Technology Inc. ASM1153 SATA 3Gb/s bridge. However, it still uses the usb-storage driver and not the correct uas driver as it does with a USB to SATA adapter cable. I don't know the answer to how to get it to use the right driver.

bpicon, 2021-05-21:
> I have the new boards installed and removed the special quirks. I have Ubuntu MATE 20.10 aarch64 (Kernel: 5.8.0-1024-raspi). When I check USB devices I get Bus 002 Device 002: ID 174c:1153 ASMedia [...]

https://deskpi.com/blogs/news/about-uasp-trim

It seems that the firmware needs to be updated

CrunchyDoodle, 2021-05-22:
Yes! The new firmware for the controller chip works. I now have the proper uas driver, and I enabled TRIM too. The disk benchmark is now what it should be.

TeoTN, 2021-05-30:
The new board actually makes the disk slower even with TRIM enabled. I've used this benchmark: https://jamesachambers.com/raspberry-pi-storage-benchmarks-2019-benchmarking-script/

- DeskPi Pro v2 scored 7560
- DeskPi Pro v3 (TRIM-enabled) scored 6832 (-10%)
- DeskPi Pro v3 (TRIM disabled) scored 4987 (-34%)

bpicon, 2021-05-31:
> The new board actually makes the disk slower even with TRIM enabled. I've used this benchmark: https://jamesachambers.com/raspberry-pi-storage-benchmarks-2019-benchmarking-script/

This is the result I get when passing the test, TRIM enabled, Xraydisk SATA3 256GB connected:

[image]

TeoTN, 2021-05-31:
> This is the result I get when passing the test, TRIM enabled, Xraydisk SATA3 256GB connected:

Okay, but the score alone on the v3 doesn't tell a lot, could you run it on the previous board and compare the individual results?
For me the performance of the newer board is worse than of the previous one.

bpicon, 2021-05-31:
> Okay, but the score alone on the v3 doesn't tell a lot, could you run it on the previous board and compare the individual results?

Sorry, I don't have the board available right now to test. I too would have liked to make the comparison.

yoyojacky, 2021-06-30:
Testing disks is never just testing disk controllers, but also considering system, disks, disk controllers and many other factors, as well as configuration and so on.

closed 2021-06-30 by yoyojacky

TeoTN, 2021-07-01:
> Testing disks is never just testing disk controllers, but also considering system, disks, disk controllers and many other factors, as well as configuration and so on.

Why would you close that issue, even without asking for details?
I've run the test on the same system, the same disk, on a pristine Raspberry Pi OS 32-bit installation, I doubt there were any other factors that I didn't take into account.

symbios24, 2022-06-29:
Hello

I have a usb 3.5 case with asm235cm in my Raspbery pi4 and getting very slow speeds is there a firmware update somewere that fixes this issue?

yoyojacky, 2022-07-01:
> Hello
> 
> I have a usb 3.5 case with asm235cm in my Raspbery pi4 and getting very slow speeds is there a firmware update somewere that fixes this issue?

Could you please paste some picture of your device ? SSD and connection?

symbios24, 2022-07-09:
Connection is usb 3.0
The hdd enclosure is the: natec nkz-0448

https://natec-zone.com/product/rhino-sata-3-5

It has the Asmedia asm235cm Chipset thats is capable of uasp and trim fuctions

The hard drive is mechanical capable of more than 100mb/sec without bad sectors.

The lsusb -t gives this

https://ibb.co/16y3Vfy

The usb-devices this

https://ibb.co/hYZXNdt

The lsusb this

https://ibb.co/qgNq8zq

If i didable quirks hdparm gives me speeds of more than 100mb/sec

The thing is the realtime perfomance of the drive is like its working with 10mb/sec like there is a huge delay/lagging in the operations of the drive

For example if i do an apt upgrade it could take like forever.

If i use the same drive in a usb2.0 case in a usb port the perfomance difference even with 32mb/sec Speed its huge.

The hdd is powered externally,i have the official Raspberry pi psu in a Raspberry pi 4B 4GB rev 1.4