DeskPi-Team/deskpi #42: Shutdown with short power off

jzazo, 2021-05-13:
Hi! Is there a way to tell the OS to shutdown by pressing the power button with a short press? I tried installing acpid tools but it did not work. Any tool that can help getting a safe shutdown? Thanks.

jzazo, 2021-05-14:
Any way we can trigger a shutdown script when pressing the button? What kind of event does it generate into the OS to configure that, similar to how it is done with GPIO? Any pointer?

yoyojacky, 2021-06-30:
hi jzazo, it seems like you cannot do that just by only one click unless you modify the firmware on the MCU on the daughter board, but you can add another button via GPIO and write a code to detect the GPIO pin's level and send `power_off` signal to `/dev/ttyUSB0` device which created by adding `dtoverlay=dwc2,dr_mode=host` to /boot/config.txt (Raspbian OS, Other OS need to enable `dwc2` overlay to let raspberry pi can communicate with MCU on daughter board.

closed 2021-10-13 by yoyojacky