DeskPi-Team/deskpi #43: DeskPi Pro with update boards - SSD not working.

nr3i, 2021-05-14:
DeskPi Pro with update boards.  Main issue: SSD not working.  
(with the OLD boards, I never got the SSD to work, either)

BUT, I could remove the SSD from DeskPi and using a USB3 to SATA cable, it would boot.
I used both a SSD and the M.2 SSD.  (SSD = WD BLUE 250GB 3D NAND) and 
(M.2 card has a Kingston SA400M8/240G)  and both work outside the Pi with the USB3 cable.

Secondary NEW issue (SD card was not working) fixed by replacing the small/short cable.  It was cut and the fan connector was broken too, the fan cable is OK and does work in the broken connector.

After reading wiki and a total reload.  
I ran the Windows “FW Flash tool” OK.

$ lsblk  still only shows SD card.  “mmcblk0”
$ lsusb now shows ASMedia Tech  ASM items both 3G/sec and 6G/sec.
“/dev/disk/by-id”  NOW shows usb DeskPi_Pro 00202011…  – Good?

BUT, using SD Copier,  when run, I get “could not write to destination” {the usb_DeskPi_Pro DISK}
AND still no SSD boot.

YES, latest boot version in USB mode and all updates.

steve

nr3i, 2021-05-15:
$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 3, If 0, Class=Mass Storage, Driver=usb-storage, 5000M  <- is the DeskPi M.2 SSD
    |__ Port 2: Dev 4, If 0, Class=Mass Storage, Driver=uas, 5000M         <-- this is the USB3 to SATA cable with WD Blue SSD

I am guessing (more reading) that "Driver=usb-storage" is the problem...   ?

HOW do I get the DeskPi to use the uas driver?   The serial # on the DeskPi USB interface card is # SN21155233 if that helps?
Even with "Driver=usb-storage", I don't understand why "SD Copier" can not write?  More than just the uas "problem" ??

steve

nr3i, 2021-05-17:
OK, I read more.  I saw others were running SSD's with 120G, and asked "Maybe size matters"...  
I have a Kingston SSD SA400S37/120G now showing (with sd copier) as: "DeskPi Pro (/dev/sda)" and it would NOT work with "SD Copier", either (“could not write to destination”).  Nor would it boot.  Using gparted, it does not show as a disk.

Class=Mass Storage, Driver=usb-storage, 5000M
ID 174c:55aa ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge

steve

nr3i, 2021-05-28:
Hello @yoyojacky @deskpi 
I am not getting very far...  I am trying TRIM, but only get so far and then ERROR and/or I am not able to figure out how to do the edit.

I ran these commands with the M.2 SSD in the DeskPi and the Kingston 120G on an external USB to SSD cable.

pi@DeskPi:~ $ lsscsi
[0:0:0:0]    disk    DeskPi   Pro              0     /dev/sda 
pi@DeskPi:~ $ lsusb
Bus 003 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 003 Device 003: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Bus 003 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 174c:55aa ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@DeskPi:~ $ lsusb  <------------------------------------------------------- after adding the external USB SSD
Bus 003 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 003 Device 003: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Bus 003 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 004: ID 174c:55aa ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
Bus 002 Device 003: ID 174c:55aa ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@DeskPi:~ $ lsscsi
[0:0:0:0]    disk    DeskPi   Pro              0     /dev/sda 
[1:0:0:0]    disk    KINGSTON  SA400S37120G    0     /dev/sdb 
pi@DeskPi:~ $ lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=dwc2/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=ch341, 12M
        |__ Port 3: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 3: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 3: Dev 4, If 2, Class=Human Interface Device, Driver=usbhid, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 3, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
    |__ Port 2: Dev 4, If 0, Class=Mass Storage, Driver=uas, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
pi@DeskPi:~ $ 

"you" cans see the external SSD is "uas" ...

NOW this is where I get stuck...
pi@DeskPi:~ $ sudo find /sys/ -name provisioning_mode -exec grep -H . {} + | sort
/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/host0/target0:0:0/0:0:0:0/scsi_disk/0:0:0:0/provisioning_mode:full
pi@DeskPi:~ $ sudo echo unmap > /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/host0/target0:0:0/0:0:0:0/scsi_disk/0:0:0:0/provisioning_mode
bash: /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/host0/target0:0:0/0:0:0:0/scsi_disk/0:0:0:0/provisioning_mode: **Permission denied**

I found I needed the "sudo" ???  (better example ??)

steve

jzazo, 2021-05-28:
Try `sudo su` to enter root prompt, and repeat without sudo. Make sure you copy the line from the terminal.

nr3i, 2021-05-30:
> Try `sudo su` to enter root prompt, and repeat without sudo. Make sure you copy the line from the terminal.

Thanks.  That worked.  NEXT PROBLEM...  unmap didn't work??  But, "...mode:unmap" shows?

steve

pi@DeskPi:~ $ sudo su
root@DeskPi:/home/pi# find /sys/ -name provisioning_mode -exec grep -H . {} + | sort
/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/host0/target0:0:0/0:0:0:0/scsi_disk/0:0:0:0/provisioning_mode:full
root@DeskPi:/home/pi# echo unmap > /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/host0/target0:0:0/0:0:0:0/scsi_disk/0:0:0:0/provisioning_mode
root@DeskPi:/home/pi# find /sys/ -name provisioning_mode -exec grep -H . {} + | sort
/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/host0/target0:0:0/0:0:0:0/scsi_disk/0:0:0:0/**provisioning_mode:unmap**
root@DeskPi:/home/pi# sg_vpd -p bl /dev/sda
Block limits VPD page (SBC):
  Write same non-zero (WSNZ): 0
  Maximum compare and write length: 0 blocks [Command not implemented]
  Optimal transfer length granularity: 1 blocks
  Maximum transfer length: 65535 blocks
  Optimal transfer length: 65535 blocks
  Maximum prefetch transfer length: 65535 blocks
  Maximum unmap LBA count: 0 [Unmap command not implemented]
  **Maximum unmap block descriptor count: 0 [Unmap command not implemented]**
  Optimal unmap granularity: 0 blocks [not reported]
  Unmap granularity alignment valid: false
  Unmap granularity alignment: 0 [invalid]
  Maximum write same length: 0 blocks [not reported]
  Maximum atomic transfer length: 0 blocks [not reported]
  Atomic alignment: 0 [unaligned atomic writes permitted]
  Atomic transfer length granularity: 0 [no granularity requirement
  Maximum atomic transfer length with atomic boundary: 0 blocks [not reported]
  Maximum atomic boundary size: 0 blocks [can only write atomic 1 block]
root@DeskPi:/home/pi#

jzazo, 2021-05-31:
Your Kingston drive appears to be in `/dev/sdb`, from the `lsscsi` command you posted above. Try `lsblk -D` to list the available devices and confirm if your usb drive is in `\dev\sda` or `\dev\sdb`. If that's not the problem, I am not sure why you don't see larger unmap blocks.

nr3i, 2021-06-05:
NEXT TRY...
I still think that IF I can plug a USB3 to SSD cable with disk into the USB3 port and have it work - I should be able to plug the DeskPi SSD into the same port and have it work, TOO.

I did find on https://wiki.geekworm.com/X862_V1.0 a comment that "they" were having an issue too ?? as it says: "X862 V1.0 version supports TRIM but not UASP, V2.0 version supports UASP but not TRIM."  The X862 is a M.2 SSD simular to the DeskPi M.2 SSD and adapter board.

THAT Said...  I am going to try again AND make sure I copy the commands...
I DO like the idea of the DeskPi, the PROGRAMMABLE Ice Tower cooling, and the M.2 SSD.

For those that HAVE the M.2 SSD working - what brand/size/model are you using and web?  I would like to get the DeskPi Pro working @deskpi ...

Off to slay the dragon.  It has been a learning experience.  A good thing.
steve

nr3i, 2021-06-05:
Thank you.

I will try again.

steve

yoyojacky, 2021-06-30:
> DeskPi Pro with update boards. Main issue: SSD not working.
> (with the OLD boards, I never got the SSD to work, either)
>
> BUT, I could remove the SSD from DeskPi and using a USB3 to SATA cable, it [...]

Did you get the preinstall OS TF card and it out of work ?

yoyojacky, 2021-06-30:
> $ lsusb -t
> /: Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
> |__ Port 1: Dev 3, If 0, Class=Mass Storage, Driver=usb-storage, 5000M <- is the DeskPi M.2 SSD
> |__ Port 2: Dev 4, If 0, [...]

If you want to clean the TF card, you need to use this software `SD card formatter`  due to windows OS can not read `ext4` filesystem on the TF card, and if you can not flash image to the TF card, please just using `etcher` instead. 
https://www.balena.io/etcher/

yoyojacky, 2021-06-30:
> 4c:55aa ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge

Did you connect the `USB adapter` to USB3.0 port on Raspberry Pi? it will connect the daughter board to your disk. please make sure installation is ok and try again, when you execute `lsblk` it should be show your disk such as "/dev/sda" or something else, if not, means your disk has not been detect and recognized yet. please try to reboot your Raspberry Pi again.

yoyojacky, 2021-06-30:
> Hello @yoyojacky @deskpi
> I am not getting very far... I am trying TRIM, but only get so far and then ERROR and/or I am not able to figure out how to do the edit.
>
> I ran these commands with the M.2 [...]

please just using this command :  `sudo sh -c 'echo unmap > /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/host0/target0:0:0/0:0:0:0/scsi_disk/0:0:0:0/provisioning_mode' `

yoyojacky, 2021-06-30:
> Your Kingston drive appears to be in `/dev/sdb`, from the `lsscsi` command you posted above. Try `lsblk -D` to list the available devices and confirm if your usb drive is in `\dev\sda` or `\dev\sdb`. If that's not the problem, I am not sure why you don't see larger unmap blocks.

please read this and follow the steps : https://github.com/DeskPi-Team/deskpi/blob/master/DeskPi_v3_firmware_upgrade_manual.md

yoyojacky, 2021-06-30:
> It has been a learning experience. A good thing.
> steve

Hi Steve, 
Have you make your deskpi work properly ?  Seems like there some different between your disk and mine, mine is working properly, and my disk is `Kingston A400 M.2 2280 SATA Rev.3.0(6Gb/s) 120GB` and it works perfect, no matter what in `usb booting` or just file storaging , and it works fine. you mentioned before about the TF card out of working ?

closed 2021-10-13 by yoyojacky

PiersAll, 2022-05-04:
Raspbarian 4 64bit installed on Deskpi running well. EXCEPT when SSD card connected wont boot or crashes when USB jumper installed. Also note that plugging in jumper apparently draws excessive current as low voltage warning appears. Power supply Raspberry Pi USB-C 5.1volt 3A. SSD card NeTec NVME M.2 Gen 3X4.  

AhHa found a note NOT SUPPORT M.2 NVME SSD just m.2 SATA SSD Bugger by a SATA SSD!!!!!!!!!!!!!

yoyojacky, 2022-05-23:
> card NeTec NVME M.2 Gen 3X4.

Yes, that's right, this version does not support M.2 NVME but M.2 SATA will work fine.

cal1br8tr, 2023-07-10:
I just bought a deskpi-pro with a m2 NVME SSD adapter card but it has the same problem. If I plug the same drive in to the pi via a USB adapter then it boots fine. If going through the USB U-link, all I see is: 

/: Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
|__ Port 1: Dev 3, If 0, Class=Mass Storage, Driver=usb-storage, 5000M

did the flash update on the adapter, checked the physical. Using the disks app it sees the drive but states no media. 

The ssd board shows a white light but only when the USB U-link is plugged in. What is the purpose of the touch points to the gpio pins?

When using a standard deskpi case with a 5V 4A 20W power supply, I still get the voltage error.

Couple of different issues but seems like this product is going in the junk drawer.