DeskPi-Team/deskpi #44: DeskPi Upgrade Boards - Instructions unclear

jacobacci, 2021-05-21:
I have received my upgrade board, thanks.
In the package this item was included:

[image]

However it is unclear to me where this fits. One possibility would be like this:

[image]

Where is the correct location to fit this?
Thanks

CrunchyDoodle, 2021-05-21:
You're close. I believe it goes to the left between the USB ports. I watched a couple of YouTube videos where it was demonstrated how to install it.

jacobacci, 2021-05-23:
Thanks, I got a response from pi-mania.com.au which confirmed the following location

[image]

CrunchyDoodle, 2021-05-23:
I hate belaboring this, and here are the instructions you should have received with the upgrade.
[image]

jacobacci, 2021-05-24:
Thanks @CrunchyDoodle 
You are not belaboring anything. You actually gave me the correct answer. I was a bit doubtful myself, as I assumed the T-piece is conductive and it would be strange to put it in contact with the solder on the bottom of the board. I will put it in correctly today.
@yoyojacky Could you possibly post the complete instructions somewhere on the DeskPi website? Thank you

TeoTN, 2021-05-30:
The docs should mention that the previous (black) USB bridge isn't compatible with the new board, and you have to use the new USB bridge (gray)... That would save me quite some time

jojobrogess, 2021-05-30:
AFAIK my old black one from v1 is working. I'll know more tonight when I try to enable TRIM. But when I put everything together initially, I used the black one and my SSD booted up without problems. What is different about it specifically ??

TeoTN, 2021-05-31:
For me it wouldn't boot, throwing loads of errors on boot up time, many services (Disk Manager, Account Service, containerd, OpenBSD SSH server, Snap Daemon) crashing and eventually a long (like, really long) list of errors while accessing /dev/sda would appear on the screen and eventually kernel would go to tainted state.

jojobrogess, 2021-06-01:
I got mine installed and working correctly without any issues, yet.. And I was able to use either bridge interchangeably without issue. Could be something else happening with your setup.

yoyojacky, 2021-06-30:
The T-type anti-static cotton should be inserted between USB3.0 and USB2.0, this is to solve the problem of the Raspberry Pi USB interface grounding.

yoyojacky, 2021-06-30:
> 谢谢@CrunchyDoodle
> 你没有在做任何事情。你实际上给了我正确的答案。我自己有点怀疑，因为我认为 T 型件是导电的，将它与电路板底部的焊料接触会很奇怪。我今天会正确地把它放进去。
> @yoyojacky您能否在 DeskPi 网站上的某处发布完整的说明？谢谢

OK, we are working on it.

closed 2021-06-30 by yoyojacky