DeskPi-Team/deskpi #45: Defective board?

jacobacci, 2021-05-23:
Hi all
I have installed the new board and updated the FW to V3. Everything seems to work, my 1TB Crucial MX500 SSD is detected as UASP and works ok.
However two things don't work:
- the two USB2 ports on the front seem to be non functional
- when I power down the RP4 via software, the power button on the enclosure stays blue. Am I right in assuming it should turn red once the RP4 has actually powered down?
Do these two issues point to a HW defect on the board?

ddb157, 2021-05-25:
Did you run the install.sh script?  It adds overlay entries to activate the USB ports.

Pi-Mania, 2021-06-17:
Depending on the O/S you are running the scripts may not work. If not,  Just edit/add: **dtoverlay=dwc,dr_mode=host** to config.txt file should fix problem. Did for me. Thanks

closed 2021-06-30 by yoyojacky