DeskPi-Team/deskpi #46: Help enabling trim

jzazo, 2021-05-24:
Hi! I have downloaded the new [firmware](https://deskpi.com/blogs/news/about-uasp-trim) and successfully installed it. 
I have tried then activating Trim following the steps provided, and after changing `provisioning_mode` to `unmap` successfully, and adding the `discard_max_bytes` setting to `/sys/block/sda/queque/discard_max_bytes`, TRIM fails with
```
fstrim -v /
fstrim: /: the discard operation is not supported
```
@bpicon, I noticed that you succedeed on this operation in another post. Can you help me? Did you experience similar result?

bpicon, 2021-05-25:
> @bpicon, I noticed that you succedeed on this operation in another post. Can you help me? Did you experience similar result?

Activating TRIM has not given me any problems.
If you want you can put the output of the following commands to try to see where the error is:
(Assuming /dev/sda)
.- lsusb
.- lsusb -t
.- lsblk --discard
.- find /sys/ -name provisioning_mode -exec grep -H . {} + | sort
.- sg_vpd -p bl /dev/sda
.- sg_readcap -l /dev/sda
.- cat /sys/block/sda/queque/discard_max_bytes
.- cat /etc/udev/rules.d/10-trim.rules

jzazo, 2021-05-25:
After reboot, `provisioning_mode` switches to full, and `discard_max_bytes` to 0. When I change them, TRIM does not work.

I wonder if the instructions fail because I am booting from the ssd / usb, should I boot from the sdcard?

Here is the output of commands:
```
# lsusb
Bus 003 Device 003: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Bus 003 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 174c:55aa ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

# lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=dwc2/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=ch341, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Mass Storage, Driver=uas, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M

# lsblk -t
NAME        ALIGNMENT MIN-IO   OPT-IO PHY-SEC LOG-SEC ROTA SCHED       RQ-SIZE    RA WSAME
sda                 0    512 33553920     512     512    0 mq-deadline      58 65532   32M
├─sda1              0    512 33553920     512     512    0 mq-deadline      58 65532   32M
├─sda2              0    512 33553920     512     512    0 mq-deadline      58 65532   32M
│ └─ssd             0    512 33553920     512     512    0                 128 65532    0B
└─sda3              0    512 33553920     512     512    0 mq-deadline      58 65532   32M
  └─home            0    512 33553920     512     512    0                 128 65532    0B
mmcblk0             0    512        0     512     512    0 mq-deadline     128   128    0B
├─mmcblk0p1         0    512        0     512     512    0 mq-deadline     128   128    0B
└─mmcblk0p2         0    512        0     512     512    0 mq-deadline     128   128    0B

# lsblk --discard
NAME        DISC-ALN DISC-GRAN DISC-MAX DISC-ZERO
sda                0        0B       0B         0
├─sda1             0        0B       0B         0
├─sda2             0        0B       0B         0
│ └─ssd            0        0B       0B         0
└─sda3             0        0B       0B         0
  └─home           0        0B       0B         0
mmcblk0            0        4M   158.8G         0
├─mmcblk0p1        0        4M   158.8G         0
└─mmcblk0p2        0        4M   158.8G         0

# find /sys/ -name provisioning_mode -exec grep -H . {} + | sort
/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/host0/target0:0:0/0:0:0:0/scsi_disk/0:0:0:0/provisioning_mode:unmap

# sg_vpd -p bl /dev/sda
Block limits VPD page (SBC):
  Write same non-zero (WSNZ): 0
  Maximum compare and write length: 0 blocks [Command not implemented]
  Optimal transfer length granularity: 1 blocks
  Maximum transfer length: 65535 blocks
  Optimal transfer length: 65535 blocks
  Maximum prefetch transfer length: 65535 blocks
  Maximum unmap LBA count: 4194240
  Maximum unmap block descriptor count: 1
  Optimal unmap granularity: 1 blocks
  Unmap granularity alignment valid: false
  Unmap granularity alignment: 0 [invalid]
  Maximum write same length: 0 blocks [not reported]
  Maximum atomic transfer length: 0 blocks [not reported]
  Atomic alignment: 0 [unaligned atomic writes permitted]
  Atomic transfer length granularity: 0 [no granularity requirement
  Maximum atomic transfer length with atomic boundary: 0 blocks [not reported]
  Maximum atomic boundary size: 0 blocks [can only write atomic 1 block]

# cat /sys/block/sda/queue/discard_max_bytes
2147450880

# cat /etc/udev/rules.d/10-trim.rules
ACTION=="add|change", ATTRS{idVendor}=="174c", ATTRS{idPro duct}=="55aa", SUBSYSTEM=="scsi_disk", ATTR{provisioning_m ode}="unmap"
```

Thanks a lot for your help!

PD: there is a typo in the pdf instructions, in `cat /sys/block/sda/queque/discard_max_bytes`, should be `queue`, other than that I have been following the instructions to the letter.

bpicon, 2021-05-25:
> # cat /etc/udev/rules.d/10-trim.rules
ACTION=="add|change", ATTRS{idVendor}=="174c", ATTRS{idPro duct}=="55aa", SUBSYSTEM=="scsi_disk", ATTR{provisioning_m ode}="unmap"

I see some blank space that shouldn't be: 
.- {idPro duct}  --> {idProduct}
.- {provisioning_m ode} --> {provisioning_mode}

jzazo, 2021-05-25:
Thanks for noticing the spaces! After fixing them and running `lsblk --discard` I now get:
```
# lsblk -D
NAME        DISC-ALN DISC-GRAN DISC-MAX DISC-ZERO
sda                0      512B       2G         0
├─sda1             0      512B       2G         0
├─sda2             0      512B       2G         0
│ └─ssd            0        0B       0B         0
└─sda3             0      512B       2G         0
  └─home           0        0B       0B         0
mmcblk0            0        4M   158.8G         0
├─mmcblk0p1        0        4M   158.8G         0
└─mmcblk0p2        0        4M   158.8G         0
```

Still `fstrim -v /` says operation is not supported. Is a configuration problem or something else? Is it the deskpi pro board or the ssd?

jzazo, 2021-05-25:
I see that TRIM is disabled by default on encrypted partitions and it seems I need to enable it when decrypting the partitions:
https://askubuntu.com/a/115850
https://askubuntu.com/questions/59519/do-you-recommend-luks-encryption-on-a-ssd-trim-support

I will read a bit about enabling it to understand the implications. Apart from that issue, TRIM support seems plausible to you with previous output? Thank you! 

(I will keep the issue open until I figure out what to do).

jzazo, 2021-05-25:
I finally managed to set it up by activating TRIM in `/etc/crypttab` and `/etc/fstab`. I was surprised that the instructions did not mention the `fstab` edit, do you know why? Is it because by default it is active? (I did not try without it)

Thanks a lot for you help!

closed 2021-05-25 by jzazo