DeskPi-Team/deskpi #47: shutdown works, reboot hangs

ddb157, 2021-05-25:
Deskpi is hanging on reboot.

Steps taken:

Power status jumper is in default setting.

Disabled all installed services such as databases and webgui.

Set /etc/systemd/journald.conf to persistent logging.

Using:

sudo systemctl reboot

System hangs, deskpi has blue power light, intermittent white flashing behind case ?HD activity light?

Hold power button 3s and system turns off, fan turns off.  

Tap power button, system boots.

# journalctl -r -b-1

May 25 09:25:51 smarthome-hub systemd-journald[109]: Journal stopped
May 25 09:25:51 smarthome-hub systemd-shutdown[1]: Sending SIGTERM to remaining processes...
May 25 09:25:51 smarthome-hub systemd-shutdown[1]: Syncing filesystems and block devices.
May 25 09:25:51 smarthome-hub kernel: printk: systemd-shutdow: 38 output lines suppressed due to ratelimiting
May 25 09:25:51 smarthome-hub systemd[1]: Failed to set timeout to 600s: Invalid argument
May 25 09:25:51 smarthome-hub systemd[1]: Hardware watchdog 'Broadcom BCM2835 Watchdog timer', version 0
May 25 09:25:51 smarthome-hub systemd[1]: Shutting down.
May 25 09:25:51 smarthome-hub systemd[1]: Reached target Reboot.

But deskpi stays in stalled state.  Needs to be manually power cycled.

systemctl shutdown results in Deskpi powered down with red light on power button.

If I can't reboot system for unattended upgrades, I'll have to toss the Deskpi pro.

ddb157, 2021-05-25:
On the last working shutdown I got these entries in the log:

```
May 25 10:09:11 smarthome-hub systemd[1]: dhcpcd.service: Succeeded.
May 25 10:09:10 smarthome-hub systemd[1]: Failed to start DeskPi Safeshutdown Service.
May 25 10:09:10 smarthome-hub systemd[1]: deskpi-safeshut.service: Failed with result 'timeout'.
May 25 10:09:10 smarthome-hub systemd[1]: deskpi-safeshut.service: Main process exited, code=killed, status=15/TERM
May 25 10:09:10 smarthome-hub sudo[587]: pam_unix(sudo:session): session closed for user root
May 25 10:09:10 smarthome-hub systemd[1]: deskpi-safeshut.service: Start operation timed out. Terminating.
May 25 10:09:10 smarthome-hub kernel: ieee80211 phy0: brcmf_fweh_event_worker: event handler failed (69)
May 25 10:09:10 smarthome-hub kernel: ieee80211 phy0: brcmf_cfg80211_escan_handler: scan not ready, bsscfgidx=0
```

Ironically, I think the safeshutdown service may be the culprit.

ddb157, 2021-05-25:
For comparison, this resulted in a hanged system needing physical intervention:

```
May 25 10:13:29 smarthome-hub sudo[585]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/systemctl reboot
May 25 10:13:29 smarthome-hub sudo[585]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
May 25 10:13:29 smarthome-hub systemd-logind[353]: System is rebooting.

May 25 10:13:29 smarthome-hub systemd[1]: Condition check resulted in Turns off Raspberry Pi display backlight on shutdown/reboot being skipped.
May 25 10:13:29 smarthome-hub systemd[1]: Stopped target Graphical Interface.
May 25 10:13:29 smarthome-hub systemd[1]: Stopping DeskPi PWM Control Fan Service...
May 25 10:13:29 smarthome-hub sudo[550]: pam_unix(sudo:session): session closed for user root
May 25 10:13:29 smarthome-hub systemd[1]: deskpi.service: Main process exited, code=killed, status=15/TERM
May 25 10:13:29 smarthome-hub systemd[1]: deskpi.service: Succeeded.
May 25 10:13:29 smarthome-hub systemd[1]: Stopped DeskPi PWM Control Fan Service.

May 25 10:13:30 smarthome-hub systemd[1]: Reached target Shutdown.
May 25 10:13:30 smarthome-hub systemd[1]: Reached target Final Step.
May 25 10:13:30 smarthome-hub systemd[1]: systemd-reboot.service: Succeeded.
May 25 10:13:30 smarthome-hub systemd[1]: Started Reboot.
May 25 10:13:30 smarthome-hub systemd[1]: Reached target Reboot.
May 25 10:13:30 smarthome-hub systemd[1]: Stopping Restore / save the current clock...
May 25 10:13:30 smarthome-hub systemd[1]: Shutting down.
May 25 10:13:31 smarthome-hub systemd[1]: Hardware watchdog 'Broadcom BCM2835 Watchdog timer', version 0
May 25 10:13:31 smarthome-hub systemd[1]: Failed to set timeout to 600s: Invalid argument
May 25 10:13:31 smarthome-hub kernel: printk: systemd-shutdow: 38 output lines suppressed due to ratelimiting
May 25 10:13:31 smarthome-hub systemd-shutdown[1]: Syncing filesystems and block devices.
May 25 10:13:31 smarthome-hub systemd-journald[123]: Journal stopped
```

I have no idea where the 600s timeout is coming from.

ddb157, 2021-05-25:
When examining:
/usr/lib/systemd/system/deskpi-safeshut.service the command below is executed:

sudo /usr/bin/fanStop

When I enter it manually, the system hangs.  I don't use a watchdog so that may be why it's working for some.

ddb157, 2021-05-25:
To those who want a quick fix:

sudo systemctl stop deskpi-safeshut
sudo systemctl disable deskpi-safeshut
sudo systemctl mask deskpi-safeshut

As of this moment rebooting works but I guess my shutdown is no longer safe.

yoyojacky, 2021-06-30:
This is a script that executes a power_off command to /dev/ttyUSB0 in the shutdown sequence. It only sends a power-off command to the MCU of the daughter board during shutdown. It has nothing to do with reboot. You can check your /lib/ The corresponding deskpi server script in systemd/system/, where the comflict part is used, is forbidden to take effect during reboot.

yoyojacky, 2021-06-30:
Do not worry about shutdown is still safe due to everything is shutting down and then pi will send `power_off` to /dev/ttyUSB0 so that the MCU can cut off the power after 5 seconds after received the signal, raspberry pi os should turned off by that moment....

pkr1771, 2021-08-29:
I encountered the issue of the power LED not going red on halt/shutdown/poweroff which also traced back to fanStop not functioning correctly.  I'm running on 64-bit Ubuntu which I was happy to see now has installer support.  The pwmFanControl works correctly (unlike previous releases) because it is recompiled during the installation process.

I needed to manually recompile the fanStop.c using gcc for the 64 bit platform and then copy the new binary to the /usr/bin directory.  This file is not recompiled in the installation process and appears that it needs to be added.

One other difference I noted from earlier releases is the definition of the deskpi-safeshut.service.  I can't say for sure that this makes a difference, but I noticed it as a change from earlier releases that included the line:

`TimeoutSec=1`

When I made these two changes, the system shuts down as expected and the power LED goes red.

yoyojacky, 2021-10-13:
64bit system need to recompile the C code .....

closed 2021-10-13 by yoyojacky