DeskPi-Team/deskpi #48: Enabling TRIM. Max_discard_bytes not persistent after reboot

bpicon, 2021-05-25:
I have verified that when activating TRIM, the value 2147450880 assigned to 'discard_max_bytes', according to the manual, does not remain after reboot:
root@raspberrypi:/home/pi# cat /sys/block/sda/queue/discard_max_bytes
4294966784
It can also be seen in the output of the lsblk --discard command, where 4G appears instead of 2G

In my case I have created a new udev rule (99-maxdiscard.rules) to keep this value: 
ACTION == "add | change", KERNEL == "sda", SUBSYSTEM == "block", ATTR {queue / discard_max_bytes} = "2147450880"

In the manual it does not say anything about it. If I'm not mistaken, I think that to avoid problems this attribute should be kept at its correct value.
I don't know if anyone else has been able to verify this problem.

jzazo, 2021-05-25:
I experience the same issue as you describe it.

yoyojacky, 2021-06-30:
This number should be calculated based on your disk capacity. . .

yoyojacky, 2021-06-30:
Please read this :https://github.com/DeskPi-Team/deskpi/blob/master/DeskPi_v3_firmware_upgrade_manual.md

amoscaritola, 2022-01-07:
@bpicon Thank you, I had the same issue and your post helped me out.