DeskPi-Team/deskpi #49: Fan doesn't work

TeoTN, 2021-05-28:
The fan doesn't work. I'm using Raspberry Pi 4 8GB, booting from SSD with Ubuntu 21.04 64-bit. I'm using the newest PCB board you've sent me.
The error is:
```
root@ubuntu:/var/log# systemctl status deskpi.service
● deskpi.service - DeskPi PWM Control Fan Service
     Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2021-05-28 13:37:19 UTC; 27min ago
   Main PID: 6474 (code=exited, status=1/FAILURE)

May 28 13:37:19 ubuntu systemd[1]: Started DeskPi PWM Control Fan Service.
May 28 13:37:19 ubuntu sudo[6474]:     root : PWD=/ ; USER=root ; COMMAND=/usr/bin/pwmFanControl
May 28 13:37:19 ubuntu sudo[6474]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 28 13:37:19 ubuntu sudo[6474]: pam_unix(sudo:session): session closed for user root
May 28 13:37:19 ubuntu sudo[6474]: sudo: unable to execute /usr/bin/pwmFanControl: No such file or directory
May 28 13:37:19 ubuntu systemd[1]: deskpi.service: Main process exited, code=exited, status=1/FAILURE
May 28 13:37:19 ubuntu systemd[1]: deskpi.service: Failed with result 'exit-code'.
```

It seems the fan controller is broken:
```
root@ubuntu:/var/log# ls -la /usr/bin/pwmFanControl
-rwxr-xr-x 1 root root 8880 May 28 13:35 /usr/bin/pwmFanControl
root@ubuntu:/var/log# /usr/bin/pwmFanControl
-bash: /usr/bin/pwmFanControl: No such file or directory
```

TeoTN, 2021-05-29:
Even if I compile the drivers manually, or if I replace `/dev/ttyUSB0` with `/dev/ttyUSBFAN` in both C and Python sources, as someone suggested, it simply doesn't work.

bpicon, 2021-05-31:
> Even if I compile the drivers manually, or if I replace /dev/ttyUSB0 with /dev/ttyUSBFAN in both C and Python sources, as someone suggested, it simply doesn't work.

Verify that the entry:
 dtoverlay=dwc2,dr_mode=host 
was created in /boot/firmware/config.txt
Make sure there are no blank spaces on either side of the comma or elsewhere within the string..

TeoTN, 2021-05-31:
Yup, it's been there. I gave up yesterday and flashed the Pi with Raspbian 32b... That "solved" the problem with fan, but the setup keeps crashing tho. All of a sudden the device would become unavailable every few hours, I can't ssh to it, all services are down and I have to power cycle it. 😡

ddetton, 2021-06-21:
> In my case, for the fan to work in Ubuntu 64 bits I had to modify the installation .sh and recompile the .c of 'deskpi/drivers/c'. I also took the opportunity to modify the source pwmControlFan.c to implement the temperature and fan speed values ​​in the code and not use the '/etc' setting via deskpi-config.

Recompiling the c code got the fan working for me too.  Thanks for the heads up on that!  What change did you make to the installation.sh file?  Can you post your enhanced pwmFanControl?  Thanks!

yltsa, 2021-06-25:
Hi, please can you make how-to instructions for making the compilation and the installation to Ubuntu 64 bit. I a beginner in these things.. Ubuntu 64 bit works but no fan or the front panel USB ports in DeskPi pro (the newest model with 40 pin GPIO connector in the backplane).. Thank you very much..

ddetton, 2021-06-25:
Sure, no problem.  Just enter the following commands in your RPi command line.  This assumes that deskpi has been cloned to the deskpi folder of the logged in user and either the install.sh or the relevant platform specific install-xxx.sh script has been run using sudo.

```
cd ~/deskpi/drivers/c
rm pwmFanControl
make
sudo cp pwmFanControl /usr/bin
cd ~/deskpi
```
Now enter `./deskpi-config` and you should get a prompt to configure your fan control.  This worked for me, hope it works for you too.

yltsa, 2021-06-25:
Hi, thank you. I could compile the stuff and run deskpi-config. But the fan will not start. 

The fan works in 32 bit Raspian and 64 bit Home assistant.. but not in 64 bit Ubuntu. something is wrong..

yltsa, 2021-06-26:
Good morning, it works now. Maybe a restart was the thing, because DeskPi was closed during the night. Thank you very much.. this is great.

yoyojacky, 2021-06-30:
Please enter into the repository folder and execute `git pull` to up2date. 
and i have already put the 64bit binary file into drivers/c folder, and also you can just install `gcc` and compile the code by yourself.
it's open source code, change it and share it or just have fun~

closed 2021-06-30 by yoyojacky

johntdavis84, 2021-08-23:
> Sure, no problem. Just enter the following commands in your RPi command line. This assumes that deskpi has been cloned to the deskpi folder of the logged in user and either the install.sh or the [...]

@ddetton Hello! Thanks for this. :)

Will these instructions work on 64-bit Manjaro, as well? The Manjaro install script isn't working for me at all, beyond activating the front USB ports.

Thanks again!

ddetton, 2021-08-23:
Quote: Will these instructions work on 64-bit Manjaro, as well? The Manjaro install script isn't working for me at all, beyond activating the front USB ports.

Sorry, no idea!  I'm a Linux nube and I have never used Manjaro.  Best of luck to you!

johntdavis84, 2021-08-24:
> Quote: Will these instructions work on 64-bit Manjaro, as well? The Manjaro install script isn't working for me at all, beyond activating the front USB ports.
> 
> Sorry, no idea! I'm a Linux nube and I have never used Manjaro. Best of luck to you!

No need to apologize. I appreciate the help. :)

I'm running into some issues when I try the method you used.

When you're in the `/drivers/c/` folder, could you do a `cat Makefile` and paste the output here?

Mine only makes the `safecutoffpower` app.
```
]$ cat Makefile 
safecutoffpower:  safecutoffpower.c 
        gcc -o safecutoffpower safecutoffpower.c
clean:
        rm -rf safecutoffpower
```

ddetton, 2021-08-24:
Here you go:
```
pwmFanControl:  pwmControlFan.c 
	gcc -o pwmFanControl pwmControlFan.c 
	gcc -o fanStop fanStop.c
clean:
	rm -rf pwmFanControl fanStop
```

johntdavis84, 2021-08-24:
Thanks!

My copy of the Makefile, pulled this week, doesn’t have any of those. I’m not sure what’s going on.

At this point, I’ve given up on the scripts and tried to figure out how to do a manual install. I’m almost there, I think, though I’m still not sure what the safe shutoff service is supposed to do. I’m guessing it makes sure an abrupt power loss doesn’t hurt the DeskPi hardware?

In any case, if I can actually get it working, I’ll write up some directions.

-- -- --
John T Davis
***@***.***

yoyojacky, 2021-10-14:
> Thanks! My copy of the Makefile, pulled this week, doesn’t have any of those. I’m not sure what’s going on. At this point, I’ve given up on the scripts and tried to figure out how to do a manual [...]

the shutoff service is to send `power_off`  signal to MCU and let MCU cut off the power from Rapberry Pi .