DeskPi-Team/deskpi #50: Unable to control Fan Speed using replacement PCB

MadLunatic42, 2021-05-30:
Hello, 

I'm running the latest raspberry pi OS and downloaded the latest deskpi package, which installed correctly.
However, I'm no longer able to control the fan speed of my fan, which was working on the previous PCB (in fact, if I plug in the previous PCB again I regain control of the fan speed).

The fan is always running at 100% regardless of the speed setting.

The required service and USB device appears to be available.

raspi4-8:~ $ systemctl status deskpi.service
● deskpi.service - DeskPi PWM Control Fan Service
   Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2021-05-30 17:58:45 CEST; 3s ago
 Main PID: 1369 (sudo)
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/deskpi.service
           ├─1369 /usr/bin/sudo /usr/bin/pwmFanControl
           └─1370 /usr/bin/pwmFanControl

May 30 17:58:45 raspi4-8 systemd[1]: Started DeskPi PWM Control Fan Service.
May 30 17:58:45 raspi4-8 sudo[1369]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/pwmFanControl
May 30 17:58:45 raspi4-8 sudo[1369]: pam_unix(sudo:session): session opened for user root by (uid=0)
raspi4-8:~ $ ls /dev/ttyUSB0 
/dev/ttyUSB0
raspi4-8:~ $ dmesg | grep -i ttyusb
[    9.738267] usb 3-1.1: ch341-uart converter now attached to ttyUSB0

Any pointers to troubleshoot this behaviour further ?

yoyojacky, 2021-06-30:
please check if file exist in `/etc/deskpi.conf`

yoyojacky, 2021-06-30:
and you can try to control you fan by using python script which in the `driver` folder of the repository.

MadLunatic42, 2021-07-04:
Thanks for your feedback, @yoyojacky 

- ```/etc/deskpi.conf``` is available and configured.
- running the "pwmControlFan.py" manually does not change anything. I even modified the python script to output some text whenever the CPU temperature is below 40 - it perfectly outputs my debugging info, it does not change the fan speed. It's like the command "pwm_000" is sent to the serial port without any effect.

MadLunatic42, 2021-08-03:
Closing as I no longer own the deskpi pro. As there is no clear solution or feedback I expect the issue is due to a hardware problem on the PCB.

closed 2021-08-03 by MadLunatic42