DeskPi-Team/deskpi #51: DeskPi Pro V3 Firmware update Manual-ReadmeFirst - Needs Work/Fixing

Pi-Mania, 2021-06-17:
Hi.

Having just run through this (successfully, as i know just enough to spot the errors) i can see many getting seriously stuck/confused due to many spelling errors missing commands etc. Some files simply cannot be edited as stated due to permissions (I used >sudo thunar and found/edited them manually) So it does work, just needs some checking/clean-up and will save you a heap of email I'm sure. Thanks, Drew

TeoTN, 2021-06-24:
It would be nice the manual was actually written in markdown, so that people can copy commands and paste them. After all, images can also be embedded in markdown.
PDF also would do but probably editing it so that commands can be copied would be more difficult.

yoyojacky, 2021-06-30:
Okay, thank you for your suggestion. I will prepare a readme.md file immediately.

closed 2021-06-30 by yoyojacky

yoyojacky, 2021-06-30:
https://github.com/DeskPi-Team/deskpi/blob/master/DeskPi_v3_firmware_upgrade_manual.md