DeskPi-Team/deskpi #52: Installation script not working on Manjaro ARM Minimal 21.06 64-bit OS

ivanmarban, 2021-06-21:
**Linux raspberrypi4 5.10.43-1-MANJARO-ARM #1 SMP PREEMPT Mon Jun 14 14:31:41 UTC 2021 aarch64 GNU/Linux**

```shell
cd ~
git clone https://github.com/DeskPi-Team/deskpi.git
cd ~/deskpi/
chmod +x install-manjaro.sh
sudo ./install-manjaro.sh
Cloning into 'deskpi'...
remote: Enumerating objects: 373, done.
remote: Counting objects: 100% (40/40), done.
remote: Compressing objects: 100% (40/40), done.
remote: Total 373 (delta 19), reused 0 (delta 0), pack-reused 333
Receiving objects: 100% (373/373), 5.39 MiB | 9.21 MiB/s, done.
Resolving deltas: 100% (193/193), done.
./install-manjaro.sh: line 3: /lib/lsb/init-functions: No such file or directory
./install-manjaro.sh: line 12: log_action_msg: command not found
./install-manjaro.sh: line 25: log_action_msg: command not found
./install-manjaro.sh: line 26: cd: /home/user/deskpi/drivers/c/: No such file or directory
cp: cannot stat '/home/user/deskpi/drivers/c/pwmFanControl': No such file or directory
cp: cannot stat '/home/user/deskpi/drivers/c/fanStop': No such file or directory
chmod: cannot access '/usr/bin/pwmFanControl': No such file or directory
chmod: cannot access '/usr/bin/fanStop': No such file or directory
cp: cannot stat '/home/user/deskpi/deskpi-config': No such file or directory
chmod: cannot access '/usr/bin/deskpi-config': No such file or directory
cp: cannot stat '/home/user/deskpi/Deskpi-uninstall': No such file or directory
chmod: cannot access '/usr/bin/Deskpi-uninstall': No such file or directory
./install-manjaro.sh: line 60: log_action_msg: command not found
./install-manjaro.sh: line 67: log_action_msg: command not found
Created symlink /etc/systemd/system/multi-user.target.wants/deskpi.service → /usr/lib/systemd/system/deskpi.service.
Created symlink /etc/systemd/system/halt.target.wants/deskpi-safeshut.service → /usr/lib/systemd/system/deskpi-safeshut.service.
Created symlink /etc/systemd/system/shutdown.target.wants/deskpi-safeshut.service → /usr/lib/systemd/system/deskpi-safeshut.service.
Created symlink /etc/systemd/system/poweroff.target.wants/deskpi-safeshut.service → /usr/lib/systemd/system/deskpi-safeshut.service.
Job for deskpi.service failed because the control process exited with error code.
See "systemctl status deskpi.service" and "journalctl -xeu deskpi.service" for details.
```
```shell
deskpi.service - DeskPi PWM Control Fan Service
     Loaded: loaded (/usr/lib/systemd/system/deskpi.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Mon 2021-06-21 22:50:48 CEST; 6min ago
    Process: 404 ExecStart=sudo /usr/bin/pwmFanControl & (code=exited, status=1/FAILURE)
   Main PID: 404 (code=exited, status=1/FAILURE)
        CPU: 45ms

Jun 21 22:50:48 raspberrypi4 systemd[1]: Starting DeskPi PWM Control Fan Service...
Jun 21 22:50:48 raspberrypi4 sudo[404]: sudo: /usr/bin/pwmFanControl: command not found
Jun 21 22:50:48 raspberrypi4 systemd[1]: deskpi.service: Main process exited, code=exited, status=1/FAILURE
Jun 21 22:50:48 raspberrypi4 systemd[1]: deskpi.service: Failed with result 'exit-code'.
Jun 21 22:50:48 raspberrypi4 systemd[1]: Failed to start DeskPi PWM Control Fan Service.
```
```shell
ls -la /usr/bin/pwmFanControl
ls: cannot access '/usr/bin/pwmFanControl': No such file or directory
```

yoyojacky, 2021-06-30:
bug fixed, please try to download the repository  and try again.

closed 2021-06-30 by yoyojacky

Marach79, 2021-07-08:
Does´t work for Manjaro XFCE.

Marach79, 2021-07-13:
> 
> 
> Does´t work for Manjaro XFCE.

My fault! It was the 64bit version.