DeskPi-Team/deskpi #54: DeskPi Pro Firmware version check

waltwright, 2021-07-01:
I ordered my brand new Deskpi Pro on June 13, 2021 through Seeed and receive SN21205367

1. If I follow the firmware upgrade procedure, do I need to disassemble the daughterboard from the rest of the assembly and the Raspberry Pi board first?
2. Is there a way to verify the current firmware version number, to be sure that I need to update the daughterboard?
3. Ultimately, I did NOT try to update the firmware yet.
4. I enabled TRIM successfully and adding the fstrim.timer. Does that mean that the firmware is current?
5. Is there a change log to see things like what V3 firmware added?

Thanks.

Pi-Mania, 2021-07-04:
Hi. I completed mine using a USBA-USBA cable. I left it in the case but, i removed my drives so no power drain there. If you look in the case, you will see a white LED once connected. Upgrade was quick and faultless. Hope that helps.

inspector71, 2021-08-01:
>     1. If I follow the firmware upgrade procedure, do I need to disassemble the daughterboard from the rest of the assembly and the Raspberry Pi board first?

This is also a question I would love answered. Having watched the video, it is not clear.

>     2. Is there a way to verify the current firmware version number, to be sure that I need to update the daughterboard?

There should be, again, @yoyojacky needs to respond, please.

>     3. Ultimately, I did NOT try to update the firmware yet.

Same here and I am uncertain if I need to because I am not aware of any changelog. The website leaves a lot to be desired.

> 
>     4. I enabled TRIM successfully and added the fstrim.timer. Does that mean that the firmware is current?
> 
>     5. Is there a change log to see things like what V3 firmware added?

There has to be, otherwise the product is not fit for purpose as we already paid for twice the bus speed we now have (6Mbps / 3Mpbs). I am not wanting to be too direct and harsh, but enough is enough of ESLanguage, foreign OEMs hyping up products then all but going missing after under-delivering. From what I can determine ... a) @yoyojacky needs a much better network connection whilst filming videos :) and we are owed a little more in support, notwithstanding the significant efforts so far in the form of the how-to PDF and so tutorial videos. They are more than some OEMs bother to do but that's a low standard of customer service and I feel we have a reasonable expectation to a higher standard.

closed 2021-10-13 by yoyojacky