DeskPi-Team/deskpi #57: Fan eventually stops even when configured not to

Exscotticus, 2021-07-09:
I have configured the fan to always run at 100%. This is because the fan makes rattling noises at slower speeds.

Unfortunately, at some point the fan stops, and I must repeatedly reinvoke option 7.

```
pi@deskpi:~ $ deskpi-config
Job for deskpi.service canceled.
[info] Welcome to Use DeskPi-Team's Product.
[info] Please select speed level that you want: .
[info] It will always run at the speed level that you choosed..
[info] ---------------------------------------------------------------.
[info] 1 - set fan speed level to 25%.
[info] 2 - set fan speed level to 50%.
[info] 3 - set fan speed level to 75%.
[info] 4 - set fan speed level to 100%.
[info] 5 - Turn off Fan.
[info] 6 - Adjust the start speed level according to the temperature.
[info] 7 - Cancel manual control and enable automatical fan control.
[info] ---------------------------------------------------------------.
[info] Just input the number and press enter..
Your choice:7
[info] Cancel manual control and enable automatical fan control.

pi@deskpi:~ $ cat /etc/deskpi.conf
40
100
50
100
60
100
70
100
```

jojobrogess, 2021-07-17:
I've had repeated problems with using anything other than options 1-4. 

Does setting the fan to always on at 100% not work for you?(option4) So that's why you used the custom settings?

Exscotticus, 2021-07-18:
If I choose option 4 and reboot, then no fan. I don't want to manually turn on the fan every time I reboot.

I haven't examined **deskpi-config** to see if I can hack it in my **/etc/rc.local** file.

arsirantala, 2021-08-08:
I have the same exact issue. Operating system is Ubuntu 20.04.2 LTS.

When I checked the deskpi service status, its:
```
$ sudo systemctl status deskpi.service                                                                                                                                        ─╯
● deskpi.service - DeskPi PWM Control Fan Service
     Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; vendor preset: enabled)
     Active: failed (Result: signal) since Sun 2021-08-08 20:25:18 EEST; 13min ago
   Main PID: 4394 (code=killed, signal=TERM)

Aug 08 20:25:01 raspberrypi4box systemd[1]: Starting DeskPi PWM Control Fan Service...
Aug 08 20:25:18 raspberrypi4box systemd[1]: deskpi.service: Main process exited, code=killed, status=15/TERM
Aug 08 20:25:18 raspberrypi4box systemd[1]: deskpi.service: Failed with result 'signal'.
Aug 08 20:25:18 raspberrypi4box systemd[1]: Stopped DeskPi PWM Control Fan Service.
```
I noticed there is a post on some of the issues in install script in here: https://www.gitmemory.com/issue/DeskPi-Team/deskpi/39/820007904 but it seems despite fixing the user issue in the install script, the deskpi service still is in failed state after reboot.

Is this intentational that the service fails? deskpi-config is still able to configure the fan - I can hear how its speed changes, so obviously something works..

I have updated to the latest version the deskpi repo to my linux box.

yoyojacky, 2021-10-13:
> If I choose option 4 and reboot, then no fan. I don't want to manually turn on the fan every time I reboot.
> 
> I haven't examined **deskpi-config** to see if I can hack it in my **/etc/rc.local** file.

fan controlled by MCU on daughterboard as following chart:

Raspberry Pi ------> USB-C (enabled dwc2 as dr_mode=host) generate a serial device called `/dev/ttyUSB0` -----> MCU on deskpi daughterboard.
when you execute `deskpi-config` and select `4` means:
Raspberry Pi ----> sending `pwm_100` to `/dev/ttyUSB0` ----> MCU get the signal `pwm_100` and control the fan at 100% pwm signal status, so the fan will spin at 100%.....
if your screen saver interrupt the sending procedure, it will stop sending `pwm_100` to serial port, MCU will turn off the fan...