DeskPi-Team/deskpi #59: Power button light, board in general, does not shutdown

inspector71, 2021-07-31:
How to shutdown the DeskPi Pro fully in software, meaning:

# power button LED changes from blue to red
# all but one board LED switches off

In other words, the same full shutdown experience I currently get when holding in the power button for a few seconds.

Is that not what the system is designed to do?

I'm not interested in and sort of standbye state here.

FWIW, I've yet to update any firmware.

pkr1771, 2021-08-29:
Posted in to another thread:

I encountered the issue of the power LED not going red on halt/shutdown/poweroff which also traced back to fanStop not functioning correctly. I'm running on 64-bit Ubuntu which I was happy to see now has installer support. The pwmFanControl works correctly (unlike previous releases) because it is recompiled during the installation process.

I needed to manually recompile the fanStop.c using gcc for the 64 bit platform and then copy the new binary to the /usr/bin directory. This file is not recompiled in the installation process and appears that it needs to be added.

One other difference I noted from earlier releases is the definition of the deskpi-safeshut.service. I can't say for sure that this makes a difference, but I noticed it as a change from earlier releases that included the line:

TimeoutSec=1

When I made these two changes, the system shuts down as expected and the power LED goes red.

inspector71, 2021-09-01:
@pkr1771 Thanks! There sounds like bugs here that @yoyojacky needs to fix. Having discovered the shitty nature of support for the DeskPi (OSMC cannot boot from USB; LibreELEC 9.*.* not fully supported and 10 is an unusable sea change; DietPi Kodi image has hardware decoding, reliability issues despite it being limited to 18.9; etc etc) I've become a bit stuck and demotivated.

Would be great if at least one boot-to-kodi OS was properly, smoothly supported in every respect.

inspector71, 2021-10-14:
> red led on power botton will always light up. it can not turn off.

FWIW, although LEDs generally much too bright for easy night time relaxing whilst watching a video, for example, I'm not than OK with a meaningful status colour change that actually indicates the unit is in a powered but 'off' state. 

That seems to mirror the behaviour of the Pi4 board itself but the Pi4 board's LEDs are hidden from the user. So it makes sense to have that LED visible to the user for the inevitable times when the Pi locks up, as it often settings to do due to arguably a lack of power filtering.

Ideal world a light

egooner, 2023-02-12:
> Posted in to another thread:
>
> I encountered the issue of the power LED not going red on halt/shutdown/poweroff which also traced back to fanStop not functioning correctly. I'm running on 64-bit [...]

Great - thanks for this now Feb 2023 and the bug is still there ... did exactly as you mention here and voila ... thanks for documenting this

jzazo, 2023-03-25:
I pulled from master, and reinstalled. After recompiling `fanStop.c` and moving it to `/usr/bin`, and adding `TimeoutSec=1` to the shutdown service, when I shut down the raspberry it actually shuts down, and the led turns red.

However, the fan service does not seem to work:
```
$ systemctl status deskpi.service
     Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; vendor preset: enabled)
     Active: failed (Result: signal) since Sat 2023-03-25 11:31:45 UTC; 20s ago
    Process: 2811 ExecStart=sudo /usr/bin/pwmFanControl & (code=killed, signal=SEGV)
   Main PID: 2811 (code=killed, signal=SEGV)
        CPU: 33ms

Mar 25 11:31:44 raspberry systemd[1]: Starting DeskPi PWM Control Fan Service...
Mar 25 11:31:44 raspberry sudo[2811]:     root : PWD=/ ; USER=root ; COMMAND=/usr/bin/pwmFanControl &
Mar 25 11:31:44 raspberry sudo[2811]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Mar 25 11:31:45 raspberry sudo[2811]: pam_unix(sudo:session): session closed for user root
Mar 25 11:31:45 raspberry systemd[1]: deskpi.service: Main process exited, code=killed, status=11/SEGV
Mar 25 11:31:45 raspberry systemd[1]: deskpi.service: Failed with result 'signal'.
Mar 25 11:31:45 raspberry systemd[1]: Failed to start DeskPi PWM Control Fan Service.
```

Do you know why it doesn't work, and do you know how to fix? Thanks.

jzazo, 2023-03-25:
I created this [PR](https://github.com/DeskPi-Team/deskpi/pull/140) to solve the service-shutdown problems discussed [above](https://github.com/DeskPi-Team/deskpi/issues/59#issuecomment-907876439)