DeskPi-Team/deskpi #67: Front of DeskPi shows red glow when Pi is turned on and when Pi is turned off. Normal?

johntdavis84, 2021-08-28:
The subject is pretty much it.

1. When I first assembled the DeskPi, before I got the fan control service running, it only showed a red glow when the Pi was off.
2. Sometime after getting the fan control software enabled, I'm now getting the red glow all the time.
3. I'm not seeing any instablity or 

The only thing I've done since then of note is overclock the Pi. It seems stable, but that MIGHT have something to do with it.

Is it supposed to glow red all the time? Or does this indicate a problem?

Yvandiel, 2021-08-29:
The red glow should be the power LED from the Pi, wich means the Pi is still on power mode.

wich system did you use on on the Pi? If mate or Ubuntu seems that the safeshutdown script didnt work correctly normaly it should put down the pi to nothing. Should also switch the Power button from blue to red. (also didnt get it to run on my version with ubuntu 64 bit and must manually power off the deskpi)

if the red is now on everytime add these lines in die boot.txt

# Power-LED from Raspberry Pi (red)
 dtparam=pwr_led_trigger=none
 dtparam=pwr_led_activelow=off

# Status-LED from Raspberry Pi (green)
 dtparam=act_led_trigger=none
 dtparam=act_led_activelow=off

these two commands should disable both leds on your pi as long as the system is bootet

johntdavis84, 2021-08-29:
Thanks!

I'm on 64-bit Manjaro Linux. I had to manually recompile the apps and manually create the service files, as the `manjaro.sh` script threw a bunch of errors.

I suspect I may have done something wrong there. **What's the difference between the `fanStop` and `safecutoffpower` apps?** I've yet to find an explanation for this, but I think I'm probably using the wrong one.

I _am_ getting the blue power LED when the device is on, and the red power LED when it's off. So, that works.

```
 ~]$ systemctl status deskpi-safecutoffpower.service 
○ deskpi-safecutoffpower.service - DeskPi Safe Shut-Off Power Service
     Loaded: loaded (/usr/lib/systemd/system/deskpi-safecutoffpower.service; enabled; vendor preset: disabled)
     Active: inactive (dead)
```

Here's my Safe Shut-Off service file:
```
 ~]$ cat /usr/lib/systemd/system/deskpi-safecutoffpower.service 
[Unit]
Description=DeskPi Safe Shut-Off Power Service
Conflicts=reboot.target
DefaultDependencies=no

[Service]
Type=oneshot
#ExecStart=/usr/bin/sudo /usr/bin/fanStop
ExecStart=/usr/bin/sudo /usr/bin/safecutoffpower
#ExecStart=/usr/bin/sudo python3 /usr/bin/safecutoffpower.py
RemainAfterExit=yes
TimeoutStartSec=15

[Install]
WantedBy=halt.target shutdown.target poweroff.target final.target
[victorsuede2011@Nicole ~]
```

johntdavis84, 2021-08-30:
**UPDATE:**

I switched to using `fanStop`, and now don't get the red glow when the unit is off. So, good. :) Thanks!

But that begs the question: what is actually supposed to invoke `safecutoffpower`?

closed 2021-10-13 by yoyojacky