DeskPi-Team/deskpi #68: Please add support for Ubuntu Budgie 21.04

saltedlolly, 2021-08-31:
I have tried using the install script for Ubuntu 64bit but it doesn't seem to work.

https://ubuntubudgie.org/2021/03/release-notes-ubuntu-budgie-21-04-on-a-raspberry-pi-4/

reqlez, 2021-09-04:
I used the ./install-ubuntu-64.sh script but the deskpi service hangs...  "Starting DeskPi PWM Control Fan Service..."

Ubuntu 21.04 (GNU/Linux 5.11.0-1012-raspi aarch64

● deskpi.service - DeskPi PWM Control Fan Service
     Loaded: loaded (/lib/systemd/system/deskpi.service; enabled; vendor preset: enabled)
     Active: activating (start) since Sat 2021-09-04 02:59:16 UTC; 1min 39s ago
   Main PID: 2642 (sudo)
      Tasks: 2 (limit: 9313)
     Memory: 1.0M
     CGroup: /system.slice/deskpi.service
             ├─2642 sudo /usr/bin/pwmFanControl &
             └─2644 /usr/bin/pwmFanControl &

Sep 04 02:59:16 ubuntu systemd[1]: Starting DeskPi PWM Control Fan Service...
Sep 04 02:59:16 ubuntu sudo[2642]:     root : PWD=/ ; USER=root ; COMMAND=/usr/bin/pwmFanControl &
Sep 04 02:59:16 ubuntu sudo[2642]: pam_unix(sudo:session): session opened for user root by (uid=0)

friendbear, 2021-10-05:
```sh
uname -a
Linux tubasan 5.11.0-1019-raspi #20-Ubuntu SMP PREEMPT Tue Sep 21 15:23:42 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux
```

occurred fatal error.

```
kumagai@tubasan:~/deskpi$ ./install-ubuntu-64.sh
 * DeskPi Fan control script installation Start.
 * DeskPi main control service loaded.
./install-ubuntu-64.sh: line 26: cd: /home//deskpi/drivers/c/: No such file or directory
mv: cannot stat '/home//deskpi/drivers/c/pwmFanControl': No such file or directory
gcc: error: /home//deskpi/drivers/c/pwmControlFan.c: No such file or directory
gcc: fatal error: no input files
compilation terminated.
```

yoyojacky, 2021-11-08:
> nstall-ubuntu-64.sh: line 26: cd: /home//deskpi/drivers/c/: No such file or directory
> mv: cannot stat '/home//deskpi/drivers/c/pwmFanControl': No such file or directory
> gcc: error: /home//deskpi/drivers/c/pwmControlFan.c: No such file or directory
> gcc: fatal error: no input files

please add `sudo` before your the command `./install-ubuntu-64.sh`

reqlez, 2021-12-13:
Okay but once the service is installed, and you restart it... is it stuck in "Active: activating..." for you as well?

reqlez, 2021-12-25:
Okay this is definitely not working right ... 
[image]

[image]

after reboot as you can see, services not starting. I used the process from the readme step by step for Ubuntu 64 and yes I used sudo and used the ubuntu 64 install file.

reqlez, 2021-12-25:
And the fan control file does exist, by the way:
[image]

reqlez, 2021-12-25:
And the fanstop file too:
[image]

reqlez, 2021-12-25:
okay nvm the fanStop is the power off command. 
[image]
I guess it stays as "activating" because how the code was written. But it does run it seems.

yoyojacky, 2022-03-02:
> okay nvm the fanStop is the power off command. [image] I guess it stays as "activating" because how the code was written. But it does run it seems.

your system is 64bit right? 
please replace that pwmFanControl file with pwmFanControl64 file which located to drivers/c/ folder in the repository and try again.

yoyojacky, 2023-11-23:
> And the fan control file does exist, by the way: [image]

After checking the script , i found that I miss the 64bit binary file.  you can just download the repository and navigate to deskpi/drivers/c/ folder and then, typing " make" to create a new binary file and copy it to your /usr/bin/ folder, reboot service will be ok. those names may confuse a lot of people,  i will rebuild all of the script again.

closed 2023-11-23 by yoyojacky